sle118/squeezelite-esp32 #348: Disable LMS not sufficiant for portable BT speaker Use Case

spudje, 2023-11-03:
### Describe the feature request

Issue https://github.com/sle118/squeezelite-esp32/issues/282 introduced the LMS/squeezelite disable button. This is a good option. However this is not sufficient for a typical use case for a portable (hey, it has a battery) BT speaker.

When at home, it will connect to wifi and will find the LMS server, so with LMS/squeezelite enabled no issues here.

When on the move, it will not find the LMS server and hence ends-up in the repeated rebooting, since squeezlite/LMS is enabled.
Now, when I'm on the move, it seems a big hassle to go and connect to the speaker over (ad-hoc/Muse AP) wifi to first turn off LMS/squeezelite to be able to use it as a portable BT speaker. And vice versa, coming back home having to turn it back on is a hassle.

So looking for a solution that simply stops the rebooting after not having found the LMS.. Or turn off LMS after a couple of unsuccessful searches for the server and turn it back on when connected again over wifi with the saved SSID.

### Preliminary Information

I'm on this release: 2.1634-16#v4.3#Muse#master-v4.3

### Hardware Details

Please describe your hardware setup:

ESP Muse Luxe

philippe44, 2023-11-06:
It has been released

spudje, 2023-11-06:
Super, that's awesome, thanks very much! I will try in the coming days!

philippe44, 2023-11-22:
I'll probably close that soon...

sle118, 2023-12-31:
Closing now because this has become stale

closed 2023-12-31 by sle118