sle118/squeezelite-esp32 #349: VU LED: any way to turn off the VU LED when going into sleep

bench2012, 2023-11-04:
Hi,

This is not a bug as such more like a query. I noted when I put the the Squeeze player into sleep, the ws2812 remains on. Any way to turn it off when go sleep?

Many thanks.

wizmo2, 2023-11-04:
Have not looked at how sleep works.  Currently, control of the strip relies on the LMS plugin.  I assume sleep shuts down the device before the server can react.   @philippe44, there is a led_vu.clear() function,  Is there somewhere where it can be called as part of the sleep activation?

philippe44, 2023-11-05:
Yes you do. In service.h, you have these 2 functions
```
void services_sleep_setsuspend(void (*hook)(void));
void services_sleep_setsleeper(uint32_t (*sleeper)(void));
```
If you want to be called when the system goes to sleep, register a function using `services_sleep_setsuspend`. This does not give a voting right, but this allows you to do whatever you want before the system goes to sleep.

The `services_sleep_setsleeper` sets a sleeper function that is polled regularly by the system and if you want to participate (it's a poor man's voting system, but nobody cares about voting right these days, no 😄), you register a callback and at every call, it returns the number of ms it has been inactive (not since last call, you don't need to remember that). Then the system will enter sleep if everybody has been inactive more than what the user has set.

This is very simple but I'd say it works for a small embedded platform.

sle118, 2023-11-08:
@wizmo2 just be aware that a bit of manual merging will be needed with the new development branch. I hope you will like the new environment and that it will help with your future contributions.

wizmo2, 2023-11-08:
@bench2012  , can you check out the latest release and see if it works for you?

bench2012, 2023-11-09:
sure, thanks.

bench2012, 2023-11-09:
Hi, works like a charm! Many thanks!