sle118/squeezelite-esp32 #355: Boot Loop after Flashing

SupperCraft5000, 2023-11-06:
I flashed my ESP32-WROOM-32U with the https://sle118.github.io/squeezelite-esp32-installer/ and then i stuck in a Boot Loop

```
ELF file SHA256: 0d4786f4a9ebfa71

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6424
load:0x40078000,len:15632
load:0x40080400,len:4348
entry 0x40080688
I (27) boot: ESP-IDF v4.3.5-dirty 2nd stage bootloader
I (27) boot: compile time 01:27:18
I (27) boot: chip revision: v3.0
I (31) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32: SPI Speed      : 80MHz
I (41) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 recovery         factory app      00 00 00010000 00140000
I (96) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (103) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (111) boot: End of partition table
I (115) boot: Defaulting to factory image
I (120) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=53750h (341840) map
I (223) esp_image: segment 1: paddr=00063778 vaddr=3ffbdb60 size=04848h ( 18504) load
I (229) esp_image: segment 2: paddr=00067fc8 vaddr=40080000 size=08050h ( 32848) load
I (240) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=d151ch (857372) map
I (476) esp_image: segment 4: paddr=00141544 vaddr=40088050 size=0c9fch ( 51708) load
I (493) esp_image: segment 5: paddr=0014df48 vaddr=400c0000 size=00068h (   104) load
I (504) boot: Loaded app from partition at offset 0x10000
I (504) boot: Disabling RNG early entropy source...
I (515) psram: This chip is ESP32-D0WD
E (515) psram: PSRAM ID read error: 0xffffffff
E (516) cpu_start: Failed to init external RAM, needed for external .bss segment
Re-enable cpu cache.

abort() was called at PC 0x400813bf on core 0

Backtrace:0x4008192d:0x3ffe3ba0 0x4008efe9:0x3ffe3bc0 0x400930b5:0x3ffe3be0 0x400813bf:0x3ffe3c50 0x40079514:0x3ffe3c80 |<-CORRUPTED
```

I tried to flash the current Image with an other Software but get the Same result.

SupperCraft5000, 2023-11-06:
[esp-web-tools-logs.txt](https://github.com/sle118/squeezelite-esp32/files/13266338/esp-web-tools-logs.txt)

sle118, 2023-11-06:
Make sure you read the documentation. You need a module with external ram on it and a WROOM is definitely not supported because it lacks this.

https://github.com/sle118/squeezelite-esp32#supported-hardware

closed 2023-11-06 by sle118