sle118/squeezelite-esp32 #359: dac_controlset "gpio":<gpio> not working

haldi4803, 2023-11-08:
### Describe the bug
Cannot toggle GPIO Pins

### Preliminary Information

1. **Firmware Version**:  1613 (Special Edition from Artefacts with GPIO Control!)

### Hardware Details

WaveShare EPS32 One

### NVS Settings

<details>
<pre><code>
{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "Merus",
  "a2dp_sink_name": "Merus",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "Merus",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "Merus",
  "autoexec": "1",
  "autoexec1": "squeezelite -o i2s -s 192.168.1.2:3483 -b 500:2000 -C 30 -d all=info -n Merus",
  "bat_config": "",
  "bt_name": "Merus",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\t\"deviceName\":\t\"Merus\",\t\"bitrate\":\t160,\t\"zeroConf\":\t1}",
  "dac_config": "model=I2S,bck=26,ws=32,do=25,sda=18,scl=23,i2c=32",
  "dac_controlset": "{\"poweron\":{\"gpio\":19,\"level\":1}}",
  "dhcp_tmout": "8",
  "display_config": "",
  "enable_airplay": "1",
  "enable_bt_sink": "1",
  "enable_cspot": "1",
  "equalizer": "0,0,0,0,0,0,0,0,0,0",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "Merus",
  "i2c_config": "scl=23,sda=18,speed=100000,port=0",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "",
  "lms_ctrls_raw": "n",
  "loudness": "0",
  "metadata_config": "",
  "model_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
  "pollmx": "600",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "CONFIG_SQUEEZELITE_ESP32_RELEASE_URL",
  "rotary_config": "",
  "set_GPIO": "",
  "sleep_config": "",
  "spdif_config": "",
  "spi_config": "",
[... 220 lines cut ...]
I (8402) httpd_handlers: serving /config.json to peer 192.168.1.1 port 11989
[00:22:33.255] slimproto:999 connected
[00:22:33.257] sendHELO:153 mac: 24:d7:eb:28:4f:58
[00:22:33.257] sendHELO:155 cap: Model=squeezeesp32,AccuratePlayPoints=1,HasDigitalOut=1,HasPolarityInversion=1,Balance=1,Depth=16,Firmware=v1.0-1613-16,ModelName=SqueezeESP32,MaxSampleRate=192000,alc,aac,ogg,ops,ogf,flc,aif,pcm,mp3
[00:22:33.285] notify:228 notified server 192.168.1.2 hport 9000 cport 9090
I (8482) network_status: LMS IP: 192.168.1.2, hport: 9000, cport: 9090
[00:22:33.293] decode_flush:251 decode flush
[00:22:33.293] output_flush:446 flush output buffer
[00:22:33.305] decode_flush:251 decode flush
[00:22:33.305] output_flush:446 flush output buffer
[00:22:33.309] equalizer_set_gain:165 equalizer gain 0,0,0,0,0,0,0,0,0,0
[00:22:33.322] equalizer_set_loudness:187 loudness 0
W (8782) httpd_txrx: httpd_sock_err: error in recv : 104
I (10092) httpd_handlers: serving /status.json to peer 192.168.1.1 port 15573
I (16992) httpd_handlers: serving /status.json to peer 192.168.1.1 port 16085
</code></pre>
</details>

### Issue Description

Tried GPIO1 and GPIO19 set to level 0 or level 1.
1 (pin 8 onboard) was always high and had 3.3V 19 (Pin 11 onboard) always low with 0V.
According to syntax checker this should be correct `{"init":{"gpio":1,"level":0}}`

Haven't tried another Board yet... but that shouldn't make any difference right?

philippe44, 2023-11-08:
But it does not seem that you have tried to play something. The DAC will be powered on only when starting to play.

haldi4803, 2023-11-08:
> But it does not seem that you have tried to play something. The DAC will be powered on only when starting to play.

i did about 4-5 reboots with different settings and pressing play in between.
Just took the log from the last run where i pressed play after i copied the log.

philippe44, 2023-11-09:
ok, found it: two things
1- there was a bug on my side where dac_controlset was not execute if there is no i2c (now it is needed for gpio)
2- your controlset init must be an array : `{"init":[{"gpio":1,"level":0}]}`

haldi4803, 2023-11-09:
This feature hasn't been included in the main build yet, right?

haldi4803, 2023-11-09:
It Partially works...
Setting set_GPIO 1=vcc gives Voltage at boot. Then using: `{"init":[{"gpio":1,"level":0},{"reg":53,"val":0},{"reg":54,"val":17},{"gpio":3,"level":1}]}` does set it low, so no more Voltage on GPIO1 it correctly sets Registers on the chip.
But then GPIO3 is always 0.9V no matter if i use level 0 or level 1

philippe44, 2023-11-09:
No, it is not supposed to work this way, there is no need to set it twice, in fact it could/should create issues

haldi4803, 2023-11-09:
<details>
<pre><code>
I (1642) esp_app_main: Setting up config subsystem.
W (1712) nvs_utilities: Configuration memory usage.  Heap internal:228159 (min:2                                          28019) (used:216) external:4055015 (min:4047167) (used:15668)
I (1712) esp_app_main: Registering default values
I (1722) system_api: Base MAC address is not set
I (1722) system_api: read default base MAC address from EFUSE
W (1732) config: Waiting for config commit ...
I (2712) config: configuration has some uncommitted entries
I (2712) config: Committing configuration to nvs. Locking config object.
I (2712) config: Done Committing configuration to nvs.
I (2712) config: Config committed!
I (2722) esp_app_main: Configuring services
I (2722) services: set GPIO 1 to vcc
I (2732) services: set GPIO 3 to vcc
I (2732) services: Configuring I2C sda:18 scl:23 port:0 speed:400000
I (2742) services: Configuring SPI mosi:-1 miso:-1 clk:-1 host:1 dc:-1
W (2752) services: no SPI configured
W (2752) led: LED GPIO -1 ignored
W (2752) led: LED GPIO -1 ignored
I (2762) led: Configuring LEDs green:-1 (on:0 rmt:-1 -1% ), red:-1 (on:0 rmt:-1                                           -1% )
I (2772) battery: No battery
I (2772) monitor: Initializing monitoring
I (2772) monitor: Heap internal:228403 (min:227831) external:4038439 (min:403841                                          1) dma:221323 (min:220767)
I (2782) esp_app_main: Initializing display
I (2792) display: Trying to configure display with N/A
W (2802) display: No display driver
I (2802) esp_app_main: Initializing led_vu
I (2802) led_vu: led_vu configuration invalid
I (2812) esp_app_main: Checking for update url
W (2812) config: Value not found for key fwurl
W (2822) config: Value not found for key
I (2822) esp_app_main: Starting Network Manager
I (2832) uart: queue free spaces: 3
W (2832) config: Value not found for key apdelay
I (2832) network_handlers: Network manager configuration: polling max 10176, pol                                          ling min 15000, ap delay 20000, dhcp timeout 8000, eth timeout 8000
I (2852) http_server: Initializing HTTP Server
I (2852) network_ethernet: Attempting to initialize Ethernet
I (2862) network_ethernet: No Ethernet configuration, or configuration invalid
I (2882) wifi:wifi driver task: 3ffdf9dc, prio:23, stack:6656, core=0
I (2882) wifi:wifi firmware version: 9ec65ef
I (2882) wifi:wifi certification version: v7.0
I (2882) wifi:config NVS flash: enabled
I (2892) wifi:config nano formating: disabled
I (2892) wifi:Init data frame dynamic rx buffer num: 40
I (2892) wifi:Init management frame dynamic rx buffer num: 40

I (2902) wifi:Init management short buffer num: 32
I (2902) wifi:Init static tx buffer num: 12
I (2912) wifi:Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

Init tx cache buffer num: 32
I (2932) wifi:Init static rx buffer size: 1600
I (2932) wifi:Init static rx buffer num: 12To automatically execute lines at sta                                          rtup:
        Set NVS variable autoexec (U8) = 1 to enable, 0 to disable automatic exe                                          cution.
        Set NVS variable autoexec[1~9] (string)to a command that should be execu                                          ted automatically

I (2952) wifi:Init dynamic rx buffer num: 40
I (2952) wifi_init: tcpip mbox: 32

I (2972) wifi_init: udp mbox: 32
I (2972) wifi_init: tcp mbox: 32
I (2972) wifi_init: tcp tx win: 8192

****************************
Your terminal application does not support escape sequences.
Line editing and history features are disabled.
On Windows, try using Putty instead.
[... 150 lines cut ...]
I (32502) monitor:             wifi (2)  4% s: 2180    btController (2)  0% s: 2                                          192       esp_timer (3)  0% s: 3260
I (32522) monitor:          console (2)  0% s:  868            ipc1 (2)  0% s:                                            464
I (42592) monitor: Heap internal:61211 (min:60647) external:742415 (min:720339)                                           dma:54131 (min:53583)
I (42602) monitor:             IDLE (1) 96% s:  948     pseudo_idle (0)  0% s: 3                                          316            IDLE (1) 99% s:  956
I (42602) monitor:           stream (2)  0% s: 2296      output_i2s (2)  0% s: 1                                          936             tiT (2)  0% s: 1884
I (42622) monitor:          Tmr Svc (2)  0% s: 1192     squeezelite (2)  0% s: 3                                          776          decode (2)  0% s:12656
I (42642) monitor:         BTC_TASK (2)  0% s:  868            hciT (2)  0% s: 1                                          568        BTU_TASK (2)  0% s: 3268
I (42662) monitor:             ipc0 (2)  0% s:  484         sys_evt (2)  0% s:                                            584            RTSP (2)  0% s: 7392
I (42682) monitor:   playerInstance (2)  0% s:29696          telnet (2)  0% s: 3                                          192         console (2)  0% s:  868
I (42682) monitor:          network (2)  0% s:  808           httpd (2)  0% s: 1                                          148            mdns (2)  0% s: 2344
I (42702) monitor:             wifi (2)  1% s: 2180    btController (2)  1% s: 2                                          192       esp_timer (3)  0% s: 3260
[... cut ...]
I (52882) monitor:   playerInstance (2)  0% s:29696         console (2)  0% s:                                            868            ipc1 (2)  0% s:  464
I (52882) monitor:             mdns (2)  0% s: 2344         network (2)  0% s:                                            808           httpd (2)  0% s: 1148
I (52902) monitor:             wifi (2)  2% s: 2180    btController (2)  1% s: 2                                          192       esp_timer (3)  0% s: 3260
I (52922) monitor:           BtAppT (2)  0% s: 1468        BTC_TASK (2)  0% s:                                            868
</pre></code>
</details>

philippe44, 2023-11-09:
ah crap ... I did not pay attention: you're using GPIO 1 and 3, which are the default UART and need a different reprogramming to be used as generic GPIO. I'll change that, but why using these? They are the last ones you want to use

haldi4803, 2023-11-09:
Not like i got a Choice.... It's a Board with standard 40pin Header fixed in place :(

philippe44, 2023-11-11:
Did you try latest update?

haldi4803, 2023-11-11:
Backup:
1=vcc, 3=vcc, 5=vcc, 15=vcc
{"init":[{"gpio":3,"level":0},{"delay":500},{"gpio":1,"level":0},{"reg":53,"val":0},{"reg":54,"val":17},{"gpio":3,"level":1}]}

Updated to 1650

And now all Pins work as epxected without an Set_GPIO.
got 3.3V on GPIO 5 and 3

<details>
<pre><code>
I (1637) esp_app_main: Setting up config subsystem.
W (1707) nvs_utilities: Configuration memory usage.  Heap internal:228207 (min:2                                                  28067) (used:216) external:4054955 (min:4047051) (used:15720)
I (1707) esp_app_main: Registering default values
I (1717) system_api: Base MAC address is not set
I (1717) system_api: read default base MAC address from EFUSE
W (1727) config: Waiting for config commit ...
I (2707) config: configuration has some uncommitted entries
I (2707) config: Committing configuration to nvs. Locking config object.
I (2707) config: Done Committing configuration to nvs.
I (2707) config: Config committed!
I (2717) esp_app_main: Configuring services
I (2717) services: Configuring I2C sda:18 scl:23 port:0 speed:400000
I (2727) services: Configuring SPI mosi:-1 miso:-1 clk:-1 host:1 dc:-1
W (2737) services: no SPI configured
W (2737) led: LED GPIO -1 ignored
W (2737) led: LED GPIO -1 ignored
I (2747) led: Configuring LEDs green:-1 (on:0 rmt:-1 -1% ), red:-1 (on:0 rmt:-1                                                   -1% )
I (2757) battery: No battery
I (2757) monitor: Initializing monitoring
I (2767) monitor: Heap internal:228451 (min:227879) external:4038367 (min:403833                                                  9) dma:221295 (min:220739)
I (2777) esp_app_main: Initializing display
I (2777) display: Trying to configure display with N/A
W (2787) display: No display driver
I (2787) esp_app_main: Initializing led_vu
I (2797) led_vu: led_vu configuration invalid
I (2797) esp_app_main: Checking for update url
W (2807) config: Value not found for key fwurl
W (2807) config: Value not found for key
I (2817) esp_app_main: Starting Network Manager
I (2817) uart: queue free spaces: 3
W (2817) config: Value not found for key apdelay
I (2817) network_handlers: Network manager configuration: polling max 10176, pol                                                  ling min 15000, ap delay 20000, dhcp timeout 8000, eth timeout 8000
I (2837) http_server: Initializing HTTP Server
I (2847) network_ethernet: Attempting to initialize Ethernet
I (2847) network_ethernet: No Ethernet configuration, or configuration invalid
I (2867) wifi:wifi driver task: 3ffdfd18, prio:23, stack:6656, core=0
I (2867) wifi:wifi firmware version: 9ec65ef
I (2867) wifi:wifi certification version: v7.0
I (2867) wifi:config NVS flash: enabled
I (2877) wifi:config nano formating: disabled
I (2877) wifi:Init data frame dynamic rx buffer num: 40
I (2877) wifi:Init management frame dynamic rx buffer num: 40

I (2887) wifi:Init management short buffer num: 32
I (2897) wifi:Init static tx buffer num: 12
I (2897) wifi:Init tx cache buffer num: 32Type 'help' to get the list of command                                                  s.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

I (2917) wifi:Init static rx buffer size: 1600
I (2917) wifi:Init static rx buffer num: 12
I (2917) wifi:To automatically execute lines at startup:
        Set NVS variable autoexec (U8) = 1 to enable, 0 to disable automatic exe                                                  cution.
        Set NVS variable autoexec[1~9] (string)to a command that should be execu                                                  ted automatically
Init dynamic rx buffer num: 40
I (2947) wifi_init: tcpip mbox: 32

I (2957) wifi_init: udp mbox: 32
I (2957) wifi_init: tcp mbox: 32
[... 196 lines cut ...]
I (63067) monitor: Heap internal:61955 (min:61091) external:750323 (min:736947)                                                   dma:54799 (min:53951)
I (63097) monitor:             IDLE (1) 63% s:  948     pseudo_idle (0)  0% s: 3                                                  312            IDLE (1) 88% s:  948
I (63107) monitor:           decode (2) 33% s:12660     squeezelite (2)  0% s: 3                                                  776             tiT (2)  2% s: 1880
I (63117) monitor:          Tmr Svc (2)  0% s: 1200          stream (2)  1% s: 2                                                  272        BTU_TASK (2)  0% s: 3304
I (63137) monitor:             ipc0 (2)  0% s:  484         sys_evt (2)  0% s:                                                    596            RTSP (2)  0% s: 7392
I (63157) monitor:   playerInstance (2)  0% s:29696          telnet (2)  0% s: 3                                                  184         console (2)  0% s:  876
I (63167) monitor:             ipc1 (2)  0% s:  464          BtAppT (2)  0% s: 1                                                  468            mdns (2)  0% s: 2348
I (63177) monitor:          network (2)  0% s:  812           httpd (2)  0% s: 1                                                  140    btController (2)  0% s: 2112
I (63187) monitor:             wifi (2)  4% s: 2188      output_i2s (2)  3% s: 2                                                  016       esp_timer (3)  0% s: 3260
I (63207) monitor:         BTC_TASK (2)  0% s:  872            hciT (2)  0% s: 1                                                  572
I (66227) telnet: We have a new client connection 57
[... cut ...]
I (73337) monitor:           BtAppT (2)  0% s: 1468        BTC_TASK (2)  0% s:  872            mdns (2)  0% s: 2348
I (73347) monitor:          network (2)  0% s:  812            wifi (2)  2% s: 2188       esp_timer (3)  0% s: 3260
I (73357) monitor:     btController (2)  0% s: 2112           httpd (2)  0% s: 1140      output_i2s (2)  0% s: 2016
I (73377) monitor:             hciT (2)  0% s: 1572        BTU_TASK (2)  0% s: 3304
</pre></code>
</details>

The board still doesn't output any sound but that's unrelated to GPIO issues.

closed 2023-11-11 by haldi4803