sle118/squeezelite-esp32 #360: ESP-32 Wroom bootloob

PoisonousWisper, 2023-11-12:
### Describe the bug
It seems like the ESP-32 Wroom is stuck inside a bootloob.
(The LED blinks rhythmically.)

1. Firmware Version: The most current one that is available with the Webinstaller (2.1650-32#v4.3#I2S-4MFlash#master-v4.3)

### Hardware Details

- **ESP32 Module**: ESP-WOROOM-32
- **Board Type**: Plain ESP32
- **DAC Chip**: N.A
- **Additional Hardware**: N.A

### Logs

This was displayed inside the Chrome debug console (looks normal but to be sure):

<details>
<pre><code>
Try hard reset.
Chip type ESP32
Uploading stub...
Running stub...
Stub is now running...
Detecting Flash Size
FlashId: 0x164068
Flash Manufacturer: 68
Flash Device: 4016
Auto-detected Flash size: 4MB
Image header, Magic=0xE9, FlashMode=0x02, FlashSizeFreq=0x2F
Writing data with filesize: 2639216. Compressed Size: 1741163
Took 154227ms to write 1741163 bytes
Erase size 0, blocks 0, block size 0x4000, offset 0x0000, encrypted no
Image header, Magic=0xE9, FlashMode=0x02, FlashSizeFreq=0x2F
Writing data with filesize: 1302816. Compressed Size: 868336
Took 76895ms to write 868336 bytes
Erase size 0, blocks 0, block size 0x4000, offset 0x0000, encrypted no
Image header, Magic=0xFF, FlashMode=0xFF, FlashSizeFreq=0xFF
Writing data with filesize: 8192. Compressed Size: 31
Took 10ms to write 31 bytes
Erase size 0, blocks 0, block size 0x4000, offset 0x0000, encrypted no
Image header, Magic=0xE9, FlashMode=0x02, FlashSizeFreq=0x2F
Writing data with filesize: 26496. Compressed Size: 16647
Took 1486ms to write 16647 bytes
Erase size 0, blocks 0, block size 0x4000, offset 0x0000, encrypted no
Image header, Magic=0xAA, FlashMode=0x01, FlashSizeFreq=0x02
Writing data with filesize: 3072. Compressed Size: 150
Took 22ms to write 150 bytes
Erase size 0, blocks 0, block size 0x4000, offset 0x0000, encrypted no
DISCONNECT
HARD RESET
Try hard reset.
</code></pre>
</details>

This log is from the ESP-32 itself (one full cicle, repeats afterwards):
<details>
<pre><code>
=== START OF LOG ===
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6424
load:0x40078000,len:15632
load:0x40080400,len:4348
entry 0x40080688
I (27) boot: ESP-IDF v4.3.5-dirty 2nd stage bootloader
I (27) boot: compile time 21:51:25
I (27) boot: chip revision: v3.0
I (31) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32: SPI Speed      : 80MHz
I (41) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 recovery         factory app      00 00 00010000 00140000
I (96) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (103) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (111) boot: End of partition table
I (115) boot: Defaulting to factory image
I (120) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=53830h (342064)                                                                                                                                                              map
I (223) esp_image: segment 1: paddr=00063858 vaddr=3ffbdb60 size=04848h ( 18504)                                                                                                                                                              load
I (229) esp_image: segment 2: paddr=000680a8 vaddr=40080000 size=07f70h ( 32624)                                                                                                                                                              load
I (240) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=d1580h (857472)                                                                                                                                                              map
I (476) esp_image: segment 4: paddr=001415a8 vaddr=40087f70 size=0cadch ( 51932)                                                                                                                                                              load
I (493) esp_image: segment 5: paddr=0014e08c vaddr=400c0000 size=00068h (   104)                                                                                                                                                              load
I (504) boot: Loaded app from partition at offset 0x10000
I (504) boot: Disabling RNG early entropy source...
I (515) psram: This chip is ESP32-D0WD
E (516) psram: PSRAM ID read error: 0xffffffff
E (516) cpu_start: Failed to init external RAM, needed for external .bss segment
Re-enable cpu cache.

abort() was called at PC 0x400813bf on core 0

Backtrace:0x4008192d:0x3ffe3ba0 0x4008efe9:0x3ffe3bc0 0x400930b5:0x3ffe3be0 0x40                                                                                                                                                             0813bf:0x3ffe3c50 0x40079514:0x3ffe3c80 |<-CORRUPTED

ELF file SHA256: d7e97c9cdee0304b

Rebooting...
=== END OF LOG ===   
</code></pre>
</details>

### Issue Description

1. **Observed Behavior**: The ESP-32 Doesn't create an AP to connect to
2. **Expected Behavior**: The ESP-32 Should boot
3. **Steps to Reproduce**: I just used the Webinstaller (https://sle118.github.io/squeezelite-esp32-installer/) and used an of the shelf ESP32-Wroom with 4MB Flash

Thanks for reading! I really want to get this awsome project to work on my ESP-32.
I also have tryed it on an ESP-32-S3 with the same result.

Renber78, 2023-11-12:
Hello, you need an ESP-WROVER (which includes a PSRAM), the ESP-WROOM does not have one.

haldi4803, 2023-11-12:
`E (516) psram: PSRAM ID read error: 0xffffffff
E (516) cpu_start: Failed to init external RAM, needed for external .bss segment`

yeah.... that's the line indicating that.
Seen that before as well if you don't pay enough attention at board selection.

PoisonousWisper, 2023-11-12:
Ok Thank you all for the help!
Amazon sellers write what they want..
Some even write both WROOM and WROVER into the same description :/
I'll then close the issue. 
Thanks again!!

closed 2023-11-12 by PoisonousWisper