sle118/squeezelite-esp32 #366: Audio Breaks up when using Cspot for 5+ hours

rickyfajar93, 2023-12-01:
Does anyone experiencing this issue?
cspot bitrate 320Kbps

currently running : 2.1652-16#v4.3#I2S-4MFlash#master-v4.3 on ESP32-CAM with 16Mflash & 8M PSRAM

Thankyou.

sle118, 2023-12-01:
There is a template for submitting issues. Please follow it as otherwise we spend too much time getting to the important details.

rickyfajar93, 2023-12-02:
Terribly sorry for the inconvenience , here's the complete log

`<27>[0m[00:00:50.267] <27>[0;31mD <27>[0;94mSpircHandler.cpp<27>[0m:69: Received subscription response<\r><\n>
<27>[0m[00:00:50.285] <27>[0;31mD <27>[0;37mMercurySession.cpp<27>[0m:252: Executing Mercury Request, type SEND<\r><\n>
[00:00:50.316] cspot_cmd_handler:429 CSpot volume 1024<\r><\n>
<27>[0m[00:00:50.702] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 178<\r><\n>
<27>[0m[00:00:50.705] <27>[0;31mD <27>[0;37mMercurySession.cpp<27>[0m:174: Received mercury packet<\r><\n>
<27>[0m[00:00:50.958] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 181<\r><\n>
<27>[0m[00:00:50.961] <27>[0;31mD <27>[0;94mSpircHandler.cpp<27>[0m:69: Received subscription response<\r><\n>
<27>[0m[00:00:50.986] <27>[0;31mD <27>[0;37mMercurySession.cpp<27>[0m:252: Executing Mercury Request, type SEND<\r><\n>
[00:00:51.020] cspot_cmd_handler:429 CSpot volume 2048<\r><\n>
<27>[0m[00:00:51.460] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 181<\r><\n>
<27>[0m[<27>[0m[00:0000::5100:.5100.463469] ] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 178<\r><\n>
<27>[0;31mD <27>[0;94mSpircHandler.cpp<27>[0m:69: Received subscription response<\r><\n>
<27>[0m[00:00:51.489] <27>[0;31mD <27>[0;37mMercurySession.cpp<27>[0m:252: Executing Mercury Request, type SEND<\r><\n>
[00:00:51.508] cspot_cmd_handler:429 CSpot volume 4096<\r><\n>
<27>[0m[00:00:51.515] <27>[0;31mD <27>[0;37mMercurySession.cpp<27>[0m:174: Received mercury packet<\r><\n>
<27>[0m[00:00:51.720] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 181<\r><\n>
<27>[0m[00:00:51.723] <27>[0;31mD <27>[0;94mSpircHandler.cpp<27>[0m:69: Received subscription response<\r><\n>
<27>[0m[00:00:51.740] <27>[0;31mD <27>[0;37mMercurySession.cpp<27>[0m:252: Executing Mercury Request, type SEND<\r><\n>
[00:00:51.764] cspot_cmd_handler:429 CSpot volume 12288<\r><\n>
<27>[0m[00:00:52.001] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 178<\r><\n>
<27>[0m[00:00:52.003] <27>[0;31mD <27>[0;37mMercurySession.cpp<27>[0m:174: Received mercury packet<\r><\n>
<27>[0m[00:00:52.251] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 178<\r><\n>
<27>[0m[00:00:52.<27>[0m254] [<27>[0;31mD 00:<27>[0;37mMercurySession.cpp<27>[0m00:174: :Received mercury packet<\r><\n>
52.255] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 181<\r><\n>
<27>[0m[00:00:52.269] <27>[0;31mD <27>[0;94mSpircHandler.cpp<27>[0m:69: Received subscription response<\r><\n>
<27>[0m[00:00:52.292] <27>[0;31mD <27>[0;37mMercurySession.cpp<27>[0m:252: Executing Mercury Request, type SEND<\r><\n>
[00:00:52.314] cspot_cmd_handler:429 CSpot volume 19456<\r><\n>
<27>[0m[00:00:52.511] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 181<\r><\n>
<27>[0m[00:00:52.516] <27>[0;31mD <27>[0;94mSpircHandler.cpp<27>[0m:69: Received subscription response<\r><\n>
<27>[0m[00:00:52.524] <27>[0;31mD <27>[0;37mMercurySession.cpp<27>[0m:252: Executing Mercury Request, type SEND<\r><\n>
[00:00:52.554] cspot_cmd_handler:429 CSpot volume 27648<\r><\n>
<27>[0m[00:00:52.744] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 178<\r><\n>
<27>[0m[00:00:52.746] <27>[0;31mD <27>[0;37mMercurySession.cpp<27>[0m:174: Received mercury packet<\r><\n>
<27>[0m[00:00:52.969] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 178<\r><\n>
<27>[0m[00:00:52.971] <27>[0;31mD <27>[0;37mMercurySession.cpp<27>[0m<27>[0m:174: Received mercury packet[<\r><\n>
00:00:52.972] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 181<\r><\n>
<27>[0m[00:00:52.987] <27>[0;31mD <27>[0;94mSpircHandler.cpp<27>[0m:69: Received subscription response<\r><\n>
<27>[0m[00:00:53.010] <27>[0;31mD <27>[0;37mMercurySession.cpp<27>[0m:252: Executing Mercury Request, type SEND<\r><\n>
[00:00:53.035] cspot_cmd_handler:429 CSpot volume 35839<\r><\n>
<27>[0m[00:00:53.236] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 181<\r><\n>
<27>[0m[00:00:53.239] <27>[0;31mD <27>[0;94mSpircHandler.cpp<27>[0m:69: Received subscription response<\r><\n>
<27>[0m[00:00:53.256] <27>[0;31mD <27>[0;37mMercurySession.cpp<27>[0m:252: Executing Mercury Request, type SEND<\r><\n>
[00:00:53.280] cspot_cmd_handler:429 CSpot volume 39935<\r><\n>
<27>[0m[00:00:53.481] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 181<\r><\n>
<27>[0m<27>[0m[[00:00:00:00:53.53485.] <27>[0;34mI 485] <27>[0;37mMercurySession.cpp<27>[0m<27>[0;31mD :42: <27>[0m<27>[0;94mSpircHandler.cpp<27>[0mReceived packet, command: 178<\r><\n>
:69: Received subscription response<\r><\n>
<27>[0m[00:00:53.503] <27>[0;31mD <27>[0;37mMercurySession.cpp<27>[0m:252: Executing Mercury Request, type SEND<\r><\n>
[00:00:53.540] cspot_cmd_handler:429 CSpot volume 44031<\r><\n>
<27>[0m[00:00:53.543] <27>[0;31mD <27>[0;37mMercurySession.cpp<27>[0m:174: Received mercury packet<\r><\n>
<27>[0m[00:00:53.722] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 178<\r><\n>
<27>[0m[<27>[0m[00:00:53.725] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 181<\r><\n>
[... cut ...]
  "rotary_config": "",
  "set_GPIO": "",
  "sleep_config": "",
  "spdif_config": "",
  "spi_config": "",
  "stats": "n",
  "target": "",
  "telnet_block": "500",
  "telnet_buffer": "40000",
  "telnet_enable": ""
}
`

currently running : 2.1652-16#v4.3#I2S-4MFlash#master-v4.3 on ESP32-CAM with 16Mflash & 8M PSRAM , PCM5102A DAC with MCLK (mck=0) in nvs

after a few day , i observe that audio breaks up and sometimes totally stopped , it can happen immediately or in between 3 - 5 hours of non-stop spotify playlists.

rickyfajar93, 2023-12-02:
Update : after leaving it on for a while without playing anything , it keeps receiving 
`<27>[0m[01:36:32.588] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 4<\r><\n>
<27>[0m[01:36:32.590] <27>[0;31mD <27>[0;91mTimeProvider.cpp<27>[0m:15: Time synced with spotify servers<\r><\n>
<27>[0m[01:36:32.834] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 74<\r><\n>
<27>[0m[01:38:32.601] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 4<\r><\n>
<27>[0m[01:38:32.603] <27>[0;31mD <27>[0;91mTimeProvider.cpp<27>[0m:15: Time synced with spotify servers<\r><\n>
<27>[0m[01:38:32.847] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 74<\r><\n>
<27>[0m[01:40:32.615] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 4<\r><\n>
<27>[0m[01:40:32.617] <27>[0;31mD <27>[0;91mTimeProvider.cpp<27>[0m:15: Time synced with spotify servers<\r><\n>
<27>[0m[01:40:32.902] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:42: <27>[0mReceived packet, command: 74<\r><\n>
`

rickyfajar93, 2023-12-02:
after investigating the uart further I think i found the problem

`<27>[0m[02:00:23.088] <27>[0;31mD <27>[0;94mSession.cpp<27>[0m:67: Connecting with AP **<ap-gew4.spotify.com:4070>**<\r><\n>
<27>[0m[02:00:23.330] <27>[0;31mD <27>[0;35mPlainConnection.cpp<27>[0m:101: Connected to spotify server<\r><\n>
<27>[0m[02:00:23.576] <27>[0;34mI <27>[0;94mSession.cpp<27>[0m:43: <27>[0mReceived APHello response<\r><\n>
<27>[0m[02:00:23.609] <27>[0;31mD <27>[0;94mSession.cpp<27>[0m:48: Received shannon keys<\r><\n>
<27>[0m[02:00:23.840] <27>[0;31mD <27>[0;94mSession.cpp<27>[0m:87: Authorization successful<\r><\n>
<27>[0m[02:00:23.843] <27>[0;34mI <27>[0;37mMercurySession.cpp<27>[0m:75: <27>[0mReconnection successful<\r><\n>
`

there's a problem with ap-gew4 that happened before , and i think it might happen again this time
[https://community.spotify.com/t5/App-Features/ap-gew4-spotify-com-not-functioning/td-p/5408508](url)
[https://community.spotify.com/t5/Other-Podcasts-Partners-etc/No-music-when-using-Spotty-in-Denmark/td-p/5349988](url)

are there any implementable workaround to blacklist certain ap-ge4 url or redirect it from the web interface?

or perhaps this is implementable?

> There are a few workarounds, trying to find the posts but can't right now
The solution is to manually assign an ip address to aap-gew4.spotify.com but I cannot find the address 
Edit, it's 104.199.65.124 , hopefully your router allows manual assignment 

i found it here : [https://community.spotify.com/t5/App-Features/ap-gew4-spotify-com-not-functioning/td-p/5408508](url)

sle118, 2023-12-02:
This would be something in the realm of @feelfreelinux as this is cspot related.

bungh0l10, 2023-12-03:
Hi,sorry for interrupting. Would it be an option (instead of manual override) to primarily use the fallback ap from http://apresolve.spotify.com/? instead of ap-gew4 or have the possibility on user side to do so?

cheers

rickyfajar93, 2023-12-03:
> Hi! Thanks for the thourough investigation. CSpot supports overriding an AP address, but I am not sure if its exposed in squeezelite's interface. A nicer option would be to detect broken APs, but I don't think there's an easy way to detect when Spotify's ap is broken - asides from recreating the entire connection, when we suspect that some playback failure was caused by ap

Is it doable in euphonium?

sle118, 2023-12-03:
Same problem because cspot runs on both 🙂

I am not sure if our shim exposes this option. @philippe44 would know as he integrated cspot

rickyfajar93, 2023-12-03:
> Same problem because cspot runs on both 🙂
> 
> I am not sure if our shim exposes this option. @philippe44 would know as he integrated cspot

I know , I'm just trying to 'sanity check' myself 😂 because i'm still having a few problems with euphonium.

feelfreelinux, 2023-12-05:
> Hi,sorry for interrupting. Would it be an option (instead of manual override) to primarily use the fallback ap from http://apresolve.spotify.com/? instead of ap-gew4 or have the possibility on user side to do so?
> 
> cheers 

Initially, I was kind of against blacklisting aps, as the situation with which currently works seems to change over time, and is also very dependent on the region. However, from what I see mostly ap-gew4 seems problematic, so maybe it is not as bad of a solution :) I think @rickyfajar93 had success with AP override in euphonium, but it's not exactly a very clear fix

rickyfajar93, 2023-12-05:
> Initially, I was kind of against blacklisting aps, as the situation with which currently works seems to change over time, and is also very dependent on the region. However, from what I see mostly [...]

After doing the override , it seems that occasionally after a few reboots , it comes back to ap-gew4
And there are times when the problem gets worse (1 song and then audio breaks up at the end and totally stops)

Since cspot gets the ap- list from ap resolve , is it possible to implement a drop down list where the user can select the ap- they prefer?

philippe44, 2023-12-05:
> Hi! Thanks for the thourough investigation. CSpot supports overriding an AP address, but I am not sure if its exposed in squeezelite's interface. A nicer option would be to detect broken APs, but I [...]
Couldn't we have a fault rating for the current AP and when a track fails we increase that rate so that we use that ap with a lower priority. In other words, we select from ap-list the ap's that we don't know yet or that don't have a high fault rating (or do round-robin on the ones with same rate, IDK the exact mechanism of ap-list, I would have to look at the code again, I kind of forgot that part). The fault rating may have a "forgiveness factor" with time. The rating can also be exponential so that bad actors really get punished and spend more time in the penalty box 😄

rickyfajar93, 2023-12-06:
UPDATE : i found [apresolve.spotify.com/?type=dealer&type=spclient]()

Can this be used as an alternative?
@sle118 
@feelfreelinux 
@philippe44 @

philippe44, 2023-12-07:
Not that it is the issue, but you are crazy low on internal ram... 32 bits version and spdif?

philippe44, 2023-12-07:
Does it happen if you use cspot natively on any platform?

rickyfajar93, 2023-12-07:
> Does it happen if you use cspot natively on any platform?

I tried on squeezelite & euphonium (dev branch) and got the same result

philippe44, 2023-12-07:
> > Does it happen if you use cspot natively on any platform?
> 
> I tried on squeezelite & euphonium (dev branch) and got the same result
> 
> But i haven't tried the cli version

No I mean capot on windows or Linux, the cli version

rickyfajar93, 2023-12-07:
> > > Does it happen if you use cspot natively on any platform?
> > 
> > I tried on squeezelite & euphonium (dev branch) and got the same result
> > 
> > But i haven't tried the cli version
> 
> No I mean capot on windows or Linux, the cli version

No , i haven't tried the cli version

philippe44, 2023-12-07:
> > > > Does it happen if you use cspot natively on any platform?
> > > 
> > > I tried on squeezelite & euphonium (dev branch) and got the same result
> > > 
> > > But i haven't tried the cli version
> > 
> > No I mean capot on windows or Linux, the cli version
> 
> No , i haven't tried the cli version (either in windows / linux)
>
- [ ] I think that it would tell us a lot about the cause of stuttering. In addition, investigating on these platforms is much easier.

rickyfajar93, 2023-12-13:
> * [ ]  I think that it would tell us a lot about the cause of stuttering. In addition, investigating on these platforms is much easier.

After doing more experiments starting from the weekend until today , i found that :
1. i'm still having some issues with euphonium (likely because i used version 1 of esp32)

2. observing apresolve.spotify.com , i found out that each time i refresh , the 4th data and the rest always changes
so i decided to build squeezelite and change ApResolve.cpp:35 to -> cJSON_GetArrayItem(cJSON_GetObjectItem(json, "ap_list"), 3)->valuestring); //originally 0
with this modification i can compare different ap with just a reboot. And after spending the whole weekend , doing comparison & stream test that ap-gew4 have some occasional stuttering every 2-3 hours during the day while others don't. another thing that i found out is when i let it stream overnight, ap-gew4 would stop playing in the morning when i wake up , but other ap's don't (needs more data) since i only observe it for 1 night and change to other ap's

could i suggest ApResolve.cpp:35 to -> cJSON_GetArrayItem(cJSON_GetObjectItem(json, "ap_list"), 3)->valuestring); so it "randomize" the AP used by cspot?

with this observation i think it temporarily solve the problem.

philippe44, 2023-12-13:
I understand where you want to go, but how is that randomizing? If this is an array of ap's, then we should really randomizingthe index we use.

rickyfajar93, 2023-12-13:
> I understand where you want to go, but how is that randomizing? If this is an array of ap's, then we should really randomizingthe index we use.

I think i should say , it's been randomized by spotify (the fourth data in the array) so we should just use it.

philippe44, 2023-12-14:
understood - is this documented (kind of) or just an observation?

rickyfajar93, 2023-12-14:
> understood - is this documented (kind of) or just an observation?

I haven't checked the documentation thoroughly but from what i read a few months ago in the spotify developer forum , the reason apresolve has 6 or sometimes more aplist is when 1 or more ap is unavailable , the system using it can fallback to the next index.

I don't know yet the reason behind the fourth data being rolled / randomized.

bungh0l10, 2023-12-18:
Maybe @mherger can chime in, as he is aware of the gew4 flaws from back in the days and can share opinions/thoughts on his LMS integration as it might be beneficial to others.

mherger, 2023-12-18:
My spotty/librespot implementation isn't trying to be smart about this. It's simply ignoring "ap-gew4.spotify.com" and "ap-gue1.spotify.com"... I'm way behind current librespot development, though. It looks as if they had changed (removed) that simplistic workaround.