sle118/squeezelite-esp32 #372: Can't save SPI settings - Javascript error thrown

JediMongoose, 2023-12-15:
To help us resolve your issue as quickly as possible, please follow these guidelines when submitting an issue. Providing all the necessary information will save both your time and ours.

### Describe the bug
When using the Web interface -> Hardware -> SPI Bus Parameters and entering values for Data Out GPIO, Data IN GPIO, Clock GPIO, DC GPIO, SPI Host Number and clicking "SAVE", I get the small dialog box "Executing", then a Javascript error is thrown in the console "Uncaught TypeError: r(....).html() is undefined"

[image]

### Preliminary Information

1. **Firmware Version**: I2S-4MFlash-1652
2. **Plugin Version**: N/A

### Hardware Details

Please describe your hardware setup:

- **ESP32 Module**: ESP32 WROVER
- **Board Type**: Generic ESP32
- **DAC Chip**: Generic PCM5102
- **Additional Hardware**: ST7789 screen (this is the thing I'm trying to get configured, but need to set up SPI first - then hit javascript error (see above))

### NVS Settings

<details>
<pre><code>
{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "squeezelite-51564",
  "a2dp_sink_name": "SMSL BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "squeezelite-51564",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "squeezelite-51564",
  "autoexec": "1",
  "autoexec1": "squeezelite  -o i2s  -s  http://192.168.1.2 -b  500:2000 -C  30 -d  all=info  -n \"ESP32 Player\"",
  "bat_config": "",
  "bt_name": "squeezelite-51564",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\"deviceName\":\"squeezelite-51564\",\"bitrate\":160,\"zeroConf\":1}",
  "dac_config": "model=I2S,bck=33,ws=25,do=32",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "",
  "enable_airplay": "Y",
  "enable_bt_sink": "Y",
  "enable_cspot": "N",
  "equalizer": "",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "squeezelite-51564",
  "i2c_config": "",
  "jack_mutes_amp": "y",
  "led_brightness": "",
  "led_vu_config": "",
  "lms_ctrls_raw": "n",
  "loudness": "0",
  "metadata_config": "",
  "model_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
  "pollmx": "600",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "CONFIG_SQUEEZELITE_ESP32_RELEASE_URL",
  "rotary_config": "",
  "set_GPIO": "",
  "sleep_config": "",
  "spdif_config": "",
  "spi_config": "",
  "stats": "N",
  "target": "",
  "telnet_block": "500",
  "telnet_buffer": "40000",
  "telnet_enable": "Y"
}
</code></pre>
</details>

### Logs

<details>
<pre><code>
=== START OF LOG ===
rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6424
load:0x40078000,len:15632
load:0x40080400,len:4348
entry 0x40080688
I (27) boot: ESP-IDF v4.3.5-dirty 2nd stage bootloader
I (27) boot: compile time 05:49:04
I (27) boot: chip revision: v1.0
I (30) qio_mode: Enabling QIO for flash chip GD
I (36) boot.esp32: SPI Speed      : 80MHz
I (40) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (73) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  3 recovery         factory app      00 00 00010000 00140000
I (96) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (103) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (111) boot: End of partition table
I (115) esp_image: segment 0: paddr=00150020 vaddr=3f400020 size=ae96ch (715116) map
I (320) esp_image: segment 1: paddr=001fe994 vaddr=3ffbdb60 size=01684h (  5764) load
I (322) esp_image: segment 2: paddr=00200020 vaddr=400d0020 size=1b2434h (1778740) map
I (816) esp_image: segment 3: paddr=003b245c vaddr=3ffbf1e4 size=043b8h ( 17336) load
I (822) esp_image: segment 4: paddr=003b681c vaddr=40080000 size=1e290h (123536) load
I (862) esp_image: segment 5: paddr=003d4ab4 vaddr=400c0000 size=00068h (   104) load
I (877) boot: Loaded app from partition at offset 0x150000
I (877) boot: Disabling RNG early entropy source...
I (889) psram: This chip is ESP32-D0WD
I (889) spiram: Found 64MBit SPI RAM device
I (889) spiram: SPI RAM mode: flash 80m sram 80m
I (892) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (900) cpu_start: Pro cpu up.
I (903) cpu_start: Starting app cpu, entry point is 0x40081404
I (0) cpu_start: App cpu up.
I (1406) spiram: SPI SRAM memory test OK
I (1419) cpu_start: Pro cpu start user code
I (1419) cpu_start: cpu freq: 240000000
I (1419) cpu_start: Application information:
I (1422) cpu_start: Project name:     Squeezelite-ESP32
I (1428) cpu_start: App version:      I2S-4MFlash-1652
I (1434) cpu_start: Compile time:     Nov 17 2023 05:50:36
I (1440) cpu_start: ELF file SHA256:  0000000000000000...
I (1446) cpu_start: ESP-IDF:          v4.3.5-dirty
I (1451) cpu_start: Min chip rev:     v1.0
I (1456) cpu_start: Max chip rev:     v3.99 
I (1461) cpu_start: Chip rev:         v1.0
I (1466) heap_init: Initializing. RAM available for dynamic allocation:
I (1473) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1479) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1486) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1492) heap_init: At 3FFC6EC8 len 00019138 (100 KiB): DRAM
I (1498) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1505) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1511) heap_init: At 4009E290 len 00001D70 (7 KiB): IRAM
I (1517) spiram: Adding pool of 4018K of external SPI memory to heap allocator
I (1526) spi_flash: detected chip: gd
I (1530) spi_flash: flash io: qio
I (1536) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1544) spiram: Reserving pool of 64K of internal memory for DMA/internal allocations
I (1554) esp_app_main: System is booting from power on.
I (1564) esp_app_main: Reset reason is: 7
I (1564) esp_app_main: Reboot counter=1

I (1564) esp_app_main: Starting app_main
I (1574) nvs_utilities: Initializing flash nvs 
I (1594) nvs_utilities: Initializing nvs partition settings
I (1644) esp_app_main: Setting up telnet.
I (1644) telnet: ***Redirecting log output to telnet
=== END OF LOG ===   
</code></pre>
</details>

### Issue Description

1. **Observed Behavior**: Can't save SPI settings - javascript error thrown in JS Console
2. **Expected Behavior**: SPI Settings should be saved
3. **Steps to Reproduce**: Enter values for SPI (as above), click "Save". I've tried both Firefox and Chrome - same problem.

Renber78, 2023-12-16:
I use exactly the same version as you and I don't encounter this problem. You should try another browser or disable your ADblocker and try again.
Otherwise, you can also edit directly in the NVS Editor to activate in "Credits".
Find the line: "spi_config" and put this: "mosi=23,clk=18,dc=14,miso=19" (Obviously adapt your I/O)

wizmo2, 2023-12-16:
> I use exactly the same version as you and I don't encounter this problem. 

The current build needs a mosi and a miso (data and data in) event though miso is supposed to be optional.  
As @sle118 is completely reworking this portion, you'll need to either put a spare gpio value in, or use NVS

JediMongoose, 2023-12-17:
Thanks for your responses. After much head scratching I realised the problem - I was trying to re-use a GPIO pin that was already assigned to PSRAM. If I enter an unassigned pin value, it works OK.

JediMongoose, 2023-12-17:
closing issue

closed 2023-12-17 by JediMongoose