sle118/squeezelite-esp32 #373: SSD1306 display flips but does not rotate

jds11111, 2023-12-16:
My display does not rotate 180 degrees, although it flips fine. 

I am using the latest firmware (1652-16) on Philippe's latest batch of squeezeamp, with the recommended SSD1306 display. Everything else about the display works fine.

### NVS Settings

NVS settings:

<details>
<pre><code>

  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "Upstairs Livingroom SqueezeAMP",
  "a2dp_sink_name": "SMSL BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "Upstairs Livingroom SqueezeAMP",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "Upstairs Livingroom SqueezeAMP",
  "autoexec": "1",
  "autoexec1": "squeezelite -o I2S -b 500:2000 -d all=info -C 30 -W -n \"Upstairs_Livingroom_SqueezeAMP\"",
  "bat_config": "",
  "bt_name": "Upstairs Livingroom SqueezeAMP",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\t\"deviceName\":\t\"Upstairs Livingroom SqueezeAMP\",\t\"bitrate\":\t160,\t\"zeroConf\":\t1}",
  "dac_config": "",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "I2C,width=128,height=32,address=60,driver=SSD1306,rotate",
  "enable_airplay": "Y",
  "enable_bt_sink": "Y",
  "enable_cspot": "Y",
  "equalizer": "0,0,0,0,0,0,0,0,0,0",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "Upstairs Livingroom SqueezeAMP",
  "i2c_config": "scl=1,sda=3,speed=600000,port=1",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "",
  "lms_ctrls_raw": "n",
  "loudness": "0",
  "metadata_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
  "pollmx": "600",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "https://github.com/sle118/squeezelite-esp32/releases",
  "rotary_config": "",
  "set_GPIO": "0=ir",
  "sleep_config": "",
  "spdif_config": "",
  "spi_config": "",
  "stats": "N",
  "target": "",
  "telnet_block": "500",
  "telnet_buffer": "40000",
  "telnet_enable": "Y"
</code></pre>
</details>

### Logs

Please let me know if you need logs.

sle118, 2023-12-19:
Logs are always welcomed, especially when no other user reported a similar issue.

jds11111, 2023-12-19:
OK, I figured that was the case. But that will have to wait. I would have to take one of my (4) players out of commission for a day to connect a UART to the pins to do that. (It does the same for all players, and I cannot figure out how to get logs from telnet.)

sle118, 2023-12-31:
you can also try enabling telnet

sle118, 2023-12-31:
Going to close for now. Feel free to reopen when you get some logs.

closed 2023-12-31 by sle118

fvzeppelin, 2024-02-04:
No logs, but at least I can confirm the described behavior: Rotation has no effect, but can be done by activating both flips.