sle118/squeezelite-esp32 #377: compiling fro ESP32S3: esp_eth.h: No such file or directory

mockerl, 2023-12-27:
To help us resolve your issue as quickly as possible, please follow these guidelines when submitting an issue. Providing all the necessary information will save both your time and ours.

### Describe the bug
A clear and concise description of what the bug is.
When trying to compile for ESP32s3 an error is thrown that "fatal error: esp_eth.h: No such file or directory"

### Preliminary Information

1. **Firmware Version**: I am using the 4.3 branch
2. **Plugin Version**: N/A

### Hardware Details

Please describe your hardware setup:

- **ESP32 Module**: generic ESP32S3

<details>
<pre><code>
  [1219/1351] Building C object esp-idf/wifi-manager/CMakeFiles/__idf_wifi-manager.dir/network_driver_LAN8720.c.obj
FAILED: esp-idf/wifi-manager/CMakeFiles/__idf_wifi-manager.dir/network_driver_LAN8720.c.obj
ccache /opt/esp/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -DHAVE_CONFIG_H -DHIERARCHICAL_STATES=1 -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMODEL_NAME=SqueezeESP32 -DSTATE_MACHINE_LOGGER=1 -I/workspace/squeezelite-esp32/build/config -I/workspace/squeezelite-esp32/components/wifi-manager -I/workspace/squeezelite-esp32/components/wifi-manager/webapp -I/workspace/squeezelite-esp32/components/wifi-manager/UML-State-Machine-in-C/src -I/opt/esp/idf/components/newlib/platform_include -I/opt/esp/idf/components/freertos/include -I/opt/esp/idf/components/freertos/include/esp_additions/freertos -I/opt/esp/idf/components/freertos/port/xtensa/include -I/opt/esp/idf/components/freertos/include/esp_additions -I/opt/esp/idf/components/esp_hw_support/include -I/opt/esp/idf/components/esp_hw_support/include/soc -I/opt/esp/idf/components/esp_hw_support/include/soc/esp32s3 -I/opt/esp/idf/components/esp_hw_support/port/esp32s3/. -I/opt/esp/idf/components/esp_hw_support/port/esp32s3/private_include -I/opt/esp/idf/components/heap/include -I/opt/esp/idf/components/log/include -I/opt/esp/idf/components/lwip/include/apps -I/opt/esp/idf/components/lwip/include/apps/sntp -I/opt/esp/idf/components/lwip/lwip/src/include -I/opt/esp/idf/components/lwip/port/esp32/include -I/opt/esp/idf/components/lwip/port/esp32/include/arch -I/opt/esp/idf/components/soc/include -I/opt/esp/idf/components/soc/esp32s3/. -I/opt/esp/idf/components/soc/esp32s3/include -I/opt/esp/idf/components/hal/esp32s3/include -I/opt/esp/idf/components/hal/include -I/opt/esp/idf/components/hal/platform_port/include -I/opt/esp/idf/components/esp_rom/include -I/opt/esp/idf/components/esp_rom/include/esp32s3 -I/opt/esp/idf/components/esp_rom/esp32s3 -I/opt/esp/idf/components/esp_common/include -I/opt/esp/idf/components/esp_system/include -I/opt/esp/idf/components/esp_system/port/soc -I/opt/esp/idf/components/esp_system/port/public_compat -I/opt/esp/idf/components/xtensa/include -I/opt/esp/idf/components/xtensa/esp32s3/include -I/opt/esp/idf/components/driver/include -I/opt/esp/idf/components/driver/esp32s3/include -I/opt/esp/idf/components/esp_pm/include -I/opt/esp/idf/components/esp_ringbuf/include -I/opt/esp/idf/components/efuse/include -I/opt/esp/idf/components/efuse/esp32s3/include -I/opt/esp/idf/components/vfs/include -I/opt/esp/idf/components/esp_wifi/include -I/opt/esp/idf/components/esp_event/include -I/opt/esp/idf/components/esp_netif/include -I/opt/esp/idf/components/tcpip_adapter/include -I/opt/esp/idf/components/esp_phy/include -I/opt/esp/idf/components/esp_phy/esp32s3/include -I/opt/esp/idf/components/esp_ipc/include -I/opt/esp/idf/components/app_trace/include -I/opt/esp/idf/components/esp_timer/include -I/workspace/squeezelite-esp32/components/squeezelite-ota -I/opt/esp/idf/components/app_update/include -I/opt/esp/idf/components/spi_flash/include -I/opt/esp/idf/components/bootloader_support/include -I/opt/esp/idf/components/esp_https_ota/include -I/opt/esp/idf/components/esp_http_client/include -I/opt/esp/idf/components/nghttp/port/include -I/opt/esp/idf/components/nghttp/nghttp2/lib/includes -I/opt/esp/idf/components/json/cJSON -I/opt/esp/idf/components/mdns/include -I/opt/esp/idf/components/console -I/workspace/squeezelite-esp32/components/tools -I/opt/esp/idf/components/pthread/include -I/workspace/squeezelite-esp32/components/services -I/workspace/squeezelite-esp32/components/platform_config -I/opt/esp/idf/components/nvs_flash/include -I/opt/esp/idf/components/mbedtls/port/include -I/opt/esp/idf/components/mbedtls/mbedtls/include -I/opt/esp/idf/components/mbedtls/esp_crt_bundle/include -I/workspace/squeezelite-esp32/components/display -I/workspace/squeezelite-esp32/components/display/fonts -I/workspace/squeezelite-esp32/components/display/core -I/opt/esp/idf/components/wpa_supplicant/include -I/opt/esp/idf/components/wpa_supplicant/port/include -I/opt/esp/idf/components/wpa_supplicant/esp_supplicant/include -I/workspace/squeezelite-esp32/components/platform_console -I/opt/esp/idf/components/esp_http_server/include -mlongcalls  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -fmacro-prefix-map=/workspace/squeezelite-esp32=. -fmacro-prefix-map=/opt/esp/idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4.6\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/wifi-manager/CMakeFiles/__idf_wifi-manager.dir/network_driver_LAN8720.c.obj -MF esp-idf/wifi-manager/CMakeFiles/__idf_wifi-manager.dir/network_driver_LAN8720.c.obj.d -o esp-idf/wifi-manager/CMakeFiles/__idf_wifi-manager.dir/network_driver_LAN8720.c.obj -c /workspace/squeezelite-esp32/components/wifi-manager/network_driver_LAN8720.c
/workspace/squeezelite-esp32/components/wifi-manager/network_driver_LAN8720.c:1:10: fatal error: esp_eth.h: No such file or directory
 #include "esp_eth.h"
          ^~~~~~~~~~~
compilation terminated.
[1220/1351] Building C object esp-idf/wifi-manager/CMakeFiles/__idf_wifi-manager.dir/network_driver_DM9051.c.obj
FAILED: esp-idf/wifi-manager/CMakeFiles/__idf_wifi-manager.dir/network_driver_DM9051.c.obj
ccache /opt/esp/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -DHAVE_CONFIG_H -DHIERARCHICAL_STATES=1 -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMODEL_NAME=SqueezeESP32 -DSTATE_MACHINE_LOGGER=1 -I/workspace/squeezelite-esp32/build/config -I/workspace/squeezelite-esp32/components/wifi-manager -I/workspace/squeezelite-esp32/components/wifi-manager/webapp -I/workspace/squeezelite-esp32/components/wifi-manager/UML-State-Machine-in-C/src -I/opt/esp/idf/components/newlib/platform_include -I/opt/esp/idf/components/freertos/include -I/opt/esp/idf/components/freertos/include/esp_additions/freertos -I/opt/esp/idf/components/freertos/port/xtensa/include -I/opt/esp/idf/components/freertos/include/esp_additions -I/opt/esp/idf/components/esp_hw_support/include -I/opt/esp/idf/components/esp_hw_support/include/soc -I/opt/esp/idf/components/esp_hw_support/include/soc/esp32s3 -I/opt/esp/idf/components/esp_hw_support/port/esp32s3/. -I/opt/esp/idf/components/esp_hw_support/port/esp32s3/private_include -I/opt/esp/idf/components/heap/include -I/opt/esp/idf/components/log/include -I/opt/esp/idf/components/lwip/include/apps -I/opt/esp/idf/components/lwip/include/apps/sntp -I/opt/esp/idf/components/lwip/lwip/src/include -I/opt/esp/idf/components/lwip/port/esp32/include -I/opt/esp/idf/components/lwip/port/esp32/include/arch -I/opt/esp/idf/components/soc/include -I/opt/esp/idf/components/soc/esp32s3/. -I/opt/esp/idf/components/soc/esp32s3/include -I/opt/esp/idf/components/hal/esp32s3/include -I/opt/esp/idf/components/hal/include -I/opt/esp/idf/components/hal/platform_port/include -I/opt/esp/idf/components/esp_rom/include -I/opt/esp/idf/components/esp_rom/include/esp32s3 -I/opt/esp/idf/components/esp_rom/esp32s3 -I/opt/esp/idf/components/esp_common/include -I/opt/esp/idf/components/esp_system/include -I/opt/esp/idf/components/esp_system/port/soc -I/opt/esp/idf/components/esp_system/port/public_compat -I/opt/esp/idf/components/xtensa/include -I/opt/esp/idf/components/xtensa/esp32s3/include -I/opt/esp/idf/components/driver/include -I/opt/esp/idf/components/driver/esp32s3/include -I/opt/esp/idf/components/esp_pm/include -I/opt/esp/idf/components/esp_ringbuf/include -I/opt/esp/idf/components/efuse/include -I/opt/esp/idf/components/efus [... cut ...]
### Issue Description

1. **Observed Behavior**: esp_eth.h dependency is missing
2. **Expected Behavior**: missing dependencies should be where the compiler expects them
3. **Steps to Reproduce**:  Recursively cloning the repository, starting the sle118/squeezelite-esp32-idfv446 docker (idk if that is officially supported yet), `idf.py set-target esp32s3`, in the menuconfig selecting external SPI-4. ram and octal mode, disabling Bluetooth, disabling Bluetooth as an audio input, selecting cspot as input.

sle118, 2023-12-30:
The branch you are trying to build uses : sle118/squeezelite-esp32-idfv435

sle118, 2023-12-31:
Going to close for now. Feel free to reopen if needed

closed 2023-12-31 by sle118

mockerl, 2024-01-01:
> Unless you are trying to develop a feature, there should not be a need to build your own binaries. All the features are configurable via NVS.

Thanks for the quick reply. 
What I was trying to achieve was to get a binary for the ESP32S3. As far as I can tell, the released binarys as well as the webinstaller are not compatible with the ESP32S3?

przemeklagod, 2024-01-02:
Hello all, I also got this issue. Additionally branch prepared for S3 can be compiled but it stuck on startup. It goes well but after wifi init goes to recovery app and halt:

RECOVERY APPLICATION
This mode is used to flash Squeezelite into the OTA partition
****

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

assert failed: prvNotifyQueueSetContainer queue.c:3092 (pxQueueSetContainer->uxM                                                    essagesWaiting < pxQueueSetContainer->uxLength)

Backtrace: 0x4037623a:0x3fc98660 0x40380375:0x3fc98680 0x40385a75:0x3fc986a0 0x4                                                    0380c9e:0x3fc987c0 0x40380e6b:0x3fc987f0 0x4202ed89:0x3fc98820 0x403774fd:0x3fc9                                                    8880 0x420b8e17:0x3fcf5230 0x42006025:0x3fcf5250 0x420bf6a3:0x3fcf5270

ELF file SHA256: 8e7aa6c54aca3dd3

CPU halted.