sle118/squeezelite-esp32 #390: If Wifi drops while playing through cspot, squeezeamp gets stuck

bungh0l10, 2024-01-30:
### Describe the bug
Cspot on Squeezeamp will get stuck if WiFi drops.
Usually while playing music through LMS, Squeezeamp will reboot/ rediscover connection as soon as you loose WiFi or restart LMS (eg. update to latest LMS). 
Now when you are listening through cspot instead, the Squeezeamp will get in a stuck state (LED will stay green (playing state), you need to powercycle/ press reset button on pcb). 
On the other hand stopping / restarting LMS pretty much has no sideeffects while listening through cspot but as soon as Wifi drops you need to "recover" the squeezeamp.

### Preliminary Information

1. 1670
2. 0.600
3. 3. LMS latest 8.4

### Hardware Details

- **Squeezeamp v3.20**

Sorry guys but I couldn't provide logs for now because I haven't had a squeezamp without housing available and don't wanted to dissamble one --- but definetly will if you can not reproduce on your end, just let me know ;)

sle118, 2024-01-31:
Definitely enable Telnet to get some logs as it helps to understand the issue was better than a description. This is the first step before I/we even attempt to reproduce as logs contain precious clues about what might be happening

bungh0l10, 2024-01-31:
Sorry but I can't provide sufficient logs from/through telnet because if I turn WiFi off (to reproduce the mentioned issue) there won't be wireless connection for any device anymore and telnet could not output anything of information because of that :) . 
Anyways I'll try to find some time within the next days to figure out if I can get some useful data through serial and update you.

bungh0l10, 2024-01-31:
[390_issue_complete.log](https://github.com/sle118/squeezelite-esp32/files/14110643/390_issue_complete.log)

I hope the serial log helps :)

thanks & cheers

sle118, 2024-02-01:
I have re-written a good deal of the wifi AP/STA handling in the new refactoring branch I've been working on since last September (see: https://github.com/sle118/squeezelite-esp32/discussions/342) so I think I will park this issue for now until I get to test in the new environment. The new capabilities supports multiple known access points, allowing better roaming capabilities. The main scenario being when you take a portable speaker and bring it to a different home for example. I still need to test the roaming piece and make sure it behaves well.

as for the watchdog being triggered by cspot, this is something that could also perhaps happen even when the system is able to roam to different APs.    @philippe44  @feelfreelinux ?