sle118/squeezelite-esp32 #391: Does Cspot comes with /handles different WiFi capabilities?

bungh0l10, 2024-01-30:
### Describe the bug
I'm running a somewhat WiFi router through a rpi5 running LMS upon having a LTE USB dongle attached and an ALFA AWUS036ACH WifFi antenna to spread signal.
`Bus 001 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
`
As lacking an inbuild kernel driver, I'm using the one from here https://github.com/morrownr/8812au-20210629 .
So...  while the squeezeamp using LMS and cspot work well with standard settings (144MBs) in hostapd.conf 

`country_code=DE
bridge=br0
interface=wlan0
driver=nl80211
ssid=XXXXXXXX
hw_mode=g
channel=1
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
ieee80211n=1
ht_capab=[SHORT-GI-20][DSSS_CCK-40][TX-STBC-2BY1][RX-STBC-1][MAX-AMSDU-7935]
ieee80211d=1
wmm_enabled=1
wpa=2
wpa_passphrase=XXXXXXXXXXX
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
`

I am encountering only stutter using Cspot  after enabling 300MBs
` ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][RX-STBC-1][MAX-AMSDU-7935]
`
Playing Music through LMS on the other hand works flawlessly, without any stutter and also using the spotty plugin from Michael doesn't stutter as well.

Telnet unfortuately doesn't output anything on this stutter, only the usual informations you see while playing a song through cspot  :( 
As I know this might be an edgecase for sure I still  wanted to mention i,t as it wonders me why only cpsot might suffer from increased bandwith.
 
Cheers

### Preliminary Information

1. **Firmware Version: 1670**
2. **Plugin Version: 0.600**
3. **LMS: latest 8.4**
### Hardware Details

Please describe your hardware setup:

- **Squeezeamp v3.20**

bungh0l10, 2024-01-30:
[image]

has been tested with 3 😅

sle118, 2024-01-31:
Thinking that you might be hitting some incompatibilities on the WiFi PHY layer in which case there isn't much we could do since wifi is provided by the esp-idf. Not sure if that could help, but I'm currently refactoring the platform and compiling against esp-idf 4.4.6. the jump requires a full reflash, though, and I would need to test a squeezeamp build first, but let me know if you are willing to try

bungh0l10, 2024-01-31:
Sure I'd like to give it a try.

closed 2024-02-27 by bungh0l10