sle118/squeezelite-esp32 #394: Feature request: Possibility to configure lms cli port

fvzeppelin, 2024-02-04:
The current software seems to have port 9090 hardcoded for the lms cli port. The possibility to configure this port would be great in order to easily set the software up for an lms with a different port setting.

sle118, 2024-02-04:
Have you tried with a LMS configured with a different port? If so, please share logs. If not, please try.

https://github.com/sle118/squeezelite-esp32/blob/7b9deb795cf868dd1efa15e7e1109cd40b12a288/components/squeezelite/slimproto.c#L851

Seems to me that the discovery process should take care of setting ports automatically, but this isn't my area of expertise. @philippe44 ?

fvzeppelin, 2024-02-04:
yes, I agree that it should be discovered. I was requesting this because my ssd1306 displays only the ip and the port of the server, nothing else, no metadata when playing. But seems like I was on the wrong track here.

fvzeppelin, 2024-02-04:
Ah, now that I activated the esp32 plugin on server-side, everything is perfectly fine :)

closed 2024-02-04 by fvzeppelin