sle118/squeezelite-esp32 #398: ESP32-S3 - CORRUPT HEAP: Bad tail after wifi connection.

KamilKobylarek, 2024-02-18:
Hey, 

I have problem with ESP32-S3, device just crash after wifi connection in AP Mode with log _CORRUPT HEAP: Bad tail at 0x3d81cbef. Expected 0xbaad5678 got 0xbaad5600_

I build image for ESP32-S3-DEVKIT using:

- sle118/squeezelite-esp32-idfv446 docker image
- using master-v4.3 branch with sdkconfig from 4.4-s3
- changed flash from 8 to 16 MB and FreeRTOS Number of thread local storage pointers from 1 to 2
- flash by command:
`python3 -m esptool -p /dev/cu.usbmodem14201 -b 460800 --before default_reset --after hard_reset --chip esp32-s3  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x0 bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 recovery.bin 0x150000 squeezelite.bin`

Logs from usb:

> Using '/dev/cu.usbmodem14201' as serial port.
> Showing logs:
> [12:27:30]8 2024 11:23:33
> [12:27:30]I (881) cpu_start: ELF file SHA256:  ccbdc8a869da6226...
> [12:27:30]I (887) cpu_start: ESP-IDF:          v4.4.6
> [12:27:30]I (891) cpu_start: Min chip rev:     v0.0
> [12:27:30]I (896) cpu_start: Max chip rev:     v0.99 
> [12:27:30]I (901) cpu_start: Chip rev:         v0.2
> [12:27:30]I (906) heap_init: Initializing. RAM available for dynamic allocation:
> [12:27:30]I (913) heap_init: At 3FCA7A20 len 00041CF0 (263 KiB): D/IRAM
> [12:27:30]I (919) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
> [12:27:30]I (926) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
> [12:27:30]I (932) heap_init: At 600FE02C len 00001FD4 (7 KiB): RTCRAM
> [12:27:30]I (939) spiram: Adding pool of 8192K of external SPI memory to heap allocator
> [12:27:30]I (1186) spi_flash: detected chip: winbond
> [12:27:30]I (1186) spi_flash: flash io: qio
> [12:27:30]I (1187) sleep: Configure to isolate all GPIO pins in sleep state
> [12:27:30]I (1190) sleep: Enable automatic switching of GPIO sleep configuration
> [12:27:30]I (1198) cpu_start: Starting scheduler on PRO CPU.
> [12:27:30]I (0) cpu_start: Starting scheduler on APP CPU.
> [12:27:30]I (1208) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
> [12:27:30]I (1218) esp_app_main: System is booting from power on.
> [12:27:30]I (1228) esp_app_main: Reset reason is: 1
> [12:27:30]I (1228) esp_app_main: Recovery Reboot counter=1
> [12:27:30]
> [12:27:30]I (1238) esp_app_main: Starting app_main
> [12:27:30]I (1238) nvs_utilities: Initializing flash nvs 
> [12:27:30]I (1258) nvs_utilities: Initializing nvs partition settings
> [12:27:30]I (1298) esp_app_main: Setting up telnet.
> [12:27:30]I (1298) telnet: Telnet support disabled
> [12:27:30]I (1298) esp_app_main: Setting up config subsystem.
> [12:27:30]W (1358) nvs_utilities: Configuration memory usage.  Heap internal:283359 (min:282919) (used:16292) external:8384131 (min:8377311) (used:0)
> [12:27:30]I (1358) esp_app_main: Registering default values
> [12:27:30]I (1368) system_api: Base MAC address is not set
> [12:27:30]I (1368) system_api: read default base MAC address from EFUSE
> [12:27:30]W (1378) config: Waiting for config commit ...
> [12:27:31]I (2358) config: configuration has some uncommitted entries
> [12:27:31]I (2358) config: Committing configuration to nvs. Locking config object.
> [12:27:31]I (2358) config: Done Committing configuration to nvs.
> [12:27:31]I (2358) config: Config committed!
> [12:27:31]I (2368) esp_app_main: Configuring services
> [12:27:31]I (2368) services: Configuring I2C sda:-1 scl:-1 port:1 speed:400000
> [12:27:31]W (2378) services: no I2C configured
> [12:27:31]I (2378) services: Configuring SPI mosi:-1 miso:-1 clk:-1 host:1 dc:-1
> [12:27:31]W (2388) services: no SPI configured
> [12:27:31]W (2398) led: LED GPIO -1 ignored
> [12:27:31]W (2398) led: LED GPIO -1 ignored
> [12:27:31]I (2398) led: Configuring LEDs green:-1 (on:0 rmt:-1 -1% ), red:-1 (on:0 rmt:-1 -1% )
> [12:27:31]I (2408) battery: No battery
> [12:27:31]I (2418) monitor: Initializing monitoring
> [12:27:31]I (2418) monitor: Heap internal:282631 (min:282267) external:8367671 (min:8367651) dma:274875 (min:274511)
> [12:27:31]I (2428) esp_app_main: Initializing display
> [12:27:31]I (2438) display: Trying to configure display with N/A
> [12:27:31]W (2438) display: No display driver
> [12:27:31]I (2448) esp_app_main: Initializing led_vu
> [12:27:31]I (2448) led_vu: led_vu configuration invalid
[... 63 lines cut ...]
> [12:27:32]Press TAB when typing command name to auto-complete.
> [12:27:32]
> [12:27:32]
> [12:27:32]
> [12:27:32]
> [12:27:32]****************************
> [12:27:32]Your terminal application does not support escape sequences.
> [12:27:32]Line editing and history features are disabled.
> [12:27:32]On Windows, try using Putty instead.
> [12:27:32]****************************
> [12:27:45]recovery-squeezelite-esp32>I (16728) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<0,0>, prof:1
> [12:27:45]I (16728) wifi:station: a0:99:9b:11:7a:67 join, AID=1, bgn, 20
> [12:27:47]I (17838) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
> [12:27:48]W (19748) httpd_handlers: redirect_processor(1002): 192.168.4.2 requested invalid URL: [/hotspot-detect.html]
> [12:27:49]W (19768) httpd_handlers: process_redirect(957): Redirecting host [192.168.4.2] to http://192.168.4.1/ (from uri /hotspot-detect.html)
> [12:27:49]CORRUPT HEAP: Bad tail at 0x3d81cbef. Expected 0xbaad5678 got 0xbaad5600
> [12:27:49]
> [12:27:49]assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)
> [12:27:49]
> [12:27:49]
> [12:27:49]Backtrace: 0x403762be:0x3d80bbb0 0x40380631:0x3d80bbd0 0x40385875:0x3d80bbf0 0x40385376:0x3d80bd10 0x40376875:0x3d80bd30 0x403858a5:0x3d80bd50 0x4201eeb9:0x3d80bd70 0x4201f10b:0x3d80bdc0 0x4201f199:0x3d80be30 0x42057d3e:0x3d80be50 0x42056ea8:0x3d80be90 0x4205740a:0x3d80bf40 0x420561d0:0x3d80bf60 0x420b9b3a:0x3d80bf80 0x420562db:0x3d80bfa0
> [12:27:49]

Any ideas?
Thanks!

sle118, 2024-02-18:
Wondering why you changed the RTOS config?

KamilKobylarek, 2024-02-18:
Because I got:

> FAILED: esp-idf/tools/CMakeFiles/__idf_tools.dir/tools.c.obj 
ccache /opt/esp/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -DHAVE_CONFIG_H -DHIERARCHICAL_STATES=1 -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMODEL_NAME=SqueezeESP32 -DSTATE_MACHINE_LOGGER=1 -I/workspace/squeezelite-esp32/squeezelite-esp32/build/config -I/workspace/squeezelite-esp32/squeezelite-esp32/components/tools -I/opt/esp/idf/components/newlib/platform_include -I/opt/esp/idf/components/freertos/include -I/opt/esp/idf/components/freertos/include/esp_additions/freertos -I/opt/esp/idf/components/freertos/port/xtensa/include -I/opt/esp/idf/components/freertos/include/esp_additions -I/opt/esp/idf/components/esp_hw_support/include -I/opt/esp/idf/components/esp_hw_support/include/soc -I/opt/esp/idf/components/esp_hw_support/include/soc/esp32s3 -I/opt/esp/idf/components/esp_hw_support/port/esp32s3/. -I/opt/esp/idf/components/esp_hw_support/port/esp32s3/private_include -I/opt/esp/idf/components/heap/include -I/opt/esp/idf/components/log/include -I/opt/esp/idf/components/lwip/include/apps -I/opt/esp/idf/components/lwip/include/apps/sntp -I/opt/esp/idf/components/lwip/lwip/src/include -I/opt/esp/idf/components/lwip/port/esp32/include -I/opt/esp/idf/components/lwip/port/esp32/include/arch -I/opt/esp/idf/components/soc/include -I/opt/esp/idf/components/soc/esp32s3/. -I/opt/esp/idf/components/soc/esp32s3/include -I/opt/esp/idf/components/hal/esp32s3/include -I/opt/esp/idf/components/hal/include -I/opt/esp/idf/components/hal/platform_port/include -I/opt/esp/idf/components/esp_rom/include -I/opt/esp/idf/components/esp_rom/include/esp32s3 -I/opt/esp/idf/components/esp_rom/esp32s3 -I/opt/esp/idf/components/esp_common/include -I/opt/esp/idf/components/esp_system/include -I/opt/esp/idf/components/esp_system/port/soc -I/opt/esp/idf/components/esp_system/port/public_compat -I/opt/esp/idf/components/xtensa/include -I/opt/esp/idf/components/xtensa/esp32s3/include -I/opt/esp/idf/components/driver/include -I/opt/esp/idf/components/driver/esp32s3/include -I/opt/esp/idf/components/esp_pm/include -I/opt/esp/idf/components/esp_ringbuf/include -I/opt/esp/idf/components/efuse/include -I/opt/esp/idf/components/efuse/esp32s3/include -I/opt/esp/idf/components/vfs/include -I/opt/esp/idf/components/esp_wifi/include -I/opt/esp/idf/components/esp_event/include -I/opt/esp/idf/components/esp_netif/include -I/opt/esp/idf/components/esp_eth/include -I/opt/esp/idf/components/tcpip_adapter/include -I/opt/esp/idf/components/esp_phy/include -I/opt/esp/idf/components/esp_phy/esp32s3/include -I/opt/esp/idf/components/esp_ipc/include -I/opt/esp/idf/components/app_trace/include -I/opt/esp/idf/components/esp_timer/include -I/opt/esp/idf/components/pthread/include -I/opt/esp/idf/components/esp_http_client/include -I/opt/esp/idf/components/nghttp/port/include -I/opt/esp/idf/components/nghttp/nghttp2/lib/includes -I/opt/esp/idf/components/esp-tls -I/opt/esp/idf/components/esp-tls/esp-tls-crypto -I/opt/esp/idf/components/mbedtls/port/include -I/opt/esp/idf/components/mbedtls/mbedtls/include -I/opt/esp/idf/components/mbedtls/esp_crt_bundle/include -mlongcalls  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -fmacro-prefix-map=/workspace/squeezelite-esp32/squeezelite-esp32=. -fmacro-prefix-map=/opt/esp/idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4.6\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/tools/CMakeFiles/__idf_tools.dir/tools.c.obj -MF esp-idf/tools/CMakeFiles/__idf_tools.dir/tools.c.obj.d -o esp-idf/tools/CMakeFiles/__idf_tools.dir/tools.c.obj -c /workspace/squeezelite-esp32/squeezelite-esp32/components/tools/tools.c
/workspace/squeezelite-esp32/squeezelite-esp32/components/tools/tools.c:24:2: error: #error CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS must be at least 2
 #error CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS must be at least 2
  ^~~~~
[1025/1227] Building C object esp-idf/platform_config/CMakeFiles/__idf_platform_config.dir/platform_config.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

In deeper mening, no, I don't know why.

sle118, 2024-02-18:
ok. This port was done by @philippe44  so he can possibly help.

KamilKobylarek, 2024-02-18:
To be more specific the board is ESP32-S3-DevKitC-1-N16R8V also I build image again and I saw in logs: 

> [1306/1359] Building C object esp-idf/services/CMakeFiles/__idf_services.dir/accessors.c.obj
/workspace/squeezelite-esp32/squeezelite-esp32/components/services/accessors.c: In function 'get_psram_gpio_list':
/workspace/squeezelite-esp32/squeezelite-esp32/components/services/accessors.c:1163:9: note: #pragma message: need to add esp32-s3 specific SPIRAM GPIO config code
 #pragma message("need to add esp32-s3 specific SPIRAM GPIO config code")
         ^~~~~~~

Any clue?

philippe44, 2024-02-19:
Yes, you must change rtos config on all platforms per message. Why doesn't it work on the s3, I don't know. I had the same issue the first time with my s3 and I did something in the build to fix it but I can't remember why, sorry.

KamilKobylarek, 2024-03-23:
@wizmo2 thanks! I managed to successful flash esp32-s3 with current branch and set wifi creds but I have another error with DAC i2c:

`[22:01:48]i2cdetect 
[22:01:48]
[22:01:48]E (35138) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35138) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35148) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35148) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35158) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35158) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35168) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35178) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35188) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35198) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35198) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35208) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35208) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35218) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35228) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35228) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35248) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35248) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35258) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35258) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35268) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35278) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35278) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35288) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35298) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35308) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35308) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35318) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35328) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35328) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35338) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35338) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35348) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35358) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35368) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35378) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35378) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35388) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35388) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35398) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35408) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35418) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35428) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35428) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35438) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35438) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35448) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35458) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35458) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35468) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35478) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35488) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35488) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35498) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35508) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35508) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35518) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35518) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35538) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35548) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35548) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35558) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35558) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]I (35568) httpd_handlers: serving /status.json to peer 10.138.1.202 port 25794
[22:01:49]E (35568) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[22:01:49]E (35578) i2c: i2c_master_cmd_begin(1426): i2c driver not installed
[... 72 lines cut ...]
[22:01:49]60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
[22:01:49]70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
[22:01:49]`

[... cut ...]
[21:57:05]I (128118) I2S: DMA queue destroyed
[21:57:05]E (128118) i2c: i2c_driver_delete(411): i2c driver install error
[21:57:05]W (128118) messaging: cfg-audio-tmpl
[21:57:05]squeezelite exited with error code 0
[21:57:05]
[21:57:05]W (128128) messaging: cfg-audio-tmpl
[21:57:05]squeezelite exited with error code 0
[21:57:05]
[21:57:05]E (128138) messaging: cfg-audio-tmpl
[21:57:05]Rebooting in 60 sec
[21:57:05]
[21:57:05]E (128148) messaging: cfg-audio-tmpl
[21:57:05]Rebooting in 60 sec
[21:57:05]
`

wizmo2, 2024-03-26:
Looks like a NVS configuration issue.  Check you i2c settings for the DAC and the auxiliaries.   If you are using my build, it could also be the ADC configuration (plus airplay is disabled).

KamilKobylarek, 2024-03-26:
I used the master-v4.3 branch, sdkconfig from 4.4-s3 and some parameters from your cfg. 
`#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32s3"
CONFIG_IDF_TARGET_ESP32S3=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0009

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x0
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_REGION_PROTECTION_ENABLE=y
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
CONFIG_BOOTLOADER_FLASH_XMC_SUPPORT=y
# end of Bootloader config

#
[... 1574 lines cut ...]
CONFIG_UDP_RECVMBOX_SIZE=32
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set 
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1=y
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_USB_ENABLED is not set
# CONFIG_SUPPORT_TERMIOS is not set
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options
`

My gole is to run on S3 with PCM5012 and Airplay.

arpena, 2026-09-15:
I can confirm and reproduce this exact crash on a fresh ESP32-S3-N16R8 build (16MB Quad flash, 8MB Octal PSRAM, DIO flash mode), IDF v4.4.7, `master-v4.3` branch, built from a from-scratch `idf.py set-target esp32s3` + patched `I2S-S3-sdkconfig`. Identical trigger, identical crash signature, on two separate client devices (iPhone probing `/hotspot-detect.html`, and an Android device probing `/gen_204`):

```
CORRUPT HEAP: Bad tail at 0x3d81e08f. Expected 0xbaad5678 got 0xbaad5600

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)
```

Both times it's triggered by the exact same log sequence right before the crash:
```
W (...) httpd_handlers: redirect_processor(1002): <peer> requested invalid URL: [<captive-portal-probe-uri>]
W (...) httpd_handlers: process_redirect(957): Redirecting host [<peer>] to http://192.168.4.1/ (from uri <uri>)
CORRUPT HEAP: Bad tail ...
```

## Root cause

In `components/wifi-manager/http_server_handlers.c`, `process_redirect()`, the `"302 Found"` branch:

```c
size_t buf_size = strlen(redirect_payload1) + strlen(redirect_payload2) + strlen(redirect_payload3) + 2*(strlen(location_prefix)+strlen(ap_ip_address))+1;
char * redirect = malloc_init_external(buf_size);

if(strcasestr(status,"302")){
    size_t url_buf_size = strlen(location_prefix) + strlen(ap_ip_address)+1;
    redirect_url = malloc_init_external(url_buf_size);
    memset(redirect_url,0x00,url_buf_size);
    snprintf(redirect_url, buf_size, "%s%s/", location_prefix, ap_ip_address);
    ...
```

`redirect_url` is allocated with `url_buf_size = strlen(location_prefix) + strlen(ap_ip_address) + 1`, which reserves space for `"http://" + <ip>` plus a NUL terminator — but does **not** reserve a byte for the trailing `"/"` that the format string `"%s%s/"` appends. That's a 1-byte-short allocation for the string actually written into it.

Compounding this, the `snprintf` call passes **`buf_size`** (the size of the *other*, larger `redirect` buffer) as the size limit instead of `url_buf_size` (the actual size of `redirect_url`). Since `buf_size` is always larger, `snprintf` has no way to know the real buffer is smaller, and it overflows `redirect_url` by one byte — exactly matching the `CORRUPT HEAP: Bad tail` signature (heap-poisoning's guard-byte check for a small overflow just past an allocation).

## Suggested fix

```c
size_t url_buf_size = strlen(location_prefix) + strlen(ap_ip_address) + 1 /* '/' */ + 1 /* NUL */;
redirect_url = malloc_init_external(url_buf_size);
memset(redirect_url, 0x00, url_buf_size);
snprintf(redirect_url, url_buf_size, "%s%s/", location_prefix, ap_ip_address);
```

i.e. (1) add a byte for the trailing slash to the allocation, and (2) pass `url_buf_size`, not `buf_size`, as the `snprintf` limit.

## Practical workaround for anyone hitting this

The crash only fires while the device is in AP/recovery configuration mode, when a client's OS auto-probes the captive-portal detection URL (`/hotspot-detect.html` on iOS/macOS, `/gen_204` on Android/Chrome). Get through the WiFi-join step on the setup page as fast as possible after connecting to the AP, before your device's background connectivity check fires — once it joins your home WiFi and the AP shuts down, this code path is no longer reachable during normal operation.