sle118/squeezelite-esp32 #399: Attempt to rewind song through IOS (17.3.1) spotify home screen widget restults in constant hum/noise (while listening through cspot)

bungh0l10, 2024-02-21:
Hi there,

just found that rewinding songs through IOS Spotify widget restult s in very loud noise - sounds like reaching close to a channel while seeking on an analog TV. Double tapping back rewinds to the previous track in best case but I also achieved it while pressing back and forth that the squeezeamp quits the spotify connection with a super loud crack through the speakers (unfortunately I have no logs for that scenario yet).

### Preliminary Information

1. **1670**
2. **0.600**

### Hardware Details

- **Squeezeamp v3.20**

### NVS Settings

{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "Squeezelite",
  "a2dp_sink_name": "SMSL BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "ESP32-AirPlay-ceded8",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "squeezelite-ceded8",
  "autoexec": "1",
  "autoexec1": "squeezelite -o I2S -b 500:2000 -d all=info -C 30 -W -n \"squeezelite-ceded8\"",
  "bat_config": "",
  "bt_name": "ESP32-BT-ceded8",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\"deviceName\":\"squeezelite-ceded8\",\"bitrate\":320,\"zeroConf\":1,\"volume\":32767}",
  "dac_config": "",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "",
  "enable_airplay": "Y",
  "enable_bt_sink": "Y",
  "enable_cspot": "Y",
  "equalizer": "0,0,0,0,0,0,0,0,0,0",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "squeezelite-ceded8",
  "i2c_config": "",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "",
  "lms_ctrls_raw": "n",
  "loudness": "0",
  "metadata_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
  "pollmx": "600",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "https://github.com/sle118/squeezelite-esp32/releases",
  "rotary_config": "",
  "set_GPIO": "0=ir",
  "sleep_config": "",
  "spdif_config": "",
  "spi_config": "",
  "stats": "N",
  "target": "",
  "telnet_block": "500",
  "telnet_buffer": "40000",
  "telnet_enable": "Y"
}

<details>
<pre><code>
  [02:23:47.393] I TrackPlayer.cpp:171: Got track ID=8b7eedf2eb445b768d0997f518c039451153784f
[02:23:47.396] I CDNAudioFile.cpp:43: Opening HTTP stream to https://audio-ak-spotify-com.akamaized.net/audio/8b7eedf2eb445b768d0997f518c039451153784f?__token__=exp=1708626743~hmac=1d25044d94499af1e0a18501365fefe23c33f17145733b7a337121b66d23a964
[02:23:48.578] I CDNAudioFile.cpp:70: Header and footer bytes received
[02:23:48.582] D MercurySession.cpp:252: Executing Mercury Request, type SEND
[02:23:48.594] D MercurySession.cpp:252: Executing Mercury Request, type SEND
[02:23:48.603] cspot_cmd_handler:383 CSpot start track
[02:23:48.604] I Shim.cpp:217: (re)start playing at 0
[02:23:48.606] cspot_cmd_handler:396 CSpot play
[02:23:48.660] output_thread_i2s:518 Output state is 2
[02:23:48.665] output_thread_i2s:585 Restarting I2S.
[02:23:48.696] I MercurySession.cpp:42: Received packet, command: 178
[02:23:48.698] D MercurySession.cpp:174: Received mercury packet
[02:23:48.925] I TrackPlayer.cpp:206: Playing
[02:23:48.933] I MercurySession.cpp:42: Received packet, command: 178
[02:23:48.935] D MercurySession.cpp:174: Received mercury packet
[02:23:48.973] I Shim.cpp:120: new track started <> => <8b7eedf2eb445b768d0997f518c039451153784f>
[02:23:48.975] I Shim.cpp:262: next track will play in 0 ms
[02:23:48.990] _output_frames:150 track start sample rate: 44100 replay_gain: 0
[02:23:49.138] I Shim.cpp:416: next track's audio has reached DAC (offset 0)
[02:23:49.147] D MercurySession.cpp:252: Executing Mercury Request, type SEND
[02:23:49.170] D MercurySession.cpp:252: Executing Mercury Request, type SEND
[02:23:49.303] I MercurySession.cpp:42: Received packet, command: 178
[02:23:49.305] D MercurySession.cpp:174: Received mercury packet
[02:23:49.422] I MercurySession.cpp:42: Received packet, command: 178
[02:23:49.426] D MercurySession.cpp:174: Received mercury packet
I (302715) battery: Voltage 8.73V
[02:25:28.367] I MercurySession.cpp:42: Received packet, command: 4
[02:25:28.369] D TimeProvider.cpp:15: Time synced with spotify servers
[02:25:28.483] I MercurySession.cpp:42: Received packet, command: 74
[02:27:28.428] I MercurySession.cpp:42: Received packet, command: 4
[02:27:28.430] D TimeProvider.cpp:15: Time synced with spotify servers
[02:27:28.489] I MercurySession.cpp:42: Received packet, command: 74
[02:28:42.378] I MercurySession.cpp:42: Received packet, command: 181
[02:28:42.381] D SpircHandler.cpp:69: Received subscription response
[02:28:42.387] I TrackPlayer.cpp:107: Seeking...
[02:28:42.391] D MercurySession.cpp:252: Executing Mercury Request, type SEND
[02:28:42.402] cspot_cmd_handler:401 CSpot seek by 0
[02:28:42.539] I MercurySession.cpp:42: Received packet, command: 178
[02:28:42.544] D MercurySession.cpp:174: Received mercury packet
I (602715) battery: Voltage 8.79V
[02:29:28.411] I MercurySession.cpp:42: Received packet, command: 4
[02:29:28.413] D TimeProvider.cpp:15: Time synced with spotify servers
[02:29:28.454] I MercurySession.cpp:42: Received packet, command: 74
[02:30:11.984] I TrackPlayer.cpp:224: EOF
[02:30:11.989] I TrackPlayer.cpp:255: Playing done
[02:30:11.995] I Shim.cpp:247: playlist ended, no track left to play
[02:30:23.753] I Shim.cpp:430: last track finished
[02:30:23.756] D MercurySession.cpp:252: Executing Mercury Request, type SEND
[02:30:23.769] I TrackPlayer.cpp:98: Resetting state
[02:30:23.959] I MercurySession.cpp:42: Received packet, command: 178
[02:30:23.961] D MercurySession.cpp:174: Received mercury packet
[02:30:24.042] I TrackPlayer.cpp:171: Got track ID=8b7eedf2eb445b768d0997f518c039451153784f
[02:30:24.046] I CDNAudioFile.cpp:43: Opening HTTP stream to https://audio-ak-spotify-com.akamaized.net/audio/8b7eedf2eb445b768d0997f518c039451153784f?__token__=exp=1708626743~hmac=1d25044d94499af1e0a18501365fefe23c33f17145733b7a337121b66d23a964
[02:30:25.198] I CDNAudioFile.cpp:70: Header and footer bytes received
[02:30:25.202] D MercurySession.cpp:252: Executing Mercury Request, type SEND
[02:30:25.211] D MercurySession.cpp:252: Executing Mercury Request, type SEND
[02:30:25.219] cspot_cmd_handler:383 CSpot start track
[02:30:25.220] I Shim.cpp:217: (re)start playing at 0
[02:30:25.221] cspot_cmd_handler:412 CSpot pause
[02:30:25.239] output_thread_i2s:518 Output state is 0
[02:30:25.410] I MercurySession.cpp:42: Received packet, command: 178
[02:30:25.412] D MercurySession.cpp:174: Received mercury packet
[02:30:25.587] I TrackPlayer.cpp:206: Playing
[02:30:25.634] I Shim.cpp:120: new track started <> => <8b7eedf2eb445b768d0997f518c039451153784f>
[02:30:25.636] I Shim.cpp:262: next track will play in 0 ms
[02:30:25.743] I MercurySession.cpp:42: Received packet, command: 178
[02:30:25.745] D MercurySession.cpp:174: Received mercury packet
[02:30:55.239] output_thread_i2s:518 Output state is -1
[02:30:55.239] output_thread_i2s:522 switching off amp GPIO -1
</code></pre>
</details>

sle118, 2024-02-22:
@feelfreelinux Filip might be able to help here, unless @philippe44 wants to comment

philippe44, 2024-02-22:
I happens from time to time and it's due to a byte/word misalignment when seeking and it's rather a cspot thing as it happens on all my other apps that use cspot . I looked at it a while ago but could not see anything obvious and I don't have time to investigate that currently.

sle118, 2024-02-24:
Thank you @philippe44 

Summoning the spirit of cspot here 🙂

feelfreelinux, 2024-03-09:
Thanks for reporting this, will try to investigate it in a bit. I think I was able to reproduce that before in upstream cspot too.

feelfreelinux, 2024-03-09:
Hm, not quite able to recreate it on the cli / desktop sink of cspot. Will have to retry on esp32 implementation.

philippe44, 2024-03-09:
I have it on my other cspot apps as well, the upnp and raop bridges.

philippe44, 2026-10-09:
Should be fixed

closed 2026-10-09 by philippe44