sle118/squeezelite-esp32 #400: Feature Request: I2S initialisation sequence for some DACs

alidaf, 2024-02-22:
I'd like to be able to initialise a DAC through an I2S sequence. This may be possible via an alternative dac_controlset feature in the NVS editor.

After trying to research an ES9018K2M [board](https://www.aliexpress.com/item/1005003740290057.html?spm=a2g0o.order_list.order_list_main.42.68aa1802pYWTAs) a discussion thread was [started](https://github.com/sle118/squeezelite-esp32/discussions/395#discussioncomment-8546958) that led to a comment with the possible solution involving toggling the serial and LR clocks in a specific sequence.
I did manage to get this DAC working with Moode on a RPi with no I2C access so initialisation via I2S seems plausible and implementing a generic, configurable I2S sequence via the NVS editor may extend it to other DACs that have I2S initialisation sequences.

ElFishi, 2024-02-22:
Darren, can you please try and hook up your ESP to the ES9018K2M via I2S: bck = BCK, ws = LRCK, do = DATA, flash the [latest I2S 32bit firmware](https://github.com/sle118/squeezelite-esp32/releases/tag/I2S-4MFlash.32.1670.master-v4.3) and see what happens?

sle118, 2024-02-22:
@alidaf since we're in stabilizing mode for the current release, we aren't planning on releasing new features at this point.

I will eventually get to the point where I can share the new platform where new features should be easier to implement

alidaf, 2024-02-23:
> Darren, can you please try and hook up your ESP to the ES9018K2M via I2S: bck = BCK, ws = LRCK, do = DATA, flash the [latest I2S 32bit firmware](https://github.com/sle118/squeezelite-esp32/releases/tag/I2S-4MFlash.32.1670.master-v4.3) and see what happens?

@ElFishi I have the 32bit version. I had another go, rearranged some of the power supply wires but everything still seemed dead. I hooked it up to a spectrum analyser I've been working on and got a brief peak. I started twiddling and got a longer peak so I hooked it up to the amp. Music. Sweet and beautiful but intermittent music. I have not just one, but a number of loose joints on the expansion board I've been using. Almost all of them on one side. I'm just wiring a new one up but am so confident that it will work that I'm writing this now to thank you from the bottom of my blackened heart for sticking with me. Fingers crossed, I'm going in.

I'm hoping to switch from the Dev C board coupled with a GPIO expansion board I've been using to one of [these](https://www.aliexpress.com/item/32977375539.html?spm=a2g0o.home.0.0.650c76dbSgxyin&mp=1) and solder everything. It'll reduce the component count and cost, and make things like this a little less likely. I can't mount the Dev C boards, hence the expansion board. Thanks again.

closed 2024-02-23 by alidaf

ElFishi, 2024-02-23:
Darren, I am glad you got it to work. 
Welcome to the never ending cycles of "improvements" : )