sle118/squeezelite-esp32 #401: SSD1322 driver issue

kuzmanzs, 2024-03-05:
Hi,
I am using some more cheap SSD1322 lcd. I think it is simple driver issue.
I got display from here:  
https://www.ebay.com/itm/134658751869

display settings:
`SPI:width=256,height=64,cs=4,speed=8000000,driver=SSD1322,VFlip`

Does somebofy have similar issue?
[image]

philippe44, 2024-03-07:
Seem that the display is more than 256

kuzmanzs, 2024-03-07:
yes it is strange display..

With width=284 right side of right VU meter is missing.

[image]

philippe44, 2024-03-07:
Is this the only issue then?

kuzmanzs, 2024-03-12:
Big thanks for the squeezelite!

Issue has been resolved in another lib.
https://github.com/olikraus/u8g2/issues/2386

I have checked you are using different implementation.
Could you do changes in your lib as well?

Need different x_offset, initialisation and do not need A enhancement.

philippe44, 2024-03-12:
So it's still a 256x32, not a 284x32 display?

kuzmanzs, 2024-03-13:
It has 256x32 resolution