sle118/squeezelite-esp32 #408: The ES9018 is on another level for DACs and there's plenty of documentation for the chip itself, but the problem is when certain 3rd parties release a board that is almost certainly an attempt at a knock-off of the Audiophonics boards for the RPi, which are woefully expensive. There is scant information other than a few screen shots of a configuration menu with the HifiBerry DAC selected. The are a couple of badly translated FAQs, one of which covers the MCLK, which isn't needed. I took a punt but fell short on this one.

Abonaif123, 2024-04-03:
The ES9018 is on another level for DACs and there's plenty of documentation for the chip itself, but the problem is when certain 3rd parties release a board that is almost certainly an attempt at a knock-off of the Audiophonics boards for the RPi, which are woefully expensive. There is scant information other than a few screen shots of a configuration menu with the HifiBerry DAC selected. The are a couple of badly translated FAQs, one of which covers the MCLK, which isn't needed. I took a punt but fell short on this one.

_Originally posted by @alidaf in https://github.com/sle118/squeezelite-esp32/discussions/395#discussioncomment-8439719_

sle118, 2024-04-09:
What is the intent of this issue? Not sure I see the point

alidaf, 2024-04-09:
I really don't know since I got it working with some help in the original post and it had nothing to do with the code. This is just a suspicious copy-paste so I'd suggest closing it.