sle118/squeezelite-esp32 #411: ESP32-S3 Problem: No Autoconnect to the LMS after Startup

jochen-rath, 2024-04-09:
Hello
I try to run squeezelite-esp32 on a ESP32-S3-WROOM-1 board.

### Describe the bug
After booting the ESP32-S3, the board can not connect to my LMS-Server. On "idf.py monitor" it gives me the message:
W (2938) config: Value not found for key autoexec
When I connect to the ESP32-S3 with the "idf.py monitor", I can manually run the command "squeezelite -s Servername -n ESP" and connect my ESP32-S3 device to my LMS-Server. Than I can play music over a connected IS2 DAC.

### Preliminary Information

1. **Firmware Version**: I have downloaded the actual (2024-04-09) version of "sle118 /squeezelite-esp32"
2. **sdkconfig**: I have copied the "build-scripts/I2S-4MFlash-sdkconfig.defaults " to sdkconfig.defaults. I changed the chip to esp32-s3
3. **esp-idf**: v4.4.7 
4. **menuconfig**: I disabled bluetooth and activated SPI Ram

### Hardware Details

Please describe your hardware setup:

- **ESP32 Module**:ESP32-S3-WROOM-1
- **I2S DAC**: CJMCU-1334

### NVS Settings

<details>
<pre><code>
{
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "ESP32-AirPlay-c720f4",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "squeezelite-c720f4",
  "autoexec1": "squeezelite  -o i2s  -n  esp -s  192.168.0.140 -b  500:2000 -C  30 -d  all=info ",
  "bat_config": "",
  "bypass_wm": "0",
  "cspot_config": "{\"deviceName\":\"squeezelite-c720f4\",\"bitrate\":160}",
  "dac_config": "model=I2S,bck=12,ws=14,do=13",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "",
  "enable_cspot": "1",
  "equalizer": "",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "squeezelite-c720f4",
  "i2c_config": "",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "",
  "lms_ctrls_raw": "n",
  "loudness": "0",
  "metadata_config": "",
  "model_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
  "pollmx": "600",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "CONFIG_SQUEEZELITE_ESP32_RELEASE_URL",
  "rotary_config": "",
  "set_GPIO": "",
  "sleep_config": "",
  "spdif_config": "",
  "spi_config": "",
[... 1919 lines cut ...]
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1=y
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_USB_ENABLED is not set
# CONFIG_SUPPORT_TERMIOS is not set
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options
</code></pre>
</details>

### Issue Description

1. **Observed Behavior**: Some key in the autoexec is missing or wrong.
2. **Expected Behavior**: The squeezeclient automatically connects with the LMS server.
3. **Steps to Reproduce**: See Preliminary Information. I tried different esp-idf version (4.4.4 and 4.4.7) and different/older squeezelite-esp32 versions.

anabolyc, 2024-05-05:
Hey folks, I'm working on the S3 version of the [Louder ESP](https://www.tindie.com/products/sonocotta/louder-esp32/) board, and very much look forward to having squeezelite-esp32 support for it. I tried to do the same as @jochen-rath and apart from the absence of Bluetooth found a few more issues. Long story short, it doesn't work currently. 

S3 has a more usable PSRAM, that supports a faster interface. It has a built-in debugger right via the USB port, which is a fantastic option for developers.

I'm willing to send a few board samples to the developers who are ready to look into the S3-related issues and perhaps add official support. The board itself uses a TAS5805M DAC that is already supported and works off USB (with a 5W output power limit) or an external power source (driving 2 channels 20W each)

sle118, 2024-05-06:
The missing autoexec key is fine. It's just standard behavior where during boot, the system checks for autoexec=1 to see if it should execute the start sequence. Then from there it will execute in sequence autoexec1, autoexec2, etc. The warning here is merely an indication that there is only a single autoexec to run (squeezelite).

As for not connecting to the media server, your logs seem to be showing that an attempt is made 
```
[00:04:24.067] notify:228 notified server 192.168.0.140 hport 9000 cport 9090
I (29968) network_status: LMS IP: 192.168.0.140, hport: 9000, cport: 9090
[00:04:24.187] notify:228 notified server 192.168.0.140 hport 9000 cport 9090
I (30088) network_status: LMS IP: 192.168.0.140, hport: 9000, cport: 9090
```
Can you confirm that the issue isn't on the server side?  for example a firewall issue, etc.

jochen-rath, 2024-05-06:
In this moment, I run the squeezelite command manuelly via idf.py monitor.

<pre><code>

squeezelite-esp32> I (29888) squeezelite_cmd: Calling squeezelite
</code></pre>

Your mentioned log occurs after the manuelly running of the squeezelite command, when esp32-s3 connects to the lms. Only after this connection is established, I can play music. But this connection doesn't occure automatically on start up.

<details>
<pre><code>
[00:04:24.067] notify:228 notified server 192.168.0.140 hport 9000 cport 9090
I (29968) network_status: LMS IP: 192.168.0.140, hport: 9000, cport: 9090
[00:04:24.187] notify:228 notified server 192.168.0.140 hport 9000 cport 9090
I (30088) network_status: LMS IP: 192.168.0.140, hport: 9000, cport: 9090

=== END OF LOG ===   
</code></pre>
</details>

Because the esp32-s3 can connect to the LMS, I don't think, it is a firewall problem.

sle118, 2024-05-09:
I have ordered an esp32-s3 so I can test myself

svxa, 2024-05-10:
I found a workaround.
You can add

`"autoexec": "1",`

to the NVS.

So I have 2 autoexec keys in my NVS:
```json
{
  "autoexec": "1",
  "autoexec1": "squeezelite  -o i2s  -s  x.x.x.x -b  500:2000 -C  30 -d  all=info  -n \"speaker-name\"",
}
```
This fixed it for me.

jochen-rath, 2024-05-10:
Wow, thank you very much. This solved the issue. No it works on my ESP32-S3

closed 2024-05-10 by jochen-rath