sle118/squeezelite-esp32 #412: ESP_ERR_NVS_KEY_TOO_LONG, when trying to enter wifi ssid and password

DamnDaniel-98, 2024-04-14:
Whenever I try to enter my Wifi data via the web interface, the EPS first connects to the Wifi. However, the message: "_[PlaceholderForMySSID]_ (error) ESP_ERR_NVS_KEY_TOO_LONG" appears in the LOGs (_[PlaceholderForMySSID]_ is a placeholder). If I then restart the ESP, it does not connect to the Wifi and starts its default AP.

The firmware was installed via the online installer under "[Squeezelite-ESP32 Installer (sle118.github.io)](https://sle118.github.io/squeezelite-esp32-installer/)" (Generic/I2S). The board is a ESP32-A1S.

sle118, 2024-04-14:
Please follow the bug template. 
 https://github.com/sle118/squeezelite-esp32/issues/new?assignees=&labels=&projects=&template=bug_report.md&title=

sle118, 2024-04-28:
Peugeot a duplicate old #410

closed 2024-05-15 by sle118