sle118/squeezelite-esp32 #414: Bootloop when used with Home Assistant

anabolyc, 2024-04-25:
### Describe the bug

Home Assistant / Music Assistant has a built-in LMS server that allows using slimproto clients to play the audio directly from HA. Seems that some communication that comes from the built-in server crashes squeezelite-esp32. 

When enabled squeezelite-esp32 immediately goes into the bootloop. As soon as disabled, it acts normally.

### Preliminary Information

1. fw: I2S-4MFlash-1672, mode: Squeezelite-ESP32
2. [slimproto](https://music-assistant.io/player-support/slimproto/) built-in into version 2024.3.6 

### Hardware Details

- ESP32 WROVER 16N8 as a part of [Esparagus](https://github.com/sonocotta/esparagus-media-center) board
- Same behavior with PCM5102 and MAX98357 DAC 
- IR, RGB led, doesn't seem to have any effect on this issue

### NVS Settings

[nvs_config_loud-esparagus_1714076880283json.txt](https://github.com/sle118/squeezelite-esp32/files/15120394/nvs_config_loud-esparagus_1714076880283json.txt)

<details>
<pre><code>
{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "loud-esparagus",
  "a2dp_sink_name": "SMSL BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "loud-esparagus",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "loud-esparagus",
  "autoexec": "1",
  "autoexec1": "squeezelite -o I2S -b 500:2000 -d all=info -C 30 -W -n loud-esparagus",
  "bat_config": "",
  "bt_name": "loud-esparagus",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\"deviceName\":\"loud-esparagus\",\"bitrate\":160,\"zeroConf\":1}",
  "dac_config": "model=I2S,bck=26,ws=25,do=22",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "",
  "enable_airplay": "Y",
  "enable_bt_sink": "Y",
  "enable_cspot": "Y",
  "equalizer": "",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "loud-esparagus",
  "i2c_config": "",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "type=[WS2812],length=1,gpio=33",
  "lms_ctrls_raw": "n",
  "loudness": "0",
  "metadata_config": "",
  "model_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
[... 263 lines cut ...]
PC      : 0x4026ec2f  PS      : 0x00060f30  A0      : 0x80102b8f  A1      : 0x3f801a20  
A2      : 0x00000000  A3      : 0x3ffbb1fc  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80098de7  A9      : 0x3f8019e0  
A10     : 0x3ffc4388  A11     : 0x3ffbf5a8  A12     : 0x3ffc438c  A13     : 0x00060f20  
A14     : 0x000000fe  A15     : 0x00000001  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000080  LBEG    : 0x40093da8  LEND    : 0x40093dc4  LCOUNT  : 0xffffffff  

Backtrace:0x4026ec2c:0x3f801a20 0x40102b8c:0x3f801a40 0x400fb3b9:0x3f801b20 0x400fbe1d:0x3f801b40 0x400ff57a:0x3f801b60 0x400fa0cb:0x3f801c40 0x400de558:0x3f801f70

ELF file SHA256: 0000000000000000

Rebooting...
</code></pre>
</details>

### Issue Description

1. **Observed Behavior**: Presense of the HA slimproto server in the network cause device to bootloop
2. **Expected Behavior**: Server should be discovered and paired normally
3. **Steps to Reproduce**: 
3.1. Make sure squeezelite-esp32 device boots and start LMS server discovery
3.2. Enable slimproto player provider in the HA / MA
3.3. Observe squeezelite-esp32 crash and bootloop within seconds.

If you guide me to investigate and possibly fix this issue, I would be happy to do it myself. From the looks of it, seems that message is malformed or at least does not look as expected. I tried to enable more logging via telnet, but this yield no more information that standard logs have

sle118, 2024-04-26:
for the record, the stack trace decodes as follow:
```
0x4026ec2c: GDS_DisplayOn at ../components/display/core/gds.c line 318
0x40102b8c: handler at ../components/squeezelite/displayer.c line 749
0x400fb3b9: handler at ../components/squeezelite/output_embedded.c line 70
0x400fbe1d: handler at ../components/squeezelite/output_i2s.c line 167
0x400ff57a: slimproto at ../components/squeezelite/slimproto.c line 809
0x400fa0cb: squeezelite_main at ../components/squeezelite/main.c line 820
0x400de558: squeezelite_thread at ../components/platform_console/app_squeezelite/cmd_squeezelite.c line 79
```
it seems like Music Assistant is sending some display commands even if there is no display. 

Could you try to disable the led vu and see if the issue is still happening?

@philippe44 one way to prevend this would be to ensure that we have a valid display pointer in the GDS code.

HA-TB303, 2024-04-27:
I have the same issue and I am not using Music Assistant.

sle118, 2024-04-27:
@HA-TB303 do you have the led_vu enabled?

anabolyc, 2024-04-27:
> Could you try to disable the led vu and see if the issue is still happening?

I can confirm that disabling led vu fix the issue. 

<details>
<pre><code>
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6424
load:0x40078000,len:15632
load:0x40080400,len:4348
entry 0x40080688
I (27) boot: ESP-IDF v4.3.5-dirty 2nd stage bootloader
I (27) boot: compile time 03:05:16
I (27) boot: chip revision: v3.1
I (31) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32: SPI Speed      : 80MHz
I (41) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 recovery         factory app      00 00 00010000 00140000
I (96) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (104) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (111) boot: End of partition table
I (115) esp_image: segment 0: paddr=00150020 vaddr=3f400020 size=aeaf0h (715504) map
I (321) esp_image: segment 1: paddr=001feb18 vaddr=3ffbdb60 size=01500h (  5376) load
I (323) esp_image: segment 2: paddr=00200020 vaddr=400d0020 size=1b2e5ch (1781340) map
I (818) esp_image: segment 3: paddr=003b2e84 vaddr=3ffbf060 size=0453ch ( 17724) load
I (823) esp_image: segment 4: paddr=003b73c8 vaddr=40080000 size=1e014h (122900) load
I (863) esp_image: segment 5: paddr=003d53e4 vaddr=400c0000 size=00068h (   104) load
I (878) boot: Loaded app from partition at offset 0x150000
I (879) boot: Disabling RNG early entropy source...
I (890) psram: This chip is ESP32-D0WD
I (890) spiram: Found 64MBit SPI RAM device
I (890) spiram: SPI RAM mode: flash 80m sram 80m
I (893) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (901) cpu_start: Pro cpu up.
I (904) cpu_start: Starting app cpu, entry point is 0x400813fc
I (0) cpu_start: App cpu up.
I (1410) spiram: SPI SRAM memory test OK
I (1423) cpu_start: Pro cpu start user code
I (1423) cpu_start: cpu freq: 240000000
I (1423) cpu_start: Application information:
I (1426) cpu_start: Project name:     Squeezelite-ESP32
I (1432) cpu_start: App version:      I2S-4MFlash-1672
I (1437) cpu_start: Compile time:     Mar 22 2024 03:06:48
I (1444) cpu_start: ELF file SHA256:  0000000000000000...
I (1450) cpu_start: ESP-IDF:          v4.3.5-dirty
I (1455) cpu_start: Min chip rev:     v1.0
I (1460) cpu_start: Max chip rev:     v3.99 
I (1465) cpu_start: Chip rev:         v3.1
I (1470) heap_init: Initializing. RAM available for dynamic allocation:
I (1477) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1483) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1489) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1495) heap_init: At 3FFC6EC8 len 00019138 (100 KiB): DRAM
I (1502) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1508) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1515) heap_init: At 4009E014 len 00001FEC (7 KiB): IRAM
I (1521) spiram: Adding pool of 4018K of external SPI memory to heap allocator
I (1529) spi_flash: detected chip: gd
I (1533) spi_flash: flash io: qio
W (1537) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
[... 154 lines cut ...]
[00:00:06.984] sendHELO:155 cap: Model=squeezeesp32,AccuratePlayPoints=1,HasDigitalOut=1,HasPolarityInversion=1,Balance=1,Depth=16,Firmware=v1.0-1672-16,ModelName=SqueezeESP32,MaxSampleRate=192000,alc,aac,ogg,ops,ogf,flc,wav,aif,pcm,mp3
[00:00:07.026] notify:228 notified server 192.168.1.48 hport 9000 cport 9090
I (8451) network_status: LMS IP: 192.168.1.48, hport: 9000, cport: 9090
[00:00:07.040] process:578 unhandled vers
[00:00:07.043] process:578 unhandled grfb
[00:00:07.046] process:578 unhandled grfe
[00:00:07.048] process:578 unhandled grfe
[00:00:07.061] process:578 unhandled grfe
[00:00:12.138] process:578 unhandled grfe
[00:00:17.360] process:578 unhandled grfe
[00:00:22.275] process:578 unhandled grfe
I (25281) httpd_handlers: serving /status.json to peer 192.168.1.16 port 64703
[00:00:27.095] process:578 unhandled grfe
[00:00:32.108] process:578 unhandled grfe
[00:00:37.122] process:578 unhandled grfe
[00:00:42.139] process:578 unhandled grfe
[00:00:47.158] process:578 unhandled grfe
[00:00:52.177] process:578 unhandled grfe
[00:00:57.099] process:578 unhandled grfe
[00:01:02.219] process:578 unhandled grfe
[00:01:07.109] process:578 unhandled grfe
[00:01:12.245] process:578 unhandled grfe
[00:01:17.161] process:578 unhandled grfe
</code></pre>
</details>

HA-TB303, 2024-04-27:
Yes indeed, disabling led vu fixed the issue for me as well.

wizmo2, 2024-04-28:
I'll have a look to see if I can see what's going on with led_vu

sle118, 2024-04-28:
Could be that led vu causes the system to report display support to the server. There may not be a choice other than adding some null checking in the GDS itself.

sle118, 2024-05-07:
I just pushed the following commit https://github.com/sle118/squeezelite-esp32/commit/21407e8c1c4decfc0da29812af590d98aa55d5be which I am hoping will resolve the issue. You can try installing the latest build when it will show up.

sle118, 2024-05-08:
Closing this now. Please reopen if needed

anabolyc, 2024-05-09:
Hey @sle118 

Sorry, just get to test the latest build. Unfortunately, it is still crashing with a different error. 

<details><pre><code>
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6424
load:0x40078000,len:15632
load:0x40080400,len:4348
entry 0x40080688
I (27) boot: ESP-IDF v4.3.5-dirty 2nd stage bootloader
I (27) boot: compile time 21:43:45
I (27) boot: chip revision: v3.1
I (31) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32: SPI Speed      : 80MHz
I (41) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  3 recovery         factory app      00 00 00010000 00140000
I (96) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (103) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (111) boot: End of partition table
I (115) esp_image: segment 0: paddr=00150020 vaddr=3f400020 size=aeaf0h (715504) map
I (321) esp_image: segment 1: paddr=001feb18 vaddr=3ffbdb60 size=01500h (  5376) load
I (323) esp_image: segment 2: paddr=00200020 vaddr=400d0020 size=1b2e84h (1781380) map
I (817) esp_image: segment 3: paddr=003b2eac vaddr=3ffbf060 size=0453ch ( 17724) load
I (823) esp_image: segment 4: paddr=003b73f0 vaddr=40080000 size=1e014h (122900) load
I (863) esp_image: segment 5: paddr=003d540c vaddr=400c0000 size=00068h (   104) load
I (878) boot: Loaded app from partition at offset 0x150000
I (878) boot: Disabling RNG early entropy source...
I (890) psram: This chip is ESP32-D0WD
I (890) spiram: Found 64MBit SPI RAM device
I (890) spiram: SPI RAM mode: flash 80m sram 80m
I (893) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (901) cpu_start: Pro cpu up.
I (904) cpu_start: Starting app cpu, entry point is 0x400813fc
I (897) cpu_start: App cpu up.
I (1410) spiram: SPI SRAM memory test OK
I (1422) cpu_start: Pro cpu start user code
I (1422) cpu_start: cpu freq: 240000000
I (1422) cpu_start: Application information:
I (1425) cpu_start: Project name:     Squeezelite-ESP32
I (1431) cpu_start: App version:      I2S-4MFlash-1673
I (1437) cpu_start: Compile time:     May  7 2024 21:45:16
I (1443) cpu_start: ELF file SHA256:  0000000000000000...
I (1449) cpu_start: ESP-IDF:          v4.3.5-dirty
I (1455) cpu_start: Min chip rev:     v1.0
I (1460) cpu_start: Max chip rev:     v3.99 
I (1465) cpu_start: Chip rev:         v3.1
I (1470) heap_init: Initializing. RAM available for dynamic allocation:
I (1477) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1483) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1489) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1495) heap_init: At 3FFC6EC8 len 00019138 (100 KiB): DRAM
I (1502) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1508) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1515) heap_init: At 4009E014 len 00001FEC (7 KiB): IRAM
I (1521) spiram: Adding pool of 4018K of external SPI memory to heap allocator
I (1529) spi_flash: detected chip: gd
I (1533) spi_flash: flash io: qio
W (1537) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (1553) cpu_start: Starting scheduler on PRO CPU.
[... 144 lines cut ...]
[00:02:45.806] I Shim.cpp:331: ZeroConf mode (port 80)
I (6761) httpd_handlers: serving /config.json to peer 192.168.1.16 port 22186
[00:02:48.505] discover_server:844 sending discovery 24
[00:02:48.511] discover_server:856 got response from: 192.168.1.48:3483
[00:02:48.512] slimproto:964 connecting to 192.168.1.48:3483
[00:02:48.518] slimproto:1011 connected
[00:02:48.519] sendHELO:153 mac: 64:b7:08:d1:d8:10
[00:02:48.530] sendHELO:155 cap: Model=squeezeesp32,AccuratePlayPoints=1,HasDigitalOut=1,HasPolarityInversion=1,Balance=1,Depth=16,Firmware=v1.0-1673-16,ModelName=SqueezeESP32,MaxSampleRate=192000,alc,aac,ogg,ops,ogf,flc,wav,aif,pcm,mp3
[00:02:48.544] sendSETD:440 sending dimension display:0x0 led_config:1
[00:02:48.558] notify:228 notified server 192.168.1.48 hport 9000 cport 9090
I (8431) network_status: LMS IP: 192.168.1.48, hport: 9000, cport: 9090
[00:02:48.572] process:578 unhandled vers
[00:02:48.577] grfb_handler:755 brightness 1
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x40102a62  PS      : 0x00060d30  A0      : 0x800fb3e4  A1      : 0x3f801a40  
A2      : 0x3f80a3fc  A3      : 0x00000208  A4      : 0x3f80e1a0  A5      : 0x00001000  
A6      : 0x3ffbfa88  A7      : 0x3ffc53f0  A8      : 0x80102a19  A9      : 0x00000000  
A10     : 0x00000001  A11     : 0x3ffbb034  A12     : 0x00000001  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40093da8  LEND    : 0x40093dc4  LCOUNT  : 0xffffffff  

Backtrace:0x40102a5f:0x3f801a40 0x400fb3e1:0x3f801b20 0x400fbe45:0x3f801b40 0x400ff5a2:0x3f801b60 0x400fa0f3:0x3f801c40 0x400de55c:0x3f801f70
</code></pre></details>

sle118, 2024-05-12:
I'm leaving this for the record:
```

Decoding stack results
0x40102a5f: handler at ../components/squeezelite/displayer.c line 709
0x400fb3e1: handler at ../components/squeezelite/output_embedded.c line 70
0x400fbe45: handler at ../components/squeezelite/output_i2s.c line 167
0x400ff5a2: slimproto at ../components/squeezelite/slimproto.c line 809
0x400fa0f3: squeezelite_main at ../components/squeezelite/main.c line 820
0x400de55c: squeezelite_thread at ../components/platform_console/app_squeezelite/cmd_squeezelite.c line 79

```

I'll try to have a look later tomorrow

wizmo2, 2024-05-28:
Apologies, been busy on other stuff. Just re-flashed one of my test systems with the latest version to test with HA with Slimproto,  

As I suspected, there is a divide-by-zero issue with Ledvu and a string length of 1.  I'll work on a PR to fix it.  

Saying that, Ledvu is for visual effects on a string of multiple addressable leds.  If you want to use a single WS2812 as a status light, use [set-GPIO](https://github.com/sle118/squeezelite-esp32#set-gpio) only.

wizmo2, 2024-05-29:
@sle118, the issue is with `grfe_handler`.   

Wen I added LEDVU. I included an extra field to SETD.  The display height reports 0 height, but currently `aioslimproto` sends a 'grfe' message regardless.  I opened an issue (https://github.com/home-assistant-libs/aioslimproto/issues/331).

In the time-being, we can fix with some extra checks in displayer.  Do you want me to create a PR?

closed 2024-08-03 by sle118