sle118/squeezelite-esp32 #419: ESP32-S3 Slimproto works, Cspot and Airplay don't

anabolyc, 2024-05-15:
### Describe the bug

hey folks, I'm testing squeezelite-esp32 with the ESP32-S3 board built with espidf v.4.4.7 Took me some time to prepare working sdkconfig, but code builds and runs, I was able to configure wifi and find device in the network. However only slim-proto seems to work as expected. Neither scpot nor airplay works as it should. 

### Preliminary Information

1. **Firmware Version**: [38d28a](https://github.com/sle118/squeezelite-esp32/commit/38d28ae8c44e7a967eb632addf5e30e215893331)
2. Music Assistant [slimproto](https://music-assistant.io/player-support/slimproto/) built-in into version 2024.3.6

### Hardware Details

Please describe your hardware setup:

- ESP32-S3-WROVER-N16R8
- Louder-ESP32-S3 board prototype
- TAS5805M DAC

### Airplay Logs

Airplay connects fine, but as soon as audio starts, errors are fired into the log. Speakers seem to issue occasional sounds, but you'd not recognize any audio of it.

<details>
<pre><code>
=== START OF LOG ===
I (24) boot: ESP-IDF v4.4.7 2nd stage bootloader
I (24) boot: compile time 07:36:59
I (24) boot: Multicore bootloader
I (26) boot: chip revision: v0.2
I (30) qio_mode: Enabling default flash chip QIO
I (35) boot.esp32s3: Boot SPI Speed : 80MHz
I (40) boot.esp32s3: SPI Mode       : QIO
I (45) boot.esp32s3: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (73) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  3 recovery         factory app      00 00 00010000 00140000
I (95) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (103) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (111) boot: End of partition table
I (115) esp_image: segment 0: paddr=00150020 vaddr=3c160020 size=a81f4h (688628) map
I (228) esp_image: segment 1: paddr=001f821c vaddr=3fc96300 size=04bb4h ( 19380) load
I (231) esp_image: segment 2: paddr=001fcdd8 vaddr=40374000 size=03240h ( 12864) load
I (236) esp_image: segment 3: paddr=00200020 vaddr=42000020 size=1583ach (1409964) map
I (456) esp_image: segment 4: paddr=003583d4 vaddr=40377240 size=0f0c0h ( 61632) load
I (468) esp_image: segment 5: paddr=0036749c vaddr=600fe000 size=0002ch (    44) load
I (476) boot: Loaded app from partition at offset 0x150000
I (476) boot: Disabling RNG early entropy source...
I (488) cpu_start: Multicore app
I (489) opi psram: vendor id : 0x0d (AP)
I (489) opi psram: dev id    : 0x02 (generation 3)
I (491) opi psram: density   : 0x03 (64 Mbit)
I (496) opi psram: good-die  : 0x01 (Pass)
I (501) opi psram: Latency   : 0x01 (Fixed)
I (506) opi psram: VCC       : 0x01 (3V)
I (510) opi psram: SRF       : 0x01 (Fast Refresh)
I (516) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (521) opi psram: BurstLen  : 0x01 (32 Byte)
I (526) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (532) opi psram: DriveStrength: 0x00 (1/1)
I (538) MSPI Timing: PSRAM timing tuning index: 5
I (542) spiram: Found 64MBit SPI RAM device
I (547) spiram: SPI RAM mode: sram 80m
I (552) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (559) cpu_start: Pro cpu up.
I (562) cpu_start: Starting app cpu, entry point is 0x40375314
0x40375314: call_start_cpu1 at /opt/esp/idf/components/esp_system/port/cpu_start.c:175

[... 628 lines cut ...]

0x420477df: tcp_input at /opt/esp/idf/components/lwip/lwip/src/core/tcp_in.c:490

0x42046dcd: tcp_input at /opt/esp/idf/components/lwip/lwip/src/core/tcp_in.c:238

0x42040615: lwip_setsockopt_impl at /opt/esp/idf/components/lwip/lwip/src/api/sockets.c:3854
 (inlined by) lwip_setsockopt_callback at /opt/esp/idf/components/lwip/lwip/src/api/sockets.c:3417

0x42041cbd: lwip_fcntl at ??:?

0x4203a6e2: esp_netif_destroy at /opt/esp/idf/components/esp_netif/lwip/esp_netif_lwip.c:655

0x42031576: bootloader_common_check_chip_validity at /opt/esp/idf/components/bootloader_support/src/bootloader_common_loader.c:89

0x4202f9e2: uart_read_bytes at /opt/esp/idf/components/driver/uart.c:1344

=== END OF LOG ===   
</code></pre>
</details>

### Issue Description

1. **Observed Behavior**: Neither Airplay not cspot works
2. **Expected Behavior**: Both should work
3. **Steps to Reproduce**: Build latest code with espidf 4.4.7 using attached sdkconfig and flash to the device.

anabolyc, 2024-05-15:
### sdkconfig

<details>
<pre><code>
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32s3"
CONFIG_IDF_TARGET_ESP32S3=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0009

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x0
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3

#
# Serial Flash Configurations
#
# CONFIG_BOOTLOADER_FLASH_DC_AWARE is not set
CONFIG_BOOTLOADER_FLASH_XMC_SUPPORT=y
# end of Serial Flash Configurations

CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_REGION_PROTECTION_ENABLE=y
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
[... 1683 lines cut ...]
  "led_vu_config": "",
  "lms_ctrls_raw": "n",
  "loudness": "0",
  "metadata_config": "",
  "model_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
  "pollmx": "600",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "CONFIG_SQUEEZELITE_ESP32_RELEASE_URL",
  "rotary_config": "",
  "set_GPIO": "17=vcc",
  "sleep_config": "",
  "spdif_config": "",
  "spi_config": "",
  "stats": "n",
  "target": "",
  "telnet_block": "500",
  "telnet_buffer": "40000",
  "telnet_enable": ""
}
</code></pre>
</details>

anabolyc, 2024-05-15:
### sdkconfig

[sdkconfig.esp32s3.txt](https://github.com/sle118/squeezelite-esp32/files/15318480/sdkconfig.esp32s3.txt)

sle118, 2024-05-15:
In all instances, this seems to be related to memory allocation and caused by some sdkconfig settings. Did you start from the experimental branch?

sdkconfig fine tuning is an art when pushing the limits of a system like we do. @philippe44 has mastered this more than me, so he would be better placed to answer, although he has currently limited availability. 

Make sure you stay from his sdkconfig file, or compare with yours. 
https://github.com/sle118/squeezelite-esp32/blob/4.4-s3/sdkconfig
There are several settings that lead to consuming internal memory, which is limited on this platform.

anabolyc, 2024-05-15:
Thanks @sle118 I didn't realize that there is a branch for S3. I'll re-test and drop here a line

anabolyc, 2024-05-15:
It is actually fixed both cspot and airplay. It is not particularly stable (compared to the ESP32 version), but generally works. I hope S3 support will not be dropped in the future.

sle118, 2024-05-15:
Very likely that we will support it at some point

closed 2024-05-15 by sle118

gulliver103, 2024-05-19:
Hi @anabolyc !
I am trying to run/compile this but it is far too complicated for me.
Would you sharing your bin- file with me?
I am trying to flash it on a Seeed Studio esp32-s3 with 8M flash and 8M PSRAM.
Would be great! Thanks /Gulliver

anabolyc, 2024-05-19:
@gulliver103 give me some time, I'll publish web-flasher. I'm about to release the Louder-ESP32-S3 board, so I'm going to need this build anyway. I'll try to finish it tomorrow

anabolyc, 2024-05-20:
Hey @gulliver103 please have a go at https://sonocotta.github.io/esp32-audio-dock/ web-installer.
If you need binaries only, they are under https://github.com/sonocotta/esp32-audio-dock/tree/main/docs/artifacts

gulliver103, 2024-05-21:
> Hey @gulliver103 please have a go at https://sonocotta.github.io/esp32-audio-dock/ web-installer. If you need binaries only, they are under https://github.com/sonocotta/esp32-audio-dock/tree/main/docs/artifacts

Thank you :) I´ll try it out on my ["Seeed Studio XIAO ESP32S3" ](https://wiki.seeedstudio.com/xiao_esp32s3_getting_started/) and get back to you (Y)

gulliver103, 2024-05-21:
Hi again @anabolyc  and sorry for taking your time.
After installing from the https://sonocotta.github.io/esp32-audio-dock/  the AP get disconnected with the following error at the end of the log. Please have a look when you have time. Thank you /Gulliver

I (35774) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<0,0>, prof:1
I (35774) wifi:station: 9a:9c:22:19:90:ca join, AID=1, bgn, 20
I (35774) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<0,0>, prof:1
I (35784) wifi:station: 9a:9c:22:19:90:ca join, AID=1, bgn, 20
I (37194) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
W (38424) httpd_uri: httpd_uri: URI '/chat' not found
W (38434) httpd_handlers: redirect_processor(1002): 192.168.4.2 requested invalid URL: [/chat]
W (38454) httpd_handlers: process_redirect(957): Redirecting host [192.168.4.2] to http://192.168.4.1/ (from uri /chat)
CORRUPT HEAP: Bad tail at 0x3d81dd6b. Expected 0xbaad5678 got 0xbaad5600

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)

Backtrace: 0x40376296:0x3d80bc70 0x4037fac1:0x3d80bc90 0x40384d01:0x3d80bcb0 0x40384806:0x3d80bdd0 0x4037686d:0x3d80bdf0 0x40384d31:0x3d80be10 0x4201f24d:0x3d80be30 0x4201f49f:0x3d80be80 0x4201f604:0x3d80bef0 0x4205a89d:0x3d80bf10 0x4205ac15:0x3d80bf30 0x42059db8:0x3d80bf70 0x4205a31a:0x3d80c020 0x420590a4:0x3d80c040 0x420bdc82:0x3d80c060 0x420591af:0x3d80c080

ELF file SHA256: 48c219f9a6c31c40

CPU halted.
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42006524
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1820
load:0x403c9700,len:0xe80
load:0x403cc700,len:0x3110
entry 0x403c9980
I (24) boot: ESP-IDF v4.4.7 2nd stage bootloader
I (24) boot: compile time 13:57:36
I (25) boot: Multicore bootloader
I (26) boot: chip revision: v0.2
I (30) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32s3: Boot SPI Speed : 80MHz
I (40) boot.esp32s3: SPI Mode       : QIO
I (45) boot.esp32s3: SPI Flash Size : 8MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 recovery         factory app      00 00 00010000 00140000
I (96) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (103) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (111) boot: End of partition table
I (115) boot: Defaulting to factory image
I (120) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=59548h (365896) map
I (184) esp_image: segment 1: paddr=00069570 vaddr=3fc96ce0 size=04870h ( 18544) load
I (188) esp_image: segment 2: paddr=0006dde8 vaddr=40374000 size=02230h (  8752) load
I (191) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=c64d0h (812240) map
I (321) esp_image: segment 4: paddr=001364f8 vaddr=40376230 size=10ab0h ( 68272) load
I (335) esp_image: segment 5: paddr=00146fb0 vaddr=600fe000 size=0002ch (    44) load
I (343) boot: Loaded app from partition at offset 0x10000
I (343) boot: Disabling RNG early entropy source...
I (355) cpu_start: Multicore app
I (355) opi psram: vendor id : 0x0d (AP)
I (355) opi psram: dev id    : 0x02 (generation 3)
I (358) opi psram: density   : 0x03 (64 Mbit)
I (363) opi psram: good-die  : 0x01 (Pass)
I (367) opi psram: Latency   : 0x01 (Fixed)
I (372) opi psram: VCC       : 0x01 (3V)
I (377) opi psram: SRF       : 0x01 (Fast Refresh)
I (382) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (388) opi psram: BurstLen  : 0x01 (32 Byte)
I (393) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (399) opi psram: DriveStrength: 0x00 (1/1)
I (404) MSPI Timing: PSRAM timing tuning index: 5
[... 119 lines cut ...]
recovery-squeezelite-esp32>W (5049) network_wifi: WIFI_EVENT_STA_DISCONNECTED. From BSSID: 00:00:00:00:00:00, reason code: 201 (WIFI_REASON_NO_AP_FOUND)
W (5059) wifi:Haven't to connect to a suitable AP now!
I (5069) network_wifi: Configuring Access Point.
I (5069) network: Stopping DHCP server
I (5079) network: Starting DHCP server
I (5079) network_wifi: AP SSID: louder-esp-s3
I (5079) network_wifi: AP Password: squeezelite
I (5089) wifi:mode : sta (dc:da:0c:18:44:24) + softAP (dc:da:0c:18:44:25)
I (5089) wifi:Total power save buffer number: 6
I (5099) wifi:Init max length of beacon: 752/752
I (5109) wifi:Init max length of beacon: 752/752
I (5469) wifi:Total power save buffer number: 6
I (5469) wifi:Total power save buffer number: 6
I (5469) wifi:Set ps type: 1

I (5479) network_wifi: Initiating wifi network scan
I (5479) dns_server: DNS Server listening on 53/udp
E (17649) network: network manager Unhandled Event NETWORK_WIFI_CONFIGURING_ACTIVE_STATE(WIFI_CONFIGURING_STATE) [EN_TIMER]
I (39409) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<0,0>, prof:1
I (39419) wifi:station: 9a:9c:22:19:90:ca join, AID=1, bgn, 20
I (39849) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
W (41129) httpd_uri: httpd_uri: URI '/chat' not found
W (41139) httpd_handlers: redirect_processor(1002): 192.168.4.2 requested invalid URL: [/chat]
W (41159) httpd_handlers: process_redirect(957): Redirecting host [192.168.4.2] to http://192.168.4.1/ (from uri /chat)
CORRUPT HEAP: Bad tail at 0x3d81e057. Expected 0xbaad5678 got 0xbaad5600

anabolyc, 2024-05-22:
Hey @gulliver103 

Binary has my wifi credentials (shame on me), it tries to connect and fails. So AP is started, looks like you were able to connect to it. Next if fails on /chat request. I don't think it is a part of the squeezelite application. Can you try accessing HTTP://192.168.4.1 with the browser cache disabled?

anabolyc, 2024-05-27:
> > Hey @gulliver103 please have a go at https://sonocotta.github.io/esp32-audio-dock/ web-installer. If you need binaries only, they are under https://github.com/sonocotta/esp32-audio-dock/tree/main/docs/artifacts
> 
> Hi @anabolyc! How would i use your installer with generic ESP32-S3 with DAC? :) I guess i need DAC to be connected to specific GPIO?

It should work with any ESP32-S3. All the board-specific details, like DAC config, pinouts etc are stored on the NVS, which you can change later on via web UI.

formatBCE, 2024-05-27:
> It should work with any ESP32-S3. All the board-specific details, like DAC config, pinouts etc are stored on the NVS, which you can change later on via web UI.

I use S3-Devkit-C with 16 MB flash and 8 MB PSRAM. DAC: I2S MCU-1334.
Flashed, connected to the WiFi, configured pins. Tried to play through line-in - didn't work. Wanted to check other settings, but at some point it started to freeze completely within 10 sec after reboot. 
I'm pretty sure, that it happened after i put onboard LED parameters (there's one WS2812 LED on board, connected to GPIO48). Wanted to check, if it reacts somehow on input.

formatBCE, 2024-05-28:
> It should work with any ESP32-S3. All the board-specific details, like DAC config, pinouts etc are stored on the NVS, which you can change later on via web UI.

@anabolyc 
So long story short, i tried to make ESPHome media player on that device, to check if it's working, and found that the board is behaving strange (right channel gives noise instead of output). I switched to other board (with 8 MB of flash and 2 MB of PSRAM), and ESPHome works well. However, after installing your S3 louder build, i see in logs:
```
E (355) opi psram: PSRAM ID read error: 0x00000000, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (359) cpu_start: Failed to init external RAM!
```

As far as i know, N8R2 has `quad` mode instead of `octal` in N16R8. Is it the case? Or 2 MB of PSRAM is not enough?
Or both?

Thanks in advance! :)

wizmo2, 2024-05-28:
>  I'm pretty sure, that it happened after i put onboard LED parameters (there's one WS2812 LED on board, connected to GPIO48). 

[there is a known divide-by-zero issue with Ledvu and a string length of 1...](https://github.com/sle118/squeezelite-esp32/issues/414#issuecomment-2136259231)

martymcbro, 2024-07-19:
> It should work with any ESP32-S3. All the board-specific details, like DAC config, pinouts etc are stored on the NVS, which you can change later on via web UI.

Hi @anabolyc
I'm also trying to achieve what @gulliver103 tried (run on XIAO ESP32-S3) and have the same problem with your build (installed through webinstaller for the louder esp32-s3).
Either when I connect to wifi or when I navigate to 192.168.4.1 the device crashes and the AP disappears. I can also add the logs later today. Do you have any idea what could be going on?
Thank you for the Esparagus Hifi Media Link btw, i'm a happy customer :-)

@gulliver103 Did you make any progress on this?

I also made my own built from the 4.4-s3 branch with the default sdkconfig from the branch and no changes in menuconfig. I can flash it but the last message I see is that the http server is created. Then it seems to hang. Wifi AP is not created.

martymcbro, 2024-07-19:
Here is the log with 3 connection attempts. It seems like it crashes as soon as the client connects to the AP.

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420bc98a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1820
load:0x403c9700,len:0xe80
load:0x403cc700,len:0x3110
entry 0x403c9980
I (24) boot: ESP-IDF v4.4.7 2nd stage bootloader
I (24) boot: compile time 13:57:36
I (25) boot: Multicore bootloader
I (26) boot: chip revision: v0.2
I (30) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32s3: Boot SPI Speed : 80MHz
I (40) boot.esp32s3: SPI Mode       : QIO
I (45) boot.esp32s3: SPI Flash Size : 8MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 recovery         factory app      00 00 00010000 00140000
I (96) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (103) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (111) boot: End of partition table
I (115) boot: Defaulting to factory image
I (120) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=59548h (365896) map
I (184) esp_image: segment 1: paddr=00069570 vaddr=3fc96ce0 size=04870h ( 18544) load
I (188) esp_image: segment 2: paddr=0006dde8 vaddr=40374000 size=02230h (  8752) load
I (191) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=c64d0h (812240) map
I (321) esp_image: segment 4: paddr=001364f8 vaddr=40376230 size=10ab0h ( 68272) load
I (335) esp_image: segment 5: paddr=00146fb0 vaddr=600fe000 size=0002ch (    44) load
I (343) boot: Loaded app from partition at offset 0x10000
I (343) boot: Disabling RNG early entropy source...
I (355) cpu_start: Multicore app
I (355) opi psram: vendor id : 0x0d (AP)
I (355) opi psram: dev id    : 0x02 (generation 3)
I (358) opi psram: density   : 0x03 (64 Mbit)
I (363) opi psram: good-die  : 0x01 (Pass)
I (367) opi psram: Latency   : 0x01 (Fixed)
I (372) opi psram: VCC       : 0x01 (3V)
I (377) opi psram: SRF       : 0x01 (Fast Refresh)
I (382) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (388) opi psram: BurstLen  : 0x01 (32 Byte)
I (393) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (399) opi psram: DriveStrength: 0x00 (1/1)
I (404) MSPI Timing: PSRAM timing tuning index: 5
I (409) spiram: Found 64MBit SPI RAM device
I (414) spiram: SPI RAM mode: sram 80m
I (418) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (425) cpu_start: Pro cpu up.
I (429) cpu_start: Starting app cpu, entry point is 0x40375820
I (0) cpu_start: App cpu up.
I (862) spiram: SPI SRAM memory test OK
I (870) cpu_start: Pro cpu start user code
I (871) cpu_start: cpu freq: 240000000
I (871) cpu_start: Application information:
I (873) cpu_start: Project name:     Squeezelite-ESP32
I (879) cpu_start: App version:      local.500.cmake-master
I (885) cpu_start: Compile time:     May 15 2024 13:58:06
I (891) cpu_start: ELF file SHA256:  48c219f9a6c31c40...
I (897) cpu_start: ESP-IDF:          v4.4.7
I (902) cpu_start: Min chip rev:     v0.0
I (907) cpu_start: Max chip rev:     v0.99 
I (912) cpu_start: Chip rev:         v0.2
[... 542 lines cut ...]
I (90799) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<0,0>, prof:1
I (95429) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<0,0>, prof:1
I (95439) wifi:station: 80:30:49:41:2c:af join, AID=1, bgn, 20
I (95529) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (161079) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<0,0>, prof:1
I (161079) wifi:station: 80:30:49:41:2c:af join, AID=1, bgn, 20
I (165079) wifi:station: 80:30:49:41:2c:af leave, AID = 1, bss_flags is 655458, bss:0x3d81d6a8
I (165089) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<0,0>, prof:1
I (169759) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<0,0>, prof:1
I (169759) wifi:station: 80:30:49:41:2c:af join, AID=1, bgn, 20
I (169849) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (171889) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (174689) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (179319) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (187689) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (204429) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (204649) httpd_handlers: serving / to peer 192.168.4.2 port 35529
W (204679) httpd_txrx: httpd_sock_err: error in send : 104
W (204679) httpd_txrx: httpd_sock_err: error in recv : 104
W (205529) httpd_uri: httpd_uri: URI '/canonical.html' not found
W (205549) httpd_handlers: redirect_processor(1002): 192.168.4.2 requested invalid URL: [/canonical.html]
W (205559) httpd_handlers: process_redirect(957): Redirecting host [192.168.4.2] to http://192.168.4.1/ (from uri /canonical.html)
CORRUPT HEAP: Bad tail at 0x3d81d59f. Expected 0xbaad5678 got 0xbaad5600

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)

philippe44, 2024-07-19:
Compile without heap poisoning to try

gulliver103, 2024-07-19:
I gave up!! Using esp32-wrover instead :)

philippe44, 2024-07-19:
Just do as I said. I don't have time to investigate why for now, but there is a heap issue that does not affect runtime although if heap corruption detection is activated, created a reboot.

anabolyc, 2024-07-19:
Sorry folks, I didn't have a chance to test it thoroughly. I can try recompiling sometime next week (I'm out of the lab till Tuesday). 
@philippe44 should I set `CONFIG_HEAP_POISONING_DISABLED=y` ?

BTW, current sdkconfig suggests _external_ PSRAM as far as I remember. All the boards I tested had ESP32-S3-WROVER module. I don't think it will work with _internal_ PSRAM, but I can be wrong.

anabolyc, 2024-07-23:
@martymcbro @gulliver103 I can confirm the issue. I missed it before because I have my wifi credentials flashed in the nvs storage (ups!). I will try to rebuild with philippe44 suggestion and see if it helps

anabolyc, 2024-07-23:
I've updated [online installer](https://sonocotta.github.io/esp32-audio-dock/) can someone please flash and test wifi provisioning? Works for me now.

BTW, [this](https://github.com/sle118/squeezelite-esp32/issues/187) would simplify wifi provisioning a tonn. We [integrated improv](https://github.com/CarlosDerSeher/snapclient/commit/a619587c03ced4284383fef2e6ba2343bb119bb2) in the snaplient and it works like a treat.

gulliver103, 2024-07-23:
> I've updated [online installer](https://sonocotta.github.io/esp32-audio-dock/) can someone please flash and test wifi provisioning? Works for me now.
>
> BTW, [...]

@anabolyc Works so much better :) Thank you (Y)

formatBCE, 2024-07-24:
@anabolyc Strange thing, it booted successfully, i added WiFi, changed GPIO config, rebooted, and it gives this on boot now:
```***ERROR*** A stack overflow in task Tmr Svc has been detected.```

anabolyc, 2024-07-24:
@formatBCE can you please share which settings exactly you set to make it fail?

@martymcbro seems that they use S3-WROOM module with 8Mb of PSRAM, so it should work.

formatBCE, 2024-07-24:
> @anabolyc For me the new version is working without problems on my XIAO ESP32-S3. I also connected a PCM5102 DAC and configured it. This is also working flawlessly. Thank you!
> 

Interesting! I have XIAO S3 module too, using it with UDA1334. Somehow i managed to get it stable - but it doesn't appear in either my Music Assistant, where i have another Squeezelite player already, or in Spotify (i have cspot enabled).

formatBCE, 2024-07-24:
> @formatBCE can you please share which settings exactly you set to make it fail?
> 

All the default ones:
- connected to WiFi
- changed the name to seeed-speaker
- removed Bluetooth and Airplay ticks, set Zeroconf for cspot to 1
- changed DAC to I2S, changed clock, word-select and data pins to corresponding ones
- pushed "exit recovery"

martymcbro, 2024-07-24:
OK, I did similar things. I also connected to Wifi, renamed the player, disabled Bluetooth and Airplay. Zeroconf for cspot was already set to 1. I also configured the DAC, additionaly I removed conflicts for the PINs I used for the DAC from the other hardware sections.

Maybe I did another important thing. When I left telnet enabled I could not connect to the player after leaving recovery. The serial debug console did only tell me that output was redirected to telnet so I could not see what was going wrong. After disabling telnet the player worked as expected. So maybe there is still a bug left.

martymcbro, 2024-07-24:
Now I also removed the Ethernet config using the NVS editor, in case this is causing some problems. But I'm not brave enough to turn telnet back on.

anabolyc, 2024-07-24:
Was anyone able to debug those issues with S3 built-in debugger-over-USB? I'm not sure if it is supported in the IDF4.3, but I was using it in the latest IDF and it was brilliant for fixing issues like those

formatBCE, 2024-07-24:
@anabolyc i connected to web.esphome.io, and see logs. But IDK what to debug. :)

martymcbro, 2024-07-25:
If you are talking about the "idf.py monitor" command I can try this when I find the time. I did that recently with my own build, but with idf 4.4. So I will clone your repository and give it a try. Maybe I can find out more about the telnet issue.

anabolyc, 2024-07-25:
I'm talking about [this](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/jtag-debugging/index.html). Technically all you need is a USB port.

formatBCE, 2024-07-25:
@anabolyc maybe this will help? This is work log, device doesn't hang/reboot, but doesn't show in Spotify or Music Assistant.
https://pastebin.com/9q8E1yYD

martymcbro, 2024-07-26:
@anabolyc Can I debug this just with the binary? In your Github it seems that the source is not available or I was not able to find it. I thought that I need to compile the source to use the debugger? Which IDF version did you use?

formatBCE, 2024-08-19:
Alright, i gave it another shot. 
Made just minimal changes, disabled Telnet right away (it's Telnet that gives `stack overflow`) - and it works! :)
Nice thing! Thank you @anabolyc 

Slava Ukraini!

polygon242, 2024-08-25:
Seeed XIAO ESP32-S3 board running nicely so far using the info here.  
I hope the next release will add S3 support to the Webinstaller, with telnet disabled. ;)

martymcbro, 2024-08-25:
I can try to debug the telnet problem, however I cannot find the source code @anabolyc uses to build the firmware.

anabolyc, 2024-08-25:
Hey @martymcbro I'm using top of the [master-v4.3](https://github.com/sle118/squeezelite-esp32/commit/40a698e2f1007dff47c4f180aadb0282610d08da) branch and [this sdkconfig](https://github.com/sle118/squeezelite-esp32/blob/4.4-s3/sdkconfig) from 4.4-s3 branch.

martymcbro, 2024-09-28:
Hi @anabolyc,

Sorry for the long delay, my notebook died and I had to get a new one :sunglasses:

I tried to build the git commit with the sdkconfig you mentioned above, but I ran into problems when using esp-idf 4.3.7 and also when using esp-idf 4.4.8.

Unfortunately I am not advanced enough to resolve these. I attached the build output. When using esp-idf 4.4.8 the error has something to do with cspot, so maybe this is one of the problems you had with cspot before? Do I need another change in the sdkconfig?

Best Regards

[build_4.3.7.log](https://github.com/user-attachments/files/17174490/build_4.3.7.log)
[build_4.4.8.log](https://github.com/user-attachments/files/17174491/build_4.4.8.log)

anabolyc, 2024-09-29:
@martymcbro 

Seems to be missing dependencies or whatnot, don't have a clue how to fix it. I'd advice to use esp docker image for build, I was using 4.4.7 which is the only one I know is successful. Try that

```
docker run --rm -it -v .:/project -w /project -v /dev:/dev --privileged espressif/idf:v4.4.7
```

make sure you have the right `sdkconfig` in the root and run `idf.py build flash monitor` as usual.

Just for reference, my sdkconfig for S3

[sdkconfig.txt](https://github.com/user-attachments/files/17178601/sdkconfig.txt)

martymcbro, 2024-09-29:
@anabolyc 

Thank you for the hint. Everything worked now using your docker command.

After enabling telnet the system crashed again. The reason is a stack overflow in task Tmr Svc.
Attached you find the output of "idf.py monitor" containing the backtrace.

[monitor.log](https://github.com/user-attachments/files/17180900/monitor.log)

@philippe44 

I think you are one of the few persons with enough insight to fix this. Looking at the backtrace, do you have a clue what can be done?

formatBCE, 2024-09-29:
@martymcbro Telnet never worked for me on esp32s3. I disable that before going out the recovery mode

anabolyc, 2024-09-29:
I can confirm that telent never worked, I can accept that for now.

martymcbro, 2024-09-29:
I thought we already agreed on the fact that telnet isn't working, but more information was needed to possibly fix the telnet issue 😉

philippe44, 2024-10-02:
We can fix it by increasing the stack but it is precious internal ram AFAIR. The esp-idf timer calls any callback registered under timer service using its own task (not an isr) stack. This is why one has to be careful what we do in that callback and also because it's a shared task for all callbacks, best is to either something really quick or send a message to wake up another task. I'll have to verify all the routines that we register for timer services.

pkuttiyam1, 2025-09-20:
Hello:
I have a ESP32S3 WROOM Devkit-C1 (8MB external PSRAM) that I am trying to get squeezelite loaded and working.
Mainly interested in getting the Spotify Connect part up and running.

I followed the instructions (took the Docker path):
docker pull sle118/squeezelite-esp32-idfv435
docker run -it -v `pwd`:/workspace/squeezelite-esp32 sle118/squeezelite-esp32-idfv435
cd to working dir
idf.py set-target esp32s3
idf.py build

Is ESP32S3 supported as is on the main branch? I dont see the "4.4-s3" branch anymore.
Now, I had to wade through a lot of build time options and Kconfig changes to get the build working, but when I flash using:
python3 -m esptool --chip esp32s3 -p "$PORT" -b 460800 write_flash @flash_project_args

It boots to recovery bin and runs the http webserver where I sign into the Wifi of the ESP32, go to webpage and then enter my wifi credentials. It seems like it likes that because I can connect to the webserver.

But after that: If I load the application "squeezelite.bin" from the "Updates" tab on the webserver, it is not able to boot up to the  Application.
I get this on the serial console:
****************************
Your terminal application does not support escape sequences.
Line editing and history features are disabled.
On Windows, try using Putty instead.
****************************
␛[0;32mI (3027) wifi_init: tcp mss: 1440␛[0m
␛[0;33mW (3047) console: Processing autoexec commands while network manager active.  Wifi related commands will be ignored.␛[0m
␛[0;32mI (3047) wifi_init: WiFi IRAM OP enabled␛[0m
␛[0;32mI (3057) console: autoexec is set to perform auto-process␛[0m
␛[0;32mI (3057) wifi_init: WiFi RX IRAM OP enabled␛[0m
␛[0;32mI (3067) console: Running command autoexec1 = squeezelite -o I2S -b 500:2000 -d all=info -C 30 -n squeezelite-16a5d4␛[0m
␛[0;32I2S abort!
I2S abort!
I2S abort!
I2S abort!
I2S abort! (keeps spitting out this error).

Is Squeezelite supported on ESP32S3 ?
Can @sle118  or @feelfreelinux help please ?

anabolyc, 2026-02-01:
Hey @macegr 

I did some work on SDKCONFIG and updated to the latest master code. It seems to work now with both ESP32 and ESP32-S3, you welcome to test it: https://sonocotta.github.io/esp32-audio-dock/