sle118/squeezelite-esp32 #420: support for Seeed Studio XIAO ESP32S3 - 2.4GHz WiFi, BLE 5.0, 8MB PSRAM, 8MB FLASH, Dual-core

gulliver103, 2024-05-17:
Add support for
Seeed Studio XIAO ESP32S3 - 2.4GHz WiFi, BLE 5.0, 8MB PSRAM, 8MB FLASH, Dual-core

https://www.seeedstudio.com/xiao-series-page

sle118, 2024-05-17:
There's an experimental branch for the S3 and we will likely officialise that at some point.

closed 2024-05-17 by sle118

martymcbro, 2024-07-19:
Hi,
I don't know if this is still the right place to get support for the Xiao ESP32-S3 or if I should better start a new discussion.

I have a problem getting the 4.4-s3 branch to run on a Seeed Studio XIAO [ESP32S3](https://wiki.seeedstudio.com/xiao_esp32s3_getting_started/). I just cloned the branch as it is and kept the default sdkconfig. I did not change anything in menuconfig and after fixing some errors regarding protobuf I got it to build successfully with esp-idf 4.4. However after flashing it gets stuck and I can see no Wifi AP.

I don't know if there is still a big issue left or if it's something small. Maybe someone can help me.

Thank you

Here is the log:

idf.py -p /dev/ttyACM0 monitor
/home/martin/esp/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
import pkg_resources
Executing action: monitor
Running idf_monitor in directory /home/martin/squeezelite-esp32
Executing "/home/martin/.espressif/python_env/idf4.4_py3.10_env/bin/python /home/martin/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32s3 /home/martin/squeezelite-esp32/build/recovery.elf -m '/home/martin/.espressif/python_env/idf4.4_py3.10_env/bin/python' '/home/martin/esp/esp-idf/tools/idf.py' '-p' '/dev/ttyACM0'"...
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42006324
0x42006324: esp_panic_handler at /home/martin/esp/esp-idf/components/esp_system/panic.c:382 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x178c
load:0x403b6000,len:0xe28
load:0x403ba000,len:0x30b8
entry 0x403b6274
I (24) boot: ESP-IDF v4.4 2nd stage bootloader
I (25) boot: compile time 17:14:53
I (25) boot: chip revision: 0
I (26) qio_mode: Enabling default flash chip QIO
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode : QIO
I (41) boot.esp32s3: SPI Flash Size : 8MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label Usage Type ST Offset Length
I (62) boot: 0 nvs WiFi data 01 02 00009000 00004000
I (69) boot: 1 otadata OTA data 01 00 0000d000 00002000
I (77) boot: 2 phy_init RF data 01 01 0000f000 00001000
I (84) boot: 3 recovery factory app 00 00 00010000 00140000
I (92) boot: 4 ota_0 OTA app 00 10 00150000 002a0000
I (99) boot: 5 settings WiFi data 01 02 003f0000 00010000
I (107) boot: End of partition table
I (111) boot: Defaulting to factory image
I (116) esp_image: segment 0: paddr=00010020 vaddr=3c0c0020 size=57ce4h (359652) map
I (179) esp_image: segment 1: paddr=00067d0c vaddr=3fc95a70 size=04cd0h ( 19664) load
I (183) esp_image: segment 2: paddr=0006c9e4 vaddr=40374000 size=03634h ( 13876) load
I (187) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=bd3fch (775164) map
I (310) esp_image: segment 4: paddr=0012d424 vaddr=40377634 size=0e434h ( 58420) load
I (322) esp_image: segment 5: paddr=0013b860 vaddr=50000000 size=00010h ( 16) load
I (322) esp_image: segment 6: paddr=0013b878 vaddr=600fe000 size=00028h ( 40) load
I (335) boot: Loaded app from partition at offset 0x10000
I (335) boot: Disabling RNG early entropy source...
I (351) opi psram: vendor id : 0x0d (AP)
I (351) opi psram: dev id : 0x02 (generation 3)
I (351) opi psram: density : 0x03 (64 Mbit)
I (355) opi psram: good-die : 0x01 (Pass)
I (359) opi psram: Latency : 0x01 (Fixed)
I (364) opi psram: VCC : 0x01 (3V)
I (369) opi psram: SRF : 0x01 (Fast Refresh)
I (374) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (380) opi psram: BurstLen : 0x01 (32 Byte)
I (385) opi psram: Readlatency : 0x02 (10 cycles@Fixed)
I (391) opi psram: DriveStrength: 0x00 (1/1)
W (396) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
[... 48 lines cut ...]
I (2350) esp_app_main: Configuring services
I (2350) services: Configuring I2C sda:-1 scl:-1 port:1 speed:400000
W (2360) services: no I2C configured
I (2360) services: Configuring SPI mosi:-1 miso:-1 clk:-1 host:1 dc:-1
W (2370) services: no SPI configured
E (2380) ledc: ledc_timer_config(322): speed_mode argument is invalid
W (2380) led: LED GPIO -1 ignored
W (2390) led: LED GPIO -1 ignored
I (2390) led: Configuring LEDs green:-1 (active:0 -1%), red:-1 (active:0 -1%)
I (2400) battery: No battery
I (2400) monitor: Initializing monitoring
I (2410) monitor: Heap internal:293059 (min:289875) external:8367587 (min:8367571) dma:287335 (min:284151)
I (2420) esp_app_main: Initializing display
I (2420) display: Trying to configure display with N/A
W (2430) display: No display driver
I (2430) esp_app_main: Checking for update url
W (2440) config: Value not found for key fwurl
I (2440) esp_app_main: Starting Network Manager
W (2450) config: Value not found for key apdelay
I (2450) uart: queue free spaces: 3
I (2450) network_handlers: Network manager configuration: polling max 10176, polling min 15000, ap delay 20000, dhcp timeout 8000, eth timeout 8000
W (2460) network_wifi: No known access point found
E (2470) nvs_utilities: Could not open the nvs storage.
W (2470) network_wifi: network manager has no previous configuration. ESP_FAIL
I (2480) http_server: Initializing HTTP Server