sle118/squeezelite-esp32 #422: Feature request: S3 WROOM-1 installer

formatBCE, 2024-05-24:
Hi! Awesome project!

Could we please get installer for WROOM-1 ESP32-S3 with 4+ PSRAM module? I have N16R8 Devkit-C board, and can't install on it... Although it should be more than enough for this purpose.

Thank you in advance! If you need me to test something - i'm always up to!

sle118, 2024-05-27:
Support for the S3 is still experimental at this point and neither @philippe44 note myself have time right now to integrate it to the main branch. So unless someone wants to tackle this, is not going to happen in the short term I'm afraid. 

It's better to stick to the trusted and cheap Wrover for now

formatBCE, 2024-05-27:
> Support for the S3 is still experimental at this point and neither @philippe44 note myself have time right now to integrate it to the main branch. So unless someone wants to tackle this, is not [...]

Thanks for the answer.

ithinkido, 2024-05-27:
https://github.com/sle118/squeezelite-esp32/issues/419#issuecomment-2120435169

anabolyc, 2026-02-01:
Hey @formatBCE 

I did some work on SDKCONFIG and updated to the latest master code. It seems to work now with both ESP32 and ESP32-S3, you welcome to test it: https://sonocotta.github.io/esp32-audio-dock/

nehasjains, 2026-09-04:
I am planning to buy esp32-s3 board to install squeezelite on it. The following is the link to this board.

https://hubtronics.in/esp32-s3-wroom1-n16r8-dev-kit?search=Esp32-s3&page=4

Can I flash squeezelite on this board using a web installer from the below website. If yes which model do I have to select?

https://sonocotta.github.io/esp32-audio-dock/

Kindly guide.

anabolyc, 2026-09-04:
> I am planning to buy esp32-s3 board to install squeezelite on it. The following is the link to this board.
>
> https://hubtronics.in/esp32-s3-wroom1-n16r8-dev-kit?search=Esp32-s3&page=4
>
> Can I flash [...]

PLease try installing HiFi-ESP32-S3, and then you need to enter recovery and update settings to fit your board. This board is the simplest one in terms of the DAC. One concern - your board seems to have 16mb flash, mine are all 8mb. Not sure if it will validate that

nehasjains, 2026-09-04:
> PLease try installing HiFi-ESP32-S3, and then you need to enter recovery and update settings to fit your board. This board is the simplest one in terms of the DAC. One concern - your board seems to [...]

I dont want to take any risk as I have yet to buy this, there is also one more board with 8mb psram and 8mb flash.

https://hubtronics.in/esp32-s3-dev-kit-n8r8-nh

Shall I order this one, returning back to seller is very tough here.

So please advise.

anabolyc, 2026-09-04:
@nehasjains either should work, 16mb version looks cheaper and has 2 usbs, can be useful.

nehasjains, 2026-09-04:
> [@nehasjains](https://github.com/nehasjains) either should work, 16mb version looks cheaper and has 2 usbs, can be useful.

You mentioned earlier that the board has 16mb and it might not validate!!!

Shall I go with the 16mb then ?

anabolyc, 2026-09-04:
> > [@nehasjains](https://github.com/nehasjains) either should work, 16mb version looks cheaper and has 2 usbs, can be useful.
> 
> You mentioned earlier that the board has 16mb and it might not validate!!!
> 
> Shall I go with the 16mb then ?

It is simple enough to rebuild the firmware for 16mb if it would not validate. feels strange to me buying more expensive board to match one specific binary file

nehasjains, 2026-09-04:
Ok, thank you so much. Will be ordering the 16mb flash and 8mb psram variant.

nehasjains, 2026-09-26:
I flashed HiFi-ESP32-S3 on esp32-s3 n16r8 board.

Flashing was 100% successful was able to open the squeezelite web page. But it won't update. 

What do I do now ?

Ps: using squeezelite-esp32-master-v4.3-i2s-4MFlash-16-21737.bin

anabolyc, 2026-09-26:
It will not. The S3 build is unofficial, so until maintaners would not issue S3 build in their release you're stuck with the current binary, I'm afraid

nehasjains, 2026-09-26:
Ok... so when I open the squeezelite web page, it always show "Recovery", if I choose "exit recovery", it never comes out of recovery.

Is this normal or am I missing something ?

anabolyc, 2026-09-26:
pull some logs please