sle118/squeezelite-esp32 #426: Equalizer settings only update audible when loudness setting is changed

legolas136, 2024-06-11:
### Describe the bug
When changing equalizer settings, the values for the bands only get adopted when "loudness" is enabled. strangly when loudness is disabled again, setting the band-values back to 0 works.

### Preliminary Information

1. **Firmware Version**: squeezelite-esp32-master-v4.3-I2S-4MFlash-16-21673.bin
2. Lyrion Music Server Version: 9.0.0 - 1717946032 @ Mon 10 Jun 2024 02:32:06 AM CEST

### Hardware Details

Please describe your hardware setup:

- **ESP32 Module**: ESP32 WROVER
- **DAC Chip**: PCM5102
- **Additional Hardware**: none

### NVS Settings

{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "SL-Terasse",
  "a2dp_sink_name": "TWS 20",
  "a2dp_spin": "0000",
  "actrls_config": "own_buttons",
  "airplay_name": "SL-Terasse",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "SL-Terasse",
  "autoexec": "1",
  "autoexec1": "squeezelite -o i2s -s xxx -b 500:2000 -C 30 -r 96000 -n SL-Terasse",
  "bat_config": "",
  "bt_name": "SL-Terasse",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\t\"deviceName\":\t\"SL-Terasse\",\t\"bitrate\":\t160,\t\"zeroConf\":\t1}",
  "dac_config": "model=I2S,bck=32,ws=25,do=33,mute=14:1",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "",
  "enable_airplay": "Y",
  "enable_bt_sink": "Y",
  "enable_cspot": "N",
  "equalizer": "0,0,0,0,0,0,0,0,0,0",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "SL-Terasse",
  "i2c_config": "",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "",
  "lms_ctrls_raw": "n",
  "loudness": "0",
  "metadata_config": "",
  "model_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
  "pollmx": "600",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "https://github.com/sle118/squeezelite-esp32/releases",
  "rotary_config": "",
  "set_GPIO": "",
  "sleep_config": "",
  "spdif_config": "",
  "spi_config": "mosi=15,miso=2,clk=14",
  "stats": "N",
  "target": "",
  "telnet_block": "500",
  "telnet_buffer": "40000",
  "telnet_enable": "D",
  "wifi_smode": "f"
}

<details>
<pre><code>
  Your log content here
</code></pre>
</details>

### Logs

[00:01:51.965] equalizer_set_gain:165 equalizer gain 0,0,0,0,0,0,0,0,0,0
[00:01:51.967] equalizer_process:222 equalizer deactivated
I (114050) config: configuration has some uncommitted entries
I (114050) config: Committing configuration to nvs. Locking config object.
I (114070) nvs_utilities: Value stored under key 'equalizer'
I (114070) config: Done Committing configuration to nvs.
I (119410) config: Setting changed flag config [equalizer]
I (119410) config: Updating config [equalizer]
[00:01:57.996] equalizer_set_gain:165 equalizer gain 20,0,0,0,0,0,0,0,0,0
I (120070) config: configuration has some uncommitted entries
I (120070) config: Committing configuration to nvs. Locking config object.
I (120100) nvs_utilities: Value stored under key 'equalizer'
I (120100) config: Done Committing configuration to nvs.
I (123440) config: Setting changed flag config [loudness]
I (123440) config: Updating config [loudness]
[00:02:02.031] calculate_loudness:77 loudness 2.8,1.9,1.3,0.66,0.26,0.016,-0.2,-0.45,-0.22,0.71,
[00:02:02.046] equalizer_set_loudness:187 loudness 1
[00:02:02.093] equalizer_process:222 equalizer actived
I (124100) config: configuration has some uncommitted entries
I (124100) config: Committing configuration to nvs. Locking config object.
I (124140) nvs_utilities: Value stored under key 'loudness'
I (124140) config: Done Committing configuration to nvs.
I (125470) config: Setting changed flag config [loudness]
I (125470) config: Updating config [loudness]
[00:02:04.058] calculate_loudness:77 loudness 0,0,0,0,0,0,-0,-0,-0,0,
[00:02:04.069] equalizer_set_loudness:187 loudness 0
[00:02:04.088] equalizer_process:222 equalizer actived
I (126140) config: configuration has some uncommitted entries
I (126140) config: Committing configuration to nvs. Locking config object.
I (126170) nvs_utilities: Value stored under key 'loudness'
I (126170) config: Done Committing configuration to nvs.
I (128420) config: Setting changed flag config [equalizer]
I (128420) config: Updating config [equalizer]
[00:02:07.004] equalizer_set_gain:165 equalizer gain 2,0,0,0,0,0,0,0,0,0
I (129170) config: configuration has some uncommitted entries
I (129170) config: Committing configuration to nvs. Locking config object.
I (129190) nvs_utilities: Value stored under key 'equalizer'
I (129190) config: Done Committing configuration to nvs.
I (132520) config: Setting changed flag config [equalizer]
I (132520) config: Updating config [equalizer]
[00:02:11.106] equalizer_set_gain:165 equalizer gain 1,0,0,0,0,0,0,0,0,0
I (133190) config: configuration has some uncommitted entries
I (133190) config: Committing configuration to nvs. Locking config object.
I (133230) nvs_utilities: Value stored under key 'equalizer'
I (133230) config: Done Committing configuration to nvs.
I (133450) config: Setting changed flag config [equalizer]
I (133450) config: Updating config [equalizer]
[00:02:12.034] equalizer_set_gain:165 equalizer gain 0,0,0,0,0,0,0,0,0,0
[00:02:12.075] equalizer_process:222 equalizer deactivated
I (134230) config: configuration has some uncommitted entries
I (134230) config: Committing configuration to nvs. Locking config object.
I (134260) nvs_utilities: Value stored under key 'equalizer'
I (134260) config: Done Committing configuration to nvs.

### Issue Description

1. **Observed Behavior**: The per-band values only update, when lodness value is changed. I can make any changes, the only update audible when I change the loudness value.
2. **Expected Behavior**: Update on band values should be audible as soon as I change them without changing loudness value
3. **Steps to Reproduce**: flash 16 bit fw, go to player settings -> esp32 settings change band values extremely, listen. Then change loudness setting. listen.

legolas136, 2024-06-11:
even if the changed band values are shown in the logs, there is no audible difference. only when changing the loudnes value the band values get adopted as they should.
So I have to set the new band values, change loudness to any value and then back to 0 to hear the desired result. it's a work around

sle118, 2024-06-20:
Out of curiosity, how are you changing the equalizer bands values? From the LMS plugin, I'm assuming? Also, do you think you could update the logs and insert a few comments to explain what is being done? This would be really helpful. 

Thank you!

digidocs, 2024-08-21:
@legolas136 I have a fix for this in PR #439.  If you'd like a firmware image to test to see if this resolves your issue, please let me know.

legolas136, 2024-08-22:
> @legolas136 I have a fix for this in PR #439. If you'd like a firmware image to test to see if this resolves your issue, please let me know.

Hi. thank you for solving the Bug!

Could you provide a precompiled version (I2S-4MFlash, 16 bits)?

I don't have the time to get vscode etc. working atm.

I would really appreciate testing the fix.

Thank you

digidocs, 2024-08-23:
@legolas136 Here is a 16 bit build (target: Muse speaker), let me know if this works for you:
https://github.com/digidocs/squeezelite-esp32/releases/tag/noise-burst-and-eq-update-rc1

digidocs, 2024-08-26:
@legolas136 Did this work for you?

legolas136, 2024-08-26:
I flashed the fw, but didn't get i2s DAC working. Should be a pin mapping problem. I configured the pins through nvs editor, but didn't manage to get it running.

I will re-flash the release version, make sure the pin mapping is correct and then re-test it this evening.

I'm sorry, I have 3 kids, so there is not much time to tinker.

I will replay again this evening.

legolas136, 2024-08-26:
i wired the dac as following: model=I2S,bck=32,ws=25,do=33

The (as far as i can tell) hardcoded pins in the muse fw is:
[image]

i have no chance to change the wiring as its glued together.

Could you provide a precompiled version as I2S-4MFlash, 16 bits?

digidocs, 2024-08-27:
@legolas136 Here is a build for the default I2S-4MFlash target.  I don't have a device to test this build, so fingers crossed that it works...
https://github.com/digidocs/squeezelite-esp32/releases/tag/noise-burst-and-eq-update-rc1-I2S-4MFlash

legolas136, 2024-08-27:
EQing is working in the Test-build as expected

digidocs, 2024-08-27:
Let's hold this issue open until @sle118 merges PR #439 into the main code.  At the moment, this code is only in my personal branch.

closed 2024-09-13 by philippe44