sle118/squeezelite-esp32 #429: Flash instructions

Michal-Szczepaniak, 2024-07-23:
There's no info whatsoever how to flash this thing. I assume building is optional since you provide binary so how do I flash it there's literally 0 info on that

iceman73, 2024-07-23:
https://sle118.github.io/squeezelite-esp32-installer/

Michal-Szczepaniak, 2024-07-23:
It doesn't work in every browser, and requiring browser to flash is bit meh

iceman73, 2024-07-23:
https://github.com/sle118/squeezelite-esp32/discussions/129

Michal-Szczepaniak, 2024-07-23:
So please add both to Readme under Installation secition and it will be golden

sle118, 2024-08-03:
> So please add both to Readme under Installation secition and it will be golden

You can also submit a pull request with your suggestions and I'll gladly review them. 

Thanks @iceman73

closed 2024-08-03 by sle118