sle118/squeezelite-esp32 #433: ESP32-Wrover-DEV install fails

mingratta, 2024-08-07:
[esp-web-tools-logs (14).txt](https://github.com/user-attachments/files/16521740/esp-web-tools-logs.14.txt)

### Describe the bug
When flashing the ESP32-Wrover-DEV device with Squeezelite the flash looks successful however the flash fails. 

### Hardware Details

- ESP32 Module: ESP32-WROVER-Dev
- 4MB Flash memory, and 8MB PSRAM
- Microcontroller: ESP32-WROVER-E
- Processor: Dual-core 32-bit MCU with 600 DMIPS
- Bluetooth: Bluetooth v4.2 BR/EDR and BLE

Logs Attached

### Issue Description
1. I plugged the device into the computer
2. Selected Generic/I2s  the install page. 
3. Held the boot button down and connected to serial and clicked install.
4. selected ease device and installed, no errors on install. 
5. when restarting the device its in a boot loop attached is the log.

Renber78, 2024-08-14:
Hello, do you have GPIO connected? if yes which ones?

criegit, 2024-08-14:
I actually have the same problem. I tried installers
https://sonocotta.github.io/esp32-audio-dock/
and
https://sle118.github.io/squeezelite-esp32-installer/

The first brought me usually the error very similar to the one from mingratta.
The second installer gives me a slightly different error message and also (i think) keeps resetting. (see below)
I tried two different boards of the same make. Nothing connected., no camera attached. 
Flashing and running esphome seems to work fine.

WROVER Module, holding BOOT during plugging in, then "reset Device" via console.
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6424
load:0x40078000,len:15632
load:0x40080400,len:4348
entry 0x40080688
I (27) boot: ESP-IDF v4.3.5-dirty 2nd stage bootloader
I (27) boot: compile time 21:43:45
I (27) boot: chip revision: v1.0
I (31) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32: SPI Speed      : 80MHz
I (40) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (73) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  3 recovery         factory app      00 00 00010000 00140000
I (96) boot:  4 ota_0            OTA app          00 10 00150000 002a0000
I (103) boot:  5 settings         WiFi data        01 02 003f0000 00010000
I (111) boot: End of partition table
I (115) boot: Defaulting to factory image
I (120) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=53878h (342136) map
I (223) esp_image: segment 1: paddr=000638a0 vaddr=3ffbdb60 size=04858h ( 18520) load
I (228) esp_image: segment 2: paddr=00068100 vaddr=40080000 size=07f18h ( 32536) load
I (239) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=d1744h (857924) map
I (476) esp_image: segment 4: paddr=0014176c vaddr=40087f18 size=0c8b8h ( 51384) load
I (493) esp_image: segment 5: paddr=0014e02c vaddr=400c0000 size=00068h (   104) load
I (503) boot: Loaded app from partition at offset 0x10000
I (503) boot: Disabling RNG early entropy source...
I (515) psram: This chip is ESP32-D0WD
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

.... (continues with last message indefinitely)
```

Unrelated:
Using https://sonocotta.github.io/esp32-audio-dock/ and another esp32-s3 xiao board i managed to get it installed and booted once, but it then does not boot a second time. And since i also want the BT streaming option, i decided to go to the brand new esp32 wrover modules anyway and abandonned the (faster) esp32-s3.

criegit, 2024-08-14:
can anybody confirm that the offset and length of the flashed firmware is correct?

Renber78, 2024-08-14:
ESP32-D0WD = no psram !

criegit, 2024-08-14:
ok... And based on the instructions i thought all WROVER work...

Any tips which one works? How can i identify it? Most shops Amazon/ebay only list the name but no detailed description.
And i do not need the full audio boards. i just want a simple SPDIF digital out to my stereo...

Renber78, 2024-08-14:
I use ESP32-WROVER-B or ESP32-WROVER-E which works very well. Please note, there are several variants (see expressive datasheet)

criegit, 2024-08-14:
That sounds good, but both of our modules say WROVER-E...

I know that these ESP names are confusing, but until today i thought i am able to get through it. 
Did we both just get a faked module?

Renber78, 2024-08-14:
You probably have fake ones then, yes.
To my knowledge the E & B all have PSRAM. The ID you have in your logs reminds me of an old ESP-WROOM-32.
I don't have the ID of "mingratta", so it's impossible to know if he is in the same situation as you.

Renber78, 2024-08-14:
Do you have a link or a diagram of your card?

criegit, 2024-08-14:
I just tried this script from [https://thingpulse.com/esp32-how-to-use-psram/](https://thingpulse.com/esp32-how-to-use-psram/) and  finally got it running with my home pc. (really bad Antivirus at work)
Result:
[  5033][D][main.cpp:5] setup(): Total heap: 380488
[  5039][D][main.cpp:6] setup(): Free heap: 357052
[  5045][D][main.cpp:7] setup(): Total PSRAM: 0
[  5052][D][main.cpp:8] setup(): Free PSRAM: 0

It is this amazon link:
https://www.amazon.de/dp/B0B96DKMQK

Renber78, 2024-08-14:
After reading some expressive docs, I believe it is possible to have this ID with PSRAM (but external). So this may not be the path to follow...

Renber78, 2024-08-14:
yes that's what I understood too

Renber78, 2024-08-14:
otherwise, you can flash your ESP32 with the camera example under Arduino. It will use PSRAM and you should see this in the logs. If you have the same type of error it is because the seller 'forgot' to solder it ;-)

criegit, 2024-08-14:
camera sketch via arduino has similar results. not booting up.
PIN12 will be tried in the next days.

criegit, 2024-09-04:
I got new boards from Soldered (https://soldered.com/de/produkt/dasduino-connectplus/)

They are looking really well made, price is ok and squeezelite worked immediately.

closed 2024-09-13 by philippe44