sle118/squeezelite-esp32 #436: [Question] if nothing is being streamed can it use a fallback source?

pilotkid, 2024-08-19:
Hello is there any way to make this play from a fallback source if music is not actively being streamed?

sle118, 2024-10-02:
Can you describe the scenario you have on mind?

pilotkid, 2024-10-02:
Sure

When I'm at home, I want to be able to take over a speaker and start playing music of my choosing. But when I leave, I want it to fall back to another source that can play some form of background music whether it's through an Internet radio, or another source for my animals.

sle118, 2024-10-03:
The firmware is mostly responding to requests and doesn't initiate anything on its own. This looks like something which would be better served by some home automation, for example using home assistant which has support for the Lyron media server (formerly Logitech media server).