sle118/squeezelite-esp32 #442: Plugin: cant access SqueezeESP32 settings

mockerl, 2024-08-25:
To help us resolve your issue as quickly as possible, please follow these guidelines when submitting an issue. Providing all the necessary information will save both your time and ours.

### Describe the bug
`[24-08-25` 22:55:00.0047] Slim::Networking::IO::Select::__ANON__ (130) Error: Select task failed calling Slim::Web::HTTP::processHTTP: Can't locate object method "hasLED" via package "Slim::Player::SqueezePlay" at /config/cache/InstalledPlugins/Plugins/SqueezeESP32/PlayerSettings.pm line 108.
; fh=Slim::Web::HTTP::ClientConn=GLOB(0x55d856e622b8)

[24-08-25 22:55:15.3961] Slim::Networking::IO::Select::__ANON__ (130) Error: Select task failed calling Slim::Web::HTTP::processHTTP: Can't locate object method "hasLED" via package "Slim::Player::SqueezePlay" at /config/cache/InstalledPlugins/Plugins/SqueezeESP32/PlayerSettings.pm line 108.
; `fh=Slim::Web::HTTP::ClientConn=GLOB(0x55d856c12868)`

### Preliminary Information

1. **Firmware Version**: fw: I2S-4MFlash-1673, mode: Squeezelite-ESP32
2. **Plugin Version**: v0.600

### Hardware Details

Please describe your hardware setup:

- **ESP32 Module**: ESP32 WROVER
- **Board Type**: NA
- **DAC Chip**: PCM5102
- **Additional Hardware**: NA

### NVS Settings

Follow these steps to share your NVS settings:

1. Open the web UI of your device.
2. Click on the "Credit" tab.
3. Enable the "Show NVS Editor" checkbox. This allows you to view or change the NVS configuration even when not in recovery mode.
4. Navigate to the "NVS Editor" tab.
5. Scroll to the bottom and click "Download Config".
6. Share the downloaded content here.

<details>
<pre><code>
{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "squeezelite-kitchen",
  "a2dp_sink_name": "SMSL BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "squeezelite-kitchen",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "squeezelite-kitchen",
  "autoexec": "1",
  "autoexec1": "squeezelite  -o i2s  -s  192.168.88.11 -b  500:2000 -C  30 -d  all=info  -n \"kitchen\" -Z 192000 -r \"44100-44100\"",
  "bat_config": "",
  "bt_name": "squeezelite-kitchen",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\t\"deviceName\":\t\"squeezelite-kitchen\",\t\"bitrate\":\t160}",
  "dac_config": "model=I2S,bck=33,ws=25,do=32,mute=14:0",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "",
  "enable_airplay": "N",
  "enable_bt_sink": "N",
  "enable_cspot": "N",
  "equalizer": "0,0,0,0,0,0,0,0,0,0",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "squeezelite-kitchen",
  "i2c_config": "",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "",
  "lms_ctrls_raw": "n",
  "loudness": "0",
  "metadata_config": "",
  "model_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
  "pollmx": "600",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "CONFIG_SQUEEZELITE_ESP32_RELEASE_URL",
  "rotary_config": "",
  "set_GPIO": "",
  "sleep_config": "",
  "spdif_config": "",
  "spi_config": "",
  "stats": "N",
  "target": "",
  "telnet_block": "500",
  "telnet_buffer": "40000",
  "telnet_enable": "N"
}
</code></pre>
</details>

### Logs

<details>
<pre><code>
=== START OF LOG ===
[24-08-25 22:55:00.0047] Slim::Networking::IO::Select::__ANON__ (130) Error: Select task failed calling Slim::Web::HTTP::processHTTP: Can't locate object method "hasLED" via package "Slim::Player::SqueezePlay" at /config/cache/InstalledPlugins/Plugins/SqueezeESP32/PlayerSettings.pm line 108.
; fh=Slim::Web::HTTP::ClientConn=GLOB(0x55d856e622b8)

[24-08-25 22:55:15.3961] Slim::Networking::IO::Select::__ANON__ (130) Error: Select task failed calling Slim::Web::HTTP::processHTTP: Can't locate object method "hasLED" via package "Slim::Player::SqueezePlay" at /config/cache/InstalledPlugins/Plugins/SqueezeESP32/PlayerSettings.pm line 108.
; fh=Slim::Web::HTTP::ClientConn=GLOB(0x55d856c12868)
</code></pre>
</details>

### Issue Description

1. **Observed Behavior**: When trying to access the plugin-settings an error is logged
2. **Expected Behavior**: i would expect the SqueezeESP32 settings to open
3. **Steps to Reproduce**: install plugin and try to access it via the player tab

mockerl, 2024-10-05:
Today I have updated everything to the newest versions and everything seems to work as expected. Therefore I will close this issue. Thanks for your continued development!

closed 2024-10-05 by mockerl

DavidM42, 2026-04-15:
To what version did you update everything? I am facing the same problems and can't fix them