sle118/squeezelite-esp32 #443: feature request: play mp3 file from SD card and provide uart control over it

tomliekens, 2024-09-13:
dear, to add mp3 playback from SD card, would it be best to add an extra component next to squeezeplayer/cspot/airplay/a2dp or would it be possible to reuse an existing and instruct it to play from local SD storage? many thx! Tom

sle118, 2024-10-02:
Tom, 

There are several projects dedicated to playing music from an SD card so I'm not sure about the value that this would bring to this project as I don't really see a proper use case. 

Perhaps @philippe44 could chime in?

philippe44, 2024-10-02:
Agree with @sle118, I don't see that as a value for our project, sorry

closed 2024-10-02 by sle118

tomliekens, 2024-10-02:
dear developers, thx for your great software and replies. I am modding a 1:8 scale Knight Rider (ex https://mykittbuild.com/) . I swapped the pcb with a esphome controller. Now the cool Kitt sound track and voices play via dfplayer mini on SD card, so it works fully standalone. Sqeezeplayer / Spotify connect / Airplay/ bluetooth would be a very nice addition but the primary goals is standalone functionality. Espmuse also has an sd slot so i was thinking maybe a larger user base would enjoy this functionality. I think the steps needed are to add a SD fat library and point and existing sink/decoder to the file on the SD fat, you think is a good approach or is it easier to start a sink from scratch and write to the buffer? many thx!