sle118/squeezelite-esp32 #446: AirPlay trouble on ESParagus: poor audio & "not enough space" or "packet too late"

roens, 2024-09-25:
### Describe the bug
I've got a "Loud" unit from the ESParagus folks. I'm using it for AirPlay (from Music in macOS). But it's having trouble playing consistently. In the output via telnet, I see lots of:
 - `rtp_request_resend:858 [0x3fad3c3c]: SENDTO failed (Not enough space)`

… with some lines of this mixed in:
 - `buffer_put_packet:516 [0x3fad3c3c]: packet too late seqno:10306 rtptime:1184468724 (W:10458 R:10334)`

As far as I know, no device(s) are experiencing WiFi trouble. I'm using Ruckus WiFi6-capable APs, with a 10GbE router running OPNsense.

### Preliminary Information

1. **Firmware Version**: I2S-4MFlash-1681
2. **Plugin Version**: Mode: Squeezelite-ESP32

### Hardware Details

Please describe your hardware setup:

- **ESP32 Module**: ESParagus Loud (ESP32-WROVER-IE)
- **Board Type**: Loud
- **DAC Chip**: 2x MAX98357 (1 per channel, I2C)
- **Additional Hardware**: (none)

### NVS Settings

Settings, afaik, are default, aside from setting device name and disabling modes I do not intend to use (bluetooth & spotify).

<details>
<pre><code>
{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "ESParagus-Kitchen",
  "a2dp_sink_name": "SMSL BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "ESParagus-Kitchen",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "ESParagus-Kitchen",
  "autoexec": "1",
  "autoexec1": "squeezelite -o i2s -n ESParagus-Kitchen -b 500:2000 -C 30 -d all=info",
  "bat_config": "",
  "bt_name": "ESParagus-Kitchen",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\t\"deviceName\":\t\"ESParagus-Kitchen\",\t\"bitrate\":\t160,\t\"zeroConf\":\t1}",
  "dac_config": "model=I2S,bck=26,ws=25,do=22",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "",
  "enable_airplay": "Y",
  "enable_bt_sink": "N",
  "enable_cspot": "N",
  "equalizer": "",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "ESParagus-Kitchen",
  "i2c_config": "",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "WS2812,length=1,gpio=33,scale=100",
  "lms_ctrls_raw": "n",
  "loudness": "0",
[... 708 lines cut ...]

[00:06:37.235] output_thread_i2s:518 Output state is 0
[00:06:45.075] slimproto:989 unable to connect to server 6
[00:06:48.678] handle_rtsp:444 [0x3fad0188]: received TEARDOWN
[00:06:48.765] rtp_thread_func:794 [0x3fad49bc]: terminating
[00:06:48.768] cleanup_rtsp:700 [0x3fad0188]: Remote search thread aborted
[00:06:48.768] raop_sink_cmd_handler:310 Stop
[00:06:48.770] handle_rtsp:667 [0x3fad0188]: responding:
RTSP/1.0 200 OK
Audio-Jack-Status: connected; type=analog
CSeq: 15

[00:06:50.076] discover_server:844 sending discovery 5
[00:06:51.169] rtsp_thread:413 RTSP close 56
[00:06:55.075] discover_server:844 sending discovery 4
[00:07:00.075] discover_server:844 sending discovery 3
[00:07:05.075] discover_server:844 sending discovery 2
[00:07:10.075] discover_server:844 sending discovery 1</code></pre>
</details>

### Issue Description

1. **Observed Behavior**: In this instance, I did not experience the former issue where audio ceased coming out of the speakers and the device became unavailable to my audio source: Music app on macOS. Instead, this time, audio persisted, and it played until I stopped playing. However, audio became noticeably choppy (split-second drops in audio, like a high speed pulse).
2. **Expected Behavior**: The device should remain available to send audio to via AirPlay, and the audio quality out of the speakers should be "good" (consistent, without dropping any data).
3. **Steps to Reproduce**: Reboot the ESP32 device, connect to it in Music via AirPlay, play some audio.

philippe44, 2024-09-29:
It's happens when sending audio in UDP frames. There is a backlog that is too big in the esp32 IP stack (lwIP) and packets then are dropped. Unfortunately, there are not many options to increase the backlog size as the type of memory available for this is pretty small. It does not mean that it would be a realtime issue on other systems as the IP stack would continue to accept packets until the WiFi access is granted (or the IP stack is scheduled), but here we can't.

sle118, 2024-10-03:
@roens can you see if your router has a band steering option? I've had cases in the past where such option would not play well with the esp32 in terms of stability. Also check for TX/RX retries if the router gives you such insights. 

You could also try to connect your Mac using a cable instead of using wifi. 

Having a capable router is good but might not isolate you from greedy neighbors using the 2.4G at 40Mhz bandwidth which can cause spikes of performance degradation.

roens, 2024-12-01:
From what I've been able to research, the Ruckus Unleashed APs I use do engage in "band steering". I run two WiFi6 capable Ruckus H550 APs along with an R510 with its radios disabled (as mesh "master"), all running their Unleashed firmware.

@sle118, Are you referring to potential WiFi issues with my laptop providing AirPlay stream, or my ESParagus running this repo's firmware? When I'm streaming, typically I do so from my MacBook Pro (M2 Max) to both my Yamaha receiver (supports AirPlay 2) and this ESParagus. My laptop and ESParagus are both on WiFi, the receiver is on ethernet. Audio to the receiver is always flawless, whether I'm streaming to just it or it and ESParagus. My impression is the trouble is within the ESP32 capabilities of the ESParagus.

Life got complicated as it does, and in the meantime I've set up an RPi (running `shairport-sync`) in the location where I had been trying to use the ESParagus. I'll have to circle back to testing out the different Unleashed settings after I set up my ESParagus again.

sle118, 2024-12-09:
@roens I was referring to band steering not being friendly to the ESP32 as the ESP32 only supports 2.4Ghz wifi bands. Some users have had success with setting up a separate wifi where band steering is disabled (e.g. where the router doesn't try to force clients into the 5Ghz band).

roens, 2024-12-12:
I have yet to implement the fix outlined, but it seems this is caused by a hardware issue:
- https://www.crowdsupply.com/sonocotta/esparagus-media-center/updates/a-software-update-an-upcoming-hardware-variant-and-a-repair-guide-for-an-issue-present-in-some-loud-esparagus-units

Essentially: the USB-serial chip is powered off the 5V rail, which results in passing about 4.2V back to the ESP32's RTS and DTR pins, unless a computer is connected to that USB-C port. This causes _"audio stuttering or breaking up, even close to the router"_. The fix is to disconnect the power pin of that USB-serial chip from 5V and connect it to the 3.3V supply. I might manage to apply the fix to one or both of my ESParagus units later this week.

sle118, 2024-12-12:
I am glad you finally found the cause of your troubles. Hoping you can finally enjoy your ESParagus boards!

closed 2024-12-12 by sle118

roens, 2024-12-12:
I've just bodged the _ESParagus Loud_ that I was having trouble with, leading to creation of this ticket. I can confirm the hardware fix described on the Crowdsupply update linked above did fix the problem!