sle118/squeezelite-esp32 #448: GPIO shortcut to m3u playlist

fensoft, 2024-10-02:
Hi,

I want to build a boombox for my daughter with only 1 button to play an m3u until the end, idealy 2 buttons (1 for kid music 1 for story told).

Any idea how to do this ?

Regards,
fensoft

sle118, 2024-10-03:
I recommend you read the following: 
https://github.com/sle118/squeezelite-esp32?tab=readme-ov-file#buttons

You could attach two buttons and set actions to trigger presets (e.g. BCTRLS_PS1 ) and perhaps a long press to stop, play the next, etc. Then in LMS, you navigate to the player configuration and bind the preset to what you want. 

I've never done this, but maybe someone else can give you more details (@philippe44 ?)

fensoft, 2024-10-03:
squeezelite command used in presets are done localy, right ?
i.e. I can play http hosted playlist directly ?

philippe44, 2024-10-03:
Our squeezeesp32 sends button_1 ... 10 to LMS which then will take any action you configured it to do

closed 2024-12-16 by sle118