sle118/squeezelite-esp32 #453: -s disable option seems not working

ferrarid61, 2024-11-03:
### Describe the bug
I'm using a sueezelite-esp32 on esp32  wrover E CPU. I installed it with the web installer. I want to avoid the LMS connection bat the --s disable option in autoexec1 seems not have any effect ad the cpu reboots prediodically

### Preliminary Information

1. **Firmware Version**:  fw: I2S-4MFlash-1681, mode: Squeezelite-ESP32
2. **Plugin Version**: Mention the version of the plugin installed on your LMS (Logitech Media Server).

### Hardware Details

Please describe your hardware setup:

- **ESP32 Module**: ESP32 WROVER E.
- **Board Type**: If applicable, e.g., ESP32 audio kit, etc.
- **DAC Chip**: PMC5201A.
- **Additional Hardware**: Include details about any other hardware like rotary controls, buttons, screens (SPI, I2C), Ethernet, IO expansion, etc.

### NVS Settings

Follow these steps to share your NVS settings:

1. Open the web UI of your device.
2. Click on the "Credit" tab.
3. Enable the "Show NVS Editor" checkbox. This allows you to view or change the NVS configuration even when not in recovery mode.
4. Navigate to the "NVS Editor" tab.
5. Scroll to the bottom and click "Download Config".
6. Share the downloaded content here.

<details>
<pre><code>
{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "cambridge",
  "a2dp_sink_name": "cambridge",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "cambridge",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "cambridge",
  "autoexec": "1",
  "autoexec1": "squeezelite  -o i2s  -s  disable -b  500:2000 -C  30 -d  all=info ",
  "bat_config": "",
  "bt_name": "cambridge",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\"deviceName\":\"cambridge\",\"bitrate\":160,\"zeroConf\":1}",
  "dac_config": "model=,bck=33,ws=25,do=32",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "",
  "enable_airplay": "1",
  "enable_bt_sink": "1",
  "enable_cspot": "1",
  "equalizer": "",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "cambridge",
  "i2c_config": "",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "",
  "lms_ctrls_raw": "n",
[... 236 lines cut ...]
I (5619) wifi:connected with XXXXXXX, aid = 21, channel 1, BW20, bssid = 10:13:31:b8:80:b7
I (5619) wifi:security: WPA2-PSK, phy: bgn, rssi: -62
I (5629) wifi:pm start, type: 1

I (5639) network: Got an IP address from interface Wifi. IP=192.168.1.203, Gateway=192.168.1.254, NetMask=255.255.255.0, Address was changed
I (5649) esp_netif_handlers: sta ip: 192.168.1.203, mask: 255.255.255.0, gw: 192.168.1.254
I (5659) esp_app_main: Network connected and mDNS initialized with cambridge
[00:00:04.245] raop_sink_start:178 starting Airplay for ip 192.168.1.203 with servicename cambridge
[00:00:04.248] raop_create:201 starting mDNS with A842E3AE6EEC@cambridge
I (5689) cspot: starting Spotify on host cambridge
[00:00:04.294] I Shim.cpp:344: CSpot instance service name cambridge (id 142137fd329622137a1490160000000435587463)
[00:00:04.296] I Shim.cpp:331: ZeroConf mode (port 80)
I (5769) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[00:00:06.965] discover_server:844 sending discovery 24
I (10459) httpd_handlers: serving /messages.json to peer 192.168.1.181 port 45256
[00:00:11.971] discover_server:844 sending discovery 23
[00:00:16.986] discover_server:844 sending discovery 22
</code></pre>
</details>

### Issue Description

1. **Observed Behavior**: aqueezeite still try to contact LMS server and after e while reboots.
2. **Expected Behavior**: working with BT Airplay & Spotify.
3. **Steps to Reproduce**: Provide a step-by-step guide on how to replicate the issue.

moesfeld, 2024-11-15:
It needs to be "-s -disable" not "-s disable". Notice the missing "-"

ferrarid61, 2024-11-15:
> It needs to be "-s -disable" not "-s disable". Notice the missing "-"

Thank you so much!!

closed 2024-11-15 by ferrarid61