sle118/squeezelite-esp32 #459: ESP32-WROVER-IB LMS Player/ Bluetooth/ Airplay not working

Charliestonedassgueniapig, 2024-12-25:
To help us resolve your issue as quickly as possible, please follow these guidelines when submitting an issue. Providing all the necessary information will save both your time and ours.

### Describe the bug
When installing squeezeeliteesp32 onto the device, it works you can configure ect but you cannot strream to it by any way.

### Preliminary Information

1. **Firmware Version**: I2S-4MFlash-1681
2. **Plugin Version**: /

### Hardware Details

Please describe your hardware setup:

- **ESP32 Module**: ESP32-WROVER-IB 
- **Board Type**: /
- **DAC Chip**: PCM5102
- **Additional Hardware**: /
### NVS Settings

Follow these steps to share your NVS settings:

1. Open the web UI of your device.
2. Click on the "Credit" tab.
3. Enable the "Show NVS Editor" checkbox. This allows you to view or change the NVS configuration even when not in recovery mode.
4. Navigate to the "NVS Editor" tab.
5. Scroll to the bottom and click "Download Config".
6. Share the downloaded content here.

<details>
<pre><code>
  {
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "Squeezelite",
  "a2dp_sink_name": "SMSL BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "ESP32-AirPlay-aea06c",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "squeezelite-aea06c",
  "autoexec": "1",
  "autoexec1": "squeezelite -o I2S -b 500:2000 -d all=info -C 30 -W -n squeezelite-aea06c",
  "bat_config": "",
  "bt_name": "ESP32-BT-aea06c",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\"deviceName\":\"squeezelite-aea06c\",\"bitrate\":160}",
  "dac_config": "",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "",
  "enable_airplay": "1",
  "enable_bt_sink": "1",
  "enable_cspot": "1",
  "equalizer": "",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "squeezelite-aea06c",
  "i2c_config": "",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "",
[... 208 lines cut ...]
I (111141) network_wifi: Starting wifi
I (111141) esp_app_main: Network connected and mDNS initialized with squeezelite-aea06c
I (111151) squeezelite-ota: sending LMS discovery for OTA Update
I (111261) squeezelite-ota: got response from: 192.168.178.190:3483 - EJSON9000CLIP9090
I (111261) network_status: LMS IP: 192.168.178.190, hport: 9000, cport: 9090
I (111281) esp_app_main: Network connected and mDNS initialized with squeezelite-aea06c
I (111291) squeezelite-ota: sending LMS discovery for OTA Update
I (111351) squeezelite-ota: got response from: 192.168.178.190:3483 - EJSON9000CLIP9090
I (111351) network_status: LMS IP: 192.168.178.190, hport: 9000, cport: 9090
I (1430801) httpd_handlers: serving / to peer 192.168.178.199 port 23256
I (1431311) httpd_handlers: serving /config.json to peer 192.168.178.199 port 24280
I (1431371) httpd_handlers: serving /commands.json to peer 192.168.178.199 port 30424
I (1432831) network_wifi: Initiating wifi network scan
W (1435691) httpd_txrx: httpd_sock_err: error in send : 104
W (1435691) httpd_txrx: httpd_sock_err: error in recv : 104
I (1435741) httpd_handlers: serving /messages.json to peer 192.168.178.199 port 33496
=== END OF LOG ===   
</code></pre>
</details>

### Issue Description

1. **Observed Behavior**: I think there is no response from LMS
2. **Expected Behavior**: I assume there should be a response from LMS like in other logs that I have seen from other people.
3. **Steps to Reproduce**: Install squeezelite onto ESP32-WROVER-IB--> connect to Wifi --> try to configure settings or wait that it is found by LMS and discoverble by bluetooth and Airplay ect.

sle118, 2024-12-26:
You are in recovery mode, likely. 

Open the web interface and you should see a button at the bottom to exit this mode which comes by default.

Let us know how that goes

Charliestonedassgueniapig, 2024-12-27:
> You are in recovery mode, likely.
> 
> Open the web interface and you should see a button at the bottom to exit this mode which comes by default.
> 
> Let us know how that goes

Thank you for your help! It worked and now I can use my ESP like intended. Pardon me if this problem was to rash but I have not seen any suggestion to such a step anywhere (maybe I had overlooked something).

sle118, 2024-12-28:
> Thank you for your help! It worked and now I can use my ESP like intended. Pardon me if this problem was to rash but I have not seen any suggestion to such a step anywhere (maybe I had overlooked [...]

Enjoy!

closed 2024-12-28 by sle118