sle118/squeezelite-esp32 #461: Diagnosing problem with SPDIF/Audio drop-outs; wifi/buffer starvation ?

VincentGijsen, 2024-12-27:
### Describe the bug
multiple drop-outs of music (SPDIF), via LMS, druing a single song (e.g. 8/5minutes); seemingly 'random' .
wifi seems 'stable' , yet perhaps the cruel pit? 

basically hopping to get some pointers as to what could be the issue why the sync between the receiver and Squeezelite is lost. using my (clone-probe) ,and decoding the SPDIF signal reveals that the SPDIF interface occasionally drops out few millisecs (in hindsight, decoding of SPDIF was not needed as the gasp are quite significant);

for the LOG-files i enabled also the status; but this behaviour is also observed IF
* STATS are disabled
* TELNET /serial logging is disabled

### Preliminary Information

1. **Firmware Version**: fw: I2S-4MFlash-1681, mode: Squeezelite-ESP32
2. **Plugin Version**: v0.601
3. **Wireless**: 77%
4. set DHCP leastime to 1day
5. Wifi-group-key updates (wpa2); set to 1D
6. Wifi-DKIM disabled.
7. Tried force transcoding everything into PCM instead of OGG; issue persists.

### Hardware Details
- ESP32 WROVER, (ALI)
- 'poor-mans SPDIF 2x R + C'  [GPIO32]
- Toslink (variant), same IO-pin [GPIO32]
- Onkyo receiver TX414
- I2C 128/64 display

### NVS Settings

<details>
<pre><code>
{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "squeezelite-een",
  "a2dp_sink_name": "Squeezelite ESP32 BT",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "squeezelite-een",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "squeezelite-een",
  "autoexec": "1",
  "autoexec1": "squeezelite  -o spdif  -b  500:2000 -c  pcm -C  30 -d  all=info ",
  "bat_config": "",
  "bt_name": "squeezelite-een",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\t\"deviceName\":\t\"squeezelite-een\",\t\"bitrate\":\t160,\t\"zeroConf\":\t1}",
  "dac_config": "",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "I2C,width=128,height=64,address=60,driver=SSD1306,HFlip,VFlip",
  "enable_airplay": "N",
  "enable_bt_sink": "N",
  "enable_cspot": "N",
  "equalizer": "2,0,0,0,0,0,0,0,0,0",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "squeezelite-een",
  "i2c_config": "scl=22,sda=21,speed=400000,port=0",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "",
[... 480 lines cut ...]
[00:10:00.356] i2s_stats:664               |  (bytes) |  (bytes) |        (bytes) |     |                |
[00:10:00.356] i2s_stats:665               +----------+----------+----------------+-----+----------------+
[00:10:00.356] i2s_stats:666         stream|    511999|    445856|          507478|   99|             108|
[00:10:00.362] i2s_stats:667         output|   1949696|   1921024|         1937976|   94|             108|
[00:10:00.362] i2s_stats:668               +----------+----------+----------------+-----+----------------+
[00:10:00.363] i2s_stats:669       received|      2048|      2048|            2048|  100|             108|
[00:10:00.364] i2s_stats:670               +----------+----------+----------------+-----+----------------+
[00:10:00.364] i2s_stats:671
[00:10:00.364] i2s_stats:672               ----------+----------+-----------+-----------+
[00:10:00.364] i2s_stats:673               max (us)  | min (us) |   avg(us) |  count    |
[00:10:00.365] i2s_stats:674               ----------+----------+-----------+-----------+
[00:10:00.365] i2s_stats:675  Buffering(us)      9319|      1203|       2064|        108|
[00:10:00.365] i2s_stats:676    i2s tfr(us)     95552|     35241|      44825|        107|
[00:10:00.365] i2s_stats:677               ----------+----------+-----------+-----------+
</code></pre></details>

### Issue Description

1. **Observed Behavior**: screen, audio plays; but occasional drop-outs on the SPDIF output on the receiver
2. **Expected Behavior**: Flawless playback
4. **Steps to Reproduce**: flash the ESP; connect SPDIF; hardware, configure accordingly; audio drops out occationally.
hoping to find some pointers where to look e.g.
1) does the buffer starvate / underruns?
2) other metrics i can look at/enable to drill-down into the issue
3) is there a posiblity to log an error when buffer-underrund happens or wifi reconnects / is such info perhaps already 'staring me in the face' but i've mist it?

sle118, 2024-12-28:
Hi @philippe44 . You are more familiar with the audio core and task contention stuff. Maybe this will ring a bell

philippe44, 2024-12-28:
This is unfortunately an issue that I spike my to happen with spdif. I assume this is 26 bits mode, not 32. Do you use the equalizer? What is the audio format and sample rate?

sle118, 2024-12-28:
From the NVS, it does look like there's an equalizer going on
`"equalizer": "2,0,0,0,0,0,0,0,0,0",`. There's also an SSD1306

VincentGijsen, 2024-12-31:
Yes, eq mode is activated from memory all set to 0 though, but I presume with zero scalars the logic still fires. I’ll try without eq enabled later today to assess that is a workaround.

sle118, 2025-01-01:
32 bits version will be worse if anything.

Also, equalizer settings are received from LMS, which is why we don't expose them directly in the UI. So make sure you set all bands to zero on the LMS config page for the device rather than in the NVS config.

VincentGijsen, 2025-01-02:
> 32 bits version will be worse if anything.
> 
> Also, equalizer settings are received from LMS, which is why we don't expose them directly in the UI. So make sure you set all bands to zero on the LMS config page for the device rather than in the NVS config.
> 
> Not sure if it could be a solution or not, but if I remember the ESP32-S3 with Octal psram has a lot more memory bandwidth. Maybe it would be a better fit for bit banging spdif like we do (abusing the i2s hardware and DMA).
> 
> Stability is part of the reason why I designed my audio boards using a dedicated i2s DAC capable of outputting spdif and analog in parallel.

mm that would be a pitty if the bitbanging coded cannot be improved somehow w.r.t the realtime-behaviour of scheduler, I imagine the bit-banging eats quite some cycles away with is more critical to discipline the scheduler of freertos in such a way its never too late., I fear my coding skills lack to comprehend and support possible improvement.
as means for (temp fix) I looked for some I2S -> coax/toslink boards; the Wm8805 based (15Euro on Ali) seem to deliver based on their description, too bad its 2x the price of the ESP32 itself though, therefore I so frond of the software based  SPDIF mode :)
Any (small) recommendations perhaps (premade boards)?

In mean-time happy to support/test *with logic analyser, if that's usefull)

Off topic: are there any plans to support USBdacs btw , (on an ESP32-S3 host), my dac has UAC2; I guess then a rasperry pi with USB & squeezelite is more economically viable coding/wise. basically you cough up 192byte of pcm (singed) data/frame; I did that once on a STM32, generated from the SOF (USB event). a-sync USB uac mode would be supported; but I guess i'm digressing :)

philippe44, 2025-01-02:
I don't have a plan for USB DACs. WRT to tweaking the scheduler, I've already spent a fair bit of time doing fine grain tweaks, CPU tasks allocation, buffer optimization and many other things. The spdif implementation is really pushing the envelope as it hijacks an i2s port to make it behave like spdif. But the result is super high sampling rate and 32 bits words. So it's pretty CPU intensive. The issue behind the clicks is that the DMA buffers for i2s are being starve due to the scheduler being busy. From that pov, it's not a pure load issue. In case you have not tried that, and it comes back to me as I'm writing this comment, make sure you have disabled traces and logs. The problem happens when the logger grabs the tasks states. During that time, the scheduler is locked. On esp32s3 this never happens but on regular esp32, you can make that almost disappear if you do not active any trace or log.

PR77, 2026-08-20:
@VincentGijsen did you end up resolving your issue with the bit banging solution?