sle118/squeezelite-esp32 #465: cspot seems to be stuck on I2S-4MFlash-1681

MrNonoss, 2025-01-15:
Hi there, many thanks for this great media service.

I am running into an issue on the [Esparagus Medialink hardware](https://github.com/sonocotta/esparagus-media-center/issues/18), where cspot seems to be stuck.

It was working previously, and used to stream music from Spotify, via a smartphone. 
Suddenly it stopped working correctly. I noticed different issues:
1. The device is correctly shown in the list of available players within the Spotify app, but no connection can occur to the speaker
2. Or the speaker is not seen in the list of available players.
3. Sometimes it works for a few seconds then stops

Rebooting the device do not fix the issues, but we may go from one issue to the other

**Firmware**: I was using a previous version of I2S-4MFlash-xxxx and upgraded to I2S-4MFlash-1681, hopping for a fix, but it is even worse now.
**Hardware**: the Esparagus Medialink is using a **ESP32-WROVER-N16R8**, and a **PCM5100A DAC** 

### NVS Settings

<details>
<pre><code>
{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "esparagus-Billard",
  "a2dp_sink_name": "SMSL BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "esparagus-Billard",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "esparagus-Billard",
  "autoexec": "1",
  "autoexec1": "squeezelite  -o i2s  -n  esparagus-billard -b  500:2000 -C  30 -d  all=info ",
  "bat_config": "",
  "bt_name": "esparagus-Billard",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\"deviceName\":\"esparagus-Billard\",\"bitrate\":160,\"zeroConf\":1}",
  "dac_config": "model=I2S,bck=26,ws=25,do=22",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "",
  "enable_airplay": "Y",
  "enable_bt_sink": "N",
  "enable_cspot": "Y",
  "equalizer": "",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "esparagus-Billard",
  "i2c_config": "",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "",
  "lms_ctrls_raw": "n",
  "loudness": "0",
  "metadata_config": "",
  "model_config": "",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
  "pollmx": "600",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "CONFIG_SQUEEZELITE_ESP32_RELEASE_URL",
  "rotary_config": "",
[... 439 lines cut ...]
[00:00:07.025] process:578 unhandled vers

</code></pre>
</details>

I spot this specific error:
```
E (157774) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (157774) task_wdt:  - IDLE (CPU 1)
E (157774) task_wdt: Tasks currently running:
E (157774) task_wdt: CPU 0: IDLE
E (157774) task_wdt: CPU 1: cspot_player
E (157774) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x401E2C56:0x3FFBED40 0x40084889:0x3FFBED60 0x40270227:0x3FFBD5B0 0x400DD261:0x3FFBD5D0 0x4027E6CB:0x3FFBD5F0 

E (157774) task_wdt: Print CPU 1 backtrace

Backtrace:0x4008F5DD:0x3FFBF570 0x40084889:0x3FFBF590 0x4026CE77:0x3FB01AD0 0x400F7A45:0x3FB01AF0 0x400F8258:0x3FB01B10 0x400F8335:0x3FB01B40 0x40109871:0x3FB01B60 0x401098A5:0x3FB01BD0 0x4015CD00:0x3FB01C00 0x401088AA:0x3FB01CB0 
```

Unfortunately, ESP is really new for me and I am not sure how to fix this.
Any help would be apreciated,

Many thanks

fabian727, 2025-04-22:
I was unsure, whether I have setup issues, therefore I opened a [discussion](https://github.com/sle118/squeezelite-esp32/discussions/491)

fabian727, 2025-04-22:
maybe some duplication also out of this repo: https://github.com/sonocotta/esparagus-media-center/issues/18