sle118/squeezelite-esp32 #466: Stops playing after a few days, reboot fixes it

elektrinis, 2025-01-17:
### Describe the bug
I'm running this on a Aliexpress A1S board, default settings. It works well, there's plenty of sound with quality speakers.
I use it as smart speaker with Home Assistant, mostly for internet radio.
Problem: it works well for 3-4 days, and then audio starts to drop out: I start streaming to it, music plays for a couple of seconds and stops. Or playing does not start at all. It is continuing to report status that music is still playing, indefinitely. I then visit the web interface and click on Restart. It takes some time to come up again, but then it works properly for another few days, until problems resume.
Can this be some kind of memory leak?

### Preliminary Information

1. **Firmware Version**: I2S-4MFlash-1681
2. **Plugin Version**: Lyrion Music Server Version: 9.0.0 - 1732822968

### Hardware Details

ESP32A1S V2.2+ variant 1 (ES8388)

### NVS Settings

(default, did not change anything) 
```
{
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "Squeezelite",
  "a2dp_sink_name": "SMSL BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "ESP32-AirPlay-1c4034",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "squeezelite-1c4034",
  "autoexec": "1",
  "autoexec1": "squeezelite -o I2S -b 500:2000 -d all=info -C 30 -W -n \"squeezelite-1c4034\"",
  "bat_config": "",
  "board_model": "ESP32-A1s",
  "bt_name": "ESP32-BT-1c4034",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\"deviceName\":\"squeezelite-1c4034\",\"bitrate\":160,\"zeroConf\":1}",
  "dac_config": "model=ES8388,bck=27,ws=25,do=26,sda=33,scl=32,di=35,i2c=16",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "",
  "enable_airplay": "1",
  "enable_bt_sink": "1",
  "enable_cspot": "1",
  "equalizer": "",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "squeezelite-1c4034",
  "i2c_config": "",
  "jack_mutes_amp": "n",
  "led_brightness": "",
  "led_vu_config": "",
  "lms_ctrls_raw": "n",
  "loudness": "0",
  "metadata_config": "",
  "model_config": "ESP32A1S V2.2+ variant 1 (ES8388)",
  "ota_erase_blk": "249856",
  "ota_prio": "6",
  "ota_stack": "10240",
  "pollmin": "15",
  "pollmx": "600",
  "rel_api": "https://api.github.com/repos/sle118/squeezelite-esp32/releases",
  "release_url": "CONFIG_SQUEEZELITE_ESP32_RELEASE_URL",
  "rotary_config": "",
  "set_GPIO": "21=amp,22=green:0,39=jack:0",
  "sleep_config": "",
  "spdif_config": "",
  "spi_config": "",
  "stats": "n",
  "target": "",
  "telnet_block": "500",
  "telnet_buffer": "40000",
  "telnet_enable": ""
}
```

### Logs

Was not able to obtain logs without fresh reboot, this takes days to manifest. Will wait for this to happen again and pull logs then.

### Issue Description

1. **Observed Behavior**: stops playing after few days
2. **Expected Behavior**: obvious
3. **Steps to Reproduce**: Use for a few days as internet radio.

elektrinis, 2025-01-18:
Log below:

<details>
<pre><code>
[00:00:11.986] discover_server:844 sending discovery 23
[00:00:11.987] discover_server:848 error sending discovery
I (18101) network_handlers: Timer: Wifi Polling timeout 
I (18101) network_wifi: Wifi Connecting to elektrinis2G...
[00:00:16.986] discover_server:844 sending discovery 22
[00:00:16.987] discover_server:848 error sending discovery
I (19791) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (19791) wifi:state: init -> auth (b0)
I (19791) wifi:state: auth -> assoc (0)
I (19801) wifi:state: assoc -> run (10)
I (19811) wifi:connected with elektrinis2G, aid = 5, channel 1, 40U, bssid = f2:48:b8:9b:5e:26
I (19811) wifi:security: WPA2-PSK, phy: bgn, rssi: -53
I (19821) wifi:pm start, type: 1

I (19831) network: Got an IP address from interface Wifi. IP=192.168.69.154, Gateway=192.168.69.1, NetMask=255.255.255.0, Address was changed
I (19851) esp_netif_handlers: sta ip: 192.168.69.154, mask: 255.255.255.0, gw: 192.168.69.1
I (19851) esp_app_main: Network connected and mDNS initialized with squeezelite-1c4034
[00:00:18.447] raop_sink_start:178 starting Airplay for ip 192.168.69.154 with servicename ESP32-AirPlay-1c4034
[00:00:18.460] raop_create:201 starting mDNS with C4DEE21C4034@ESP32-AirPlay-1c4034
I (19891) cspot: starting Spotify on host squeezelite-1c4034
[00:00:18.505] I Shim.cpp:344: CSpot instance service name squeezelite-1c4034 (id 142137fd329622137a1490160000003573715504)
[00:00:18.518] I Shim.cpp:331: ZeroConf mode (port 80)
I (19971) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[00:00:21.986] discover_server:844 sending discovery 21
[00:00:21.992] discover_server:856 got response from: 192.168.69.4:3483
[00:00:21.993] slimproto:964 connecting to 192.168.69.4:3483
[00:00:22.000] slimproto:1011 connected
[00:00:22.001] sendHELO:153 mac: c4:de:e2:1c:40:34
[00:00:22.002] sendHELO:155 cap: Model=squeezeesp32,AccuratePlayPoints=1,HasDigitalOut=1,HasPolarityInversion=1,Balance=1,Depth=32,Firmware=v1.0-1681-32,ModelName=SqueezeESP32,MaxSampleRate=96000,alc,aac,ogg,ops,ogf,flc,wav,aif,pcm,mp3
[00:00:22.029] notify:228 notified server 192.168.69.4 hport 9000 cport 9090
I (23441) network_status: LMS IP: 192.168.69.4, hport: 9000, cport: 9090
[00:00:22.042] process_dsco:483 got DSCO, switching from id 101 to 12
[00:00:22.043] slimproto_run:625 error reading from socket: closed
[00:00:22.161] slimproto:1011 connected
[00:00:22.162] sendHELO:153 mac: c4:de:e2:1c:40:34
[00:00:22.162] sendHELO:155 cap: Model=squeezeesp32,AccuratePlayPoints=1,HasDigitalOut=1,HasPolarityInversion=1,Balance=1,Depth=32,Firmware=v1.0-1681-32,ModelName=SqueezeESP32,MaxSampleRate=96000,alc,aac,ogg,ops,ogf,flc,wav,aif,pcm,mp3
[00:00:22.190] notify:228 notified server 192.168.69.4 hport 9000 cport 9090
I (23611) network_status: LMS IP: 192.168.69.4, hport: 9000, cport: 9090
I (24151) httpd_handlers: serving /status.json to peer 192.168.69.222 port 30151
I (36261) httpd_handlers: serving /status.json to peer 192.168.69.222 port 33991
[00:00:59.522] decode_flush:251 decode flush
[00:00:59.523] output_flush:446 flush output buffer (full)
[00:00:59.524] decode_flush:251 decode flush
[00:00:59.525] output_flush:446 flush output buffer (full)
[00:01:00.346] codec_open:283 codec open: 'm'
[00:01:00.348] stream_sock:660 connecting to 185.33.21.112:80

[00:01:01.261] stream_thread:415 headers: len: 694

[00:01:01.428] stream_thread:468 icy meta: len: 48
StreamTitle='MobyMusic - Whispering Wind';
[00:01:01.781] output_thread_i2s:518 Output state is 1
[00:01:01.785] output_thread_i2s:585 Restarting I2S.
[00:01:01.844] mad_decode:247 setting track_start
[00:01:01.925] _output_frames:65 start buffer frames: 9839
[00:01:01.927] _output_frames:150 track start sample rate: 44100 replay_gain: 0
[00:01:01.929] output_thread_i2s:518 Output state is 2
[00:01:10.893] stream_thread:468 icy meta: len: 80
StreamTitle='Ravi Costes - Lion Sleeps - Cafe Groovera Hotel Bar Del Mar Mix';
I (156871) httpd_handlers: serving /messages.json to peer 192.168.69.222 port 42951
[00:02:38.702] output_thread_i2s:518 Output state is 0
I (188251) httpd_handlers: serving /messages.json to peer 192.168.69.222 port 43719
I (188311) httpd_handlers: serving /status.json to peer 192.168.69.222 port 44231
[00:03:08.676] decode_flush:251 decode flush
[00:03:08.725] output_flush:446 flush output buffer (full)
[00:03:08.730] codec_open:283 codec open: 'm'
[... 223 lines cut ...]
[00:00:04.361] raop_sink_start:178 starting Airplay for ip 192.168.69.154 with servicename ESP32-AirPlay-1c4034
[00:00:04.374] raop_create:201 starting mDNS with C4DEE21C4034@ESP32-AirPlay-1c4034
I (5801) cspot: starting Spotify on host squeezelite-1c4034
[00:00:04.413] I Shim.cpp:344: CSpot instance service name squeezelite-1c4034 (id 142137fd329622137a1490160000003573715504)
[00:00:04.426] I Shim.cpp:331: ZeroConf mode (port 80)
I (5851) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[00:00:06.986] discover_server:844 sending discovery 24
[00:00:06.992] discover_server:856 got response from: 192.168.69.4:3483
[00:00:06.994] slimproto:964 connecting to 192.168.69.4:3483
[00:00:07.011] slimproto:1011 connected
[00:00:07.012] sendHELO:153 mac: c4:de:e2:1c:40:34
[00:00:07.013] sendHELO:155 cap: Model=squeezeesp32,AccuratePlayPoints=1,HasDigitalOut=1,HasPolarityInversion=1,Balance=1,Depth=32,Firmware=v1.0-1681-32,ModelName=SqueezeESP32,MaxSampleRate=96000,alc,aac,ogg,ops,ogf,flc,wav,aif,pcm,mp3
[... cut ...]
[00:00:07.193] sendHELO:153 mac: c4:de:e2:1c:40:34
[00:00:07.193] sendHELO:155 cap: Model=squeezeesp32,AccuratePlayPoints=1,HasDigitalOut=1,HasPolarityInversion=1,Balance=1,Depth=32,Firmware=v1.0-1681-32,ModelName=SqueezeESP32,MaxSampleRate=96000,alc,aac,ogg,ops,ogf,flc,wav,aif,pcm,mp3
[00:00:07.230] notify:228 notified server 192.168.69.4 hport 9000 cport 9090
I (8651) network_status: LMS IP: 192.168.69.4, hport: 9000, cport: 9090
I (9351) httpd_handlers: serving /status.json to peer 192.168.69.222 port 57543

</code></pre>
</details>

sle118, 2025-03-20:
@elektrinis which part of the logs show the moment when playback stops?  Also, do you mean that the player is streaming non-stop and stops after a few days, or that after a few days of being connected you are no longer able to stream?

I am assuming that streaming is from a media server?  Given that I rarely reboot any of my players here, I suspect this isn't related to our project but perhaps something on your wifi network layer or Lyron server itself. Did you check your server's log for any errors/warnings?

elektrinis, 2025-03-25:
I'm using it as radio, so basically streaming a few hours a day, in the evenings when we are home, or the whole weekend. When playing for some time, it begins to drop out, i.e. playback stops for a few seconds, then resumes, then stops and resumes again. Sometimes it does not resume at all (but status remains as "playing"). Usually when this happens, I'm able to start a new stream, but it drops out in few seconds. The longer it remains connected, the worse it gets, until new stream does not start at all, and it becomes unresponsive (via player interface in home assistant). I then load up the web interface and reboot it there, after which it works well for a few more days. If I wait a long time and let the problem become more serious, the web interface seems sluggish, reboot button does not work a few times.
I'm not sure which part of log is the exact moment, as I was getting a lot of log output, while playback situation was happening slowly.
What else I have tried:
a) remove power. Effect is the same as reboot via web interface, it recovers.
b) reboot WiFi access point. No improvement.
c) restart lyrion server and whole home assistant. No improvement.

Note that I have now replaced my hardware with "Louder ESP32-S3", which has onboard I2S amplifier.
With this one I had a similar problem, which appeared roughly after 10 days of use. I have restarted it via web interface and problem disappeared. I am still waiting for it to happen again.
I will be unable to get logs from this one, as I can't use USB while it is powered via external 19V brick. Unless this can be done wirelessly.

sle118, 2025-03-29:
You can enable telnet from the web UI. Make sure you disable it after collecting logs since it is not password protected. 

As for automating reboots, you could schedule a task in home assistant: 
https://community.home-assistant.io/t/automation-to-call-a-url-at-a-certain-time/553768

You could try something like
`curl -X POST -H "Content-Type: application/json; charset=utf-8" -d '{"timestamp": '$(date +%s)'}' http://localhost/reboot.json`

Replacing the host with the address of your player. If your player is configured in home assistant, you might even be able to use substitution to insert the address.

elektrinis, 2025-05-05:
A month has passed and it finally happened again.
When trying to telnet, I get an error "could not open connection to host". Not sure if I enabled it properly: in VNS editor I have set `telnet_enable` to `1`. Does not connect even after reboot.