sle118/squeezelite-esp32 #467: [Request] Support for MUSE RADIO analog buttons and battery voltage

MatoPotat0, 2025-01-17:
Hi! Great project! I have multiple esp32 devices for multiroom audio and all work great.

I have also purchased the new device from Raspiaudio, the [MUSE RADIO](https://raspiaudio.com/product/muse-radio/).

But it does not have the usual pinout and is not fully compatible with NVS editor ([DOCUMENTATION](https://forum.raspiaudio.com/t/muse-radio-installation-documentation/1091)). It uses analog pin (GPIO1) for 4 pushbuttons with an resistor network to determine which button was pressed.
The radio also has battery voltage monitoring pin on GPIO13, that is on the second ADC channel and as I understand ADC2 is not supported in the NVS editor so far.

It would be great if this type of configuration could be supported by the NVS editor.

Thanks in advance.

m-reiter, 2025-01-19:
Just chiming in to say I would also be very interested in having the buttons on the Muse Radio working. Any initial ideas on how feasible that would be? I guess they would need to be polled rather than handled by some kind of interrupt. There seems to be some button polling code present already, so maybe it wouldn't be that different after all?

I'd gladly try to help here or give this a try myself but I probably would need quite a bit of guidance.

RASPIAUDIO, 2025-01-20:
Because we were short on GPIOs, we used only one ADC (GPIO1) for four buttons by implementing a network of resistors. This technique is commonly used in commercial devices. You can find the example code used in our demo app [here](https://github.com/RASPIAUDIO/MuseRadio-DEMO/blob/main/RadioV01/RadioV01.ino).
```
int button_get_level(int nb)
{
#define THRESHOLD 200
#define maxB 3
  static int adcB[] = {1850, 2350, 450, 930};
  int adcValue, V;
  if ((nb > maxB) || (nb < 0))return -1;
  adcValue = analogRead(KEYs_ADC);
  V = adcB[nb];
  if (abs(V - adcValue) < THRESHOLD ) return 0; else return 1;
}
```

joede121, 2025-01-23:
It would be great to have this feature