sle118/squeezelite-esp32 #470: I2C volume control

RASPIAUDIO, 2025-01-26:
Hi!
How do I set I2C command for volume control?

i do see in the doc that I have to use the "dac_controlset"

{"volume":[{"reg":0,"val":128}...]}

but should introduce a volume variable somewhere or should I define every volume step? how?

Thanks!

RASPIAUDIO, 2025-01-28:
I have tried this in the NVS using the variable volume but using the I2C dedub tool it does not seems to use the register 46 and 47 as it should:

`{
  "init": [
    {"reg":3,"val":24}, {"reg":24,"val":2}, {"reg":38,"val":9}, {"reg":39,"val":144},
    {"reg":42,"val":144}, {"reg":43,"val":128}, {"reg":45,"val":128}, {"reg":27,"val":0},
    {"reg":26,"val":0}, {"reg":2,"val":240}, {"reg":2,"val":0}, {"reg":28,"val":32},
    {"reg":29,"val":32}, {"reg":4,"val":60}, {"reg":25,"val":0}, {"reg":46,"val":33},
    {"reg":47,"val":33}, {"reg":48,"val":33}, {"reg":49,"val":33}
  ],
  "volume": [
    {"reg":46,"val":"volume * 33 / 100"}, {"reg":47,"val":"volume * 33 / 100"}
  ]
}`

sle118, 2025-03-18:
Volume is controlled by backend and thus as far as DAC is concerned, its output volume should always be 100%. If you want to address some issues like hissing, etc when the player is idle, then there are other options. Please read the extensive documentation:
https://github.com/sle118/squeezelite-esp32?tab=readme-ov-file#daci2s

You will see the supported commands:
`Where <command> is one of init, poweron, poweroff, speakeron, speakeroff, headseton, headsetoff (it must be an array even for a single item). `

closed 2025-03-18 by sle118

RASPIAUDIOadmin, 2025-04-07:
Hi Sébastien,

Sorry to ask again—I just want to make sure I fully understand.

From your previous answer, I understood that the DAC output is always at maximum volume, and any volume reduction is done in software by adjusting the source volume directly in the backend.

Does this mean it's not possible in the current architecture to send I²C commands directly to the DAC to control volume?
The benefit of controlling the DAC volume directly would be improved audio quality, ensuring full 24-bit resolution regardless of the volume level and achieving a better signal-to-noise ratio.

Thanks for clarifying!