sle118/squeezelite-esp32 #471: Please add link to Web installer to README.me !

skiphansen, 2025-02-01:
I'm new to squeezelite-esp32 but a reasonably experienced firmware developer and this project
looks amazing!

At first I struggled to build squeezelite-esp32 from source using the current IDF and then gave up.  

Then I downloaded the most recent release and extracted squeezelite-esp32-master-v4.3-SqueezeAmp-32-21681.zip.
I found some documentation and scripts but I couldn't get that to flash because partitions.bin isn't in the release
anywhere I can find.

I started to create an issue for that problem when I discovered the webinstaller from the default issue template.
The webinstaller worked!

I had read the README multiple times and never found any mention of the webinstaller or I would
have tired that first.

Clearly lots and lots of people didn't have this problem, but they must have followed a tutorial or
documentation from elsewhere or I'm blind or more stupid that I give myself credit for !

elektrinis, 2025-02-07:
I had the same problem. +1

CornelisJ, 2025-02-11:
> Then I downloaded the most recent release and extracted squeezelite-esp32-master-v4.3-SqueezeAmp-32-21681.zip. I found some documentation and scripts but I couldn't get that to flash because partitions.bin isn't in the release anywhere I can find.

I fully agree that the webinstaller should be mentioned in the README

Just want to make the remark that partition-table.bin is included in the zip file, and that you can flash an ESP32 WRover module with squeezelite-esp32 using the zip file and the ESP32 Flash Download tool. See screenshot below.

[image]

skiphansen, 2025-02-11:
@CornelisJ You are right of course, it was right there by a slightly different name but I was blind.

I my defense had tried to build from sources first I was trying to use the generated writeSequeezeEsp.sh script:

```
#!/bin/bash

port=$1
baud=$2
esptool.py --port $port --baud $baud  write_flash --flash_mode dio --flash_freq 80m --flash_size detect  0x1000 bootloader/bootloader.bin 0x8000 partitions.bin
```

At that point I was very frustrated and missed finding partition-table.bin.

In any case since then I've been able to compile and build this amazing code for my "cheap yellow display' board
and it's working! 

[image]

One room down, 3 to go.