sle118/squeezelite-esp32 #473: Squeezelite-ESP32: Spotify Artwork Download Fails Due to HTTPS

nanostra, 2025-02-16:
Hello, 

I am using a Louder-ESP32-S3 with Squeezelite-ESP32 in Spotify Connect mode (cspot). The ESP32 is configured to display artworks for the currently playing tracks. I am using the latest firmware version provided at this address https://sonocotta.github.io/esp32-audio-dock/, as I did not see any release for the Louder-ESP32-S3 N16R8 on your GitHub. I couldn't find where the firmware version is displayed!..sorry...

🚨 Issue encountered

When Spotify sends an image URL (artwork) for the current track, the ESP32 attempts to download it but fails with the following error in the logs. Squeezelite then completely crashes, and I have to reboot and remove the artwork:1 option.

```
I (50365) cspot: requesting artwork https://i.scdn.co/image/ab67616d00001e029a494f7d8909a6cc4ceb74ac
I (50365) display: playing Chris Brown - Under The Influence
[00:05:13.786] D MercurySession.cpp:252: Executing Mercury Request, type SEND
E (50415) HTTP_CLIENT: No transport found
E (50425) HTTP_CLIENT: Please enable HTTPS at menuconfig to allow requesting via https
W (50425) cspot: artwork error or too large 0
```
Is there an option to force the request to use HTTP?
Is there a simple way to resolve this issue?

Thank you for your great work! 😊

philippe44, 2025-02-17:
It's a config error from the s3 build.. As I don't provide "official" s3 build, the author should change menuconfig to enable HTTPS

nanostra, 2025-02-17:
@anabolyc (sonocota) has added the option... but now we have this,

```
[00:00:57.427] I MercurySession.cpp:42: Received packet, command: 178
[00:00:57.431] D MercurySession.cpp:174: Received mercury packet
[00:00:57.460] I MercurySession.cpp:42: Received packet, command: 178
[00:00:57.464] D MercurySession.cpp:174: Received mercury packet
[00:00:57.795] I TrackPlayer.cpp:206: Playing
[00:00:57.797] I Shim.cpp:120: new track started <> => <de8bfbbe2281f3162ebb36b89523e4d624cc484f>
[00:00:57.809] I Shim.cpp:262: next track will play in 0 ms
[00:00:57.827] _output_frames:150 track start sample rate: 44100 replay_gain: 0
[00:00:57.885] I Shim.cpp:416: next track's audio has reached DAC (offset 6085)
[00:00:57.893] D MercurySession.cpp:252: Executing Mercury Request, type SEND
I (58949) cspot: requesting artwork https://i.scdn.co/image/ab67616d00001e02c58b581c98eafab982391a6d
I (58949) display: playing Marnik - Boyz In Paris (with VINAI)
[00:00:57.951] D MercurySession.cpp:252: Executing Mercury Request, type SEND
E (59029) esp-tls-mbedtls: No server verification option set in esp_tls_cfg_t structure. Check esp_tls API reference
E (59029) esp-tls-mbedtls: Failed to set client configurations, returned [0x8017] (ESP_ERR_MBEDTLS_SSL_SETUP_FAILED)
E (59039) esp-tls: create_ssl_handle failed
E (59049) esp-tls: Failed to open new connection
E (59049) TRANSPORT_BASE: Failed to open a new connection
E (59059) HTTP_CLIENT: Connection failed, sock < 0
W (59069) cspot: artwork error or too large 0
I2S abort!
```

https://github.com/sonocotta/esp32-audio-dock/issues/32#issuecomment-2664128966

philippe44, 2025-02-17:
I'll have to look at that, but likely there is a config to disable SSL server verification, a common issue

philippe44, 2025-02-17:
> I'll have to look at that, but likely there is a config to disable SSL server verification, a common issue
>
Indeed, in ESP-TLS, you have to:
[*] Allow potentially insecure options
[*]     Skip server certificate verification by default (WARNING: ONLY FOR TESTING PURPOSE, READ HELP)

nanostra, 2025-02-17:
Thank you for your feedback.

closed 2025-03-18 by sle118