sle118/squeezelite-esp32 #481: Bluetooth output mode is extremely unstable

xqyang2008, 2025-03-26:
First of all, I sincerely appreciate the developers' hard work.

### Describe the bug
When using the I2SDAC and SPDIF output modes, everything works fine. However, the Bluetooth output mode is extremely unstable. I kept trying for two days and was only able to successfully play a song once, after which it stopped working again.

The specific issue is that when using the Bluetooth output mode, as soon as playback starts in LMS, the player goes offline after about 2-3 seconds, displaying "Player not connected."

Additional information: My Wi-Fi router operates in high-performance mode, using channels 8 and 10 in the 2.4GHz band. The signal strength at the location of the squeezeAMP is -21dBm.

### Preliminary Information

	Firmware Version: 2.1681-16#v4.3#SqueezeAmp#master-v4.3
	Plugin Version: LMS (Logitech Media Server).
		server：
			version: 9.0.2 - 1741501853 @ Thu Mar 13 14:57:33 UTC 2025
			IP: 192.168.0.80
		Plugin：
			Additional Browse Modes 1.0
			Don't Stop The Music 1.0.0
			Drag & drop music files to play 0.1
			Full text search 1.0.0
			Material主题 5.8.3
			Online Music Library Integration 1.0
			Presets Editor 1.0
			Radio 1.0
			Radio Now Playing 0.0.49
			Random Mix 1.0
			Remote Music Libraries 1.0
			Song Scanner 2.0
			Sounds & Effects 1.1

### Hardware Details

squeezeAMP v3.2

### NVS Settings

<details>
<pre><code>
	{
	  "a2dp_ctmt": "1000",
	  "a2dp_ctrld": "500",
	  "a2dp_dev_name": "Squeezelite",
	  "a2dp_sink_name": "ULT FIELD 1",
	  "a2dp_spin": "0000",
	  "actrls_config": "",
	  "airplay_name": "ESP32-AirPlay-28957c",
	  "airplay_port": "5000",
	  "ap_channel": "1",
	  "ap_ip_address": "192.168.4.1",
	  "ap_ip_gateway": "192.168.4.1",
	  "ap_ip_netmask": "255.255.255.0",
	  "ap_pwd": "squeezelite",
	  "ap_ssid": "squeezelite-28957c",
	  "autoexec": "1",
	  "autoexec1": "squeezelite  -o bt -s 192.168.0.80 -b 500:2000 -C 30 -d log=all",
	  "bat_config": "",
	  "bt_name": "ESP32-BT-28957c",
	  "bt_sink_pin": "1234",
	  "bt_sink_volume": "127",
	  "bypass_wm": "0",
	  "cspot_config": "{\"deviceName\":\"squeezelite-28957c\",\"bitrate\":160,\"zeroConf\":1}",
	  "dac_config": "",
	  "dac_controlset": "",
	  "dhcp_tmout": "8",
	  "display_config": "",
	  "enable_airplay": "1",
	  "enable_bt_sink": "1",
	  "enable_cspot": "1",
	  "equalizer": "",
[... 225 lines cut ...]
	I (16468) bt_app_source: Discovery started.
	I (29288) bt_app_source: Device discovery failed, continue to discover...
	I (29288) bt_app_source: Discovery started.
	I (30758) bt_app_source: event: 16
	I (33668) RCCT: AVRC conn_state evt: state 1, [50:1b:6a:8b:33:b8]
	I (33668) RCCT: AVRC remote features 24b, TG features 2
	I (33738) RCCT: remote rn_cap: count 1, bitmask 0x2000
	I (39928) bt_app_source: event: 13
	I (42108) bt_app_source: Device discovery failed, continue to discover...
	I (42118) bt_app_source: Discovery started.
	I (54938) bt_app_source: Device discovery failed, continue to discover...
	I (54938) bt_app_source: Discovery started.
	I (67758) bt_app_source: Device discovery failed, continue to discover...
	I (67758) bt_app_source: Discovery started.
	[00:01:11.168] notify:228 notified server 192.168.0.80 hport 9000 cport 9090
	I (72588) network_status: LMS IP: 192.168.0.80, hport: 9000, cport: 9090
</code></pre>
</details>

### Issue Description

1. **Observed Behavior**: I know that making Wi-Fi and Bluetooth work simultaneously is difficult, but I still hope that Squeezelite-ESP32 can achieve this.
2. **Expected Behavior**:  I still hope that Squeezelite-ESP32 can achieve this.
3. **Steps to Reproduce**: Reproducing this issue on my end is very easy. After flashing the firmware using a USB-to-serial tool, setting up Wi-Fi, and configuring the Bluetooth output mode, the issue naturally occurs. I have repeated this process nearly a hundred times.
I kindly request assistance, and once again, thank you for the developers' dedicated efforts.

sle118, 2025-03-29:
Thank you for providing all the details when documenting your issue. This saves precious time.

Please have a look at this section of the documentation, which explains some required resampling parameters when streaming to Bluetooth.
https://github.com/sle118/squeezelite-esp32?tab=readme-ov-file#raw-wrover-module

Also note that while this setup might work for some users, your mileage may vary. That capability is something that I built in early into the project while I was waiting for a proper DAC to arrive in the mail so it has always been more of a hack. There is very little that could be done from a software perspective, as streaming audio in via WiFi and out to Bluetooth is very demanding for the chip's architecture as Bluetooth and WiFi need to coexist and share the radio's bandwidth. In addition to this, transmitting Bluetooth requires an extra encoding step that may overwhelm the poor chip's performances. Setting up the options to force resampling may help, but nothing is guaranteed

xqyang2008, 2025-03-29:
Yes, I carefully read the document you wrote. It mentions that when sending audio to a Bluetooth speaker, only a 44.1kHz sampling rate can be used. Therefore, from the beginning, I encoded all my music on LMS to a 44.1kHz sampling rate and 16-bit depth. During the debugging process, I also tried setting resampling to 44.1kHz, but it didn’t help.

I kept trying for two days, and it finally worked, playing ten songs stably. At that moment, I thought the issue was resolved. However, when I tried using it again about six hours later, it stopped working properly, even though I hadn’t changed any software or hardware settings, nor had I shut down or disconnected the power. So, the issue might also be related to Wi-Fi channel congestion in the room at the time. Unfortunately, I didn’t manage to save any logs.

As a software and hardware engineer myself, I understand the challenges of getting Bluetooth and Wi-Fi to work simultaneously and stably. Given the current hardware conditions, spending too much time on this issue may not be worthwhile. So, I switched to speakers with a 3.5mm analog input and an SPDIF interface, both of which have been working stably. Therefore, I don’t think you need to spend too much time on this issue either.

Finally, I sincerely appreciate your efforts.

sle118, 2025-03-29:
There's an option to dump statistics. They contain buffer info, etc. Philippe used that to optimize the system

cmonrroy88, 2025-05-13:
@xqyang2008 Same issue here. I'm trying to use an Echo Dot 5th as Bluetooth speaker and its hard to pair them with the ESP32-WROVER. So when its finally paired, play a random amount of song very nice and next the ESP32 reboot itself.

Then the ESP32 does not pair with the Bluetooth speaker. I have to restart many times both devices to make it work again

lisz8, 2025-05-14:
@cmonrroy88 The first connection attempt after ESP is turned on seems critical: when it fails, ESP sometimes becomes unresponsive: the web server does not respond, telnet does not connect. When the connection to the speaker is lost later, such dramatic effects do not occur.

What helped me to some extent was clearing other paired devices from the BT speaker. The telnet connection helps to find out what is happening (the event log on the www page does not contain useful information).

cmonrroy88, 2025-05-15:
@lisz8 You right, I did it with my echo dot and this increase the stability a little. 

But I'm still having several issues when I play music by one hour approx, the esp32 reboot with no reason so, at this moment I'm trying with stream music from music assistant (it's the music server that I'm currently using) in 44.1 kHz sampling rate with 16bits and now it's more stable.

lisz8, 2025-05-17:
@cmonrroy88 That's a very good idea, I'm sure it helps not to overload the ESP. 

Another thing that helped me was moving the ESP further away from the speaker. I noticed that there were significantly more dropouts when they were very close together. There was probably more interference then.