sle118/squeezelite-esp32 #482: Build fails because of .gitattributes

AndreKR, 2025-03-27:
When you try building on Windows using Docker as described in the readme, the build will fail with an error message:

```
[653/1551] Generating nanopb/generator/proto/nanopb_pb2.py
FAILED: esp-idf/spotify/cspot/nanopb/generator/proto/nanopb_pb2.py
cd /project/build/esp-idf/spotify/cspot && /project/components/spotify/cspot/bell/external/nanopb/generator/protoc -I/project/build/esp-idf/spotify/cspot/nanopb/generator/proto --python_out=/project/build/esp-idf/spotify/cspot/nanopb/generator/proto /project/build/esp-idf/spotify/cspot/nanopb/generator/proto/nanopb.proto
/usr/bin/env: ‘python3\r’: No such file or directory
[657/1551] Building C object esp-idf/esp_serial_slave_link/CMakeFiles/__idf_esp_serial_slave_link.dir/essl_spi.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

The key information here is:
```
/usr/bin/env: ‘python3\r’: No such file or directory
```

Note the `\r` after the command.

This is because there is [a line in the `.gitattributes` file](https://github.com/sle118/squeezelite-esp32/blob/cecb7fd87651ba12cf03e51c683c3a2404762394/.gitattributes#L2) that instructs Git to convert all LF line endings to CRLF line endings on Windows when checking out the project.

These CRLF line endings then cause the build scripts in the Docker container to fail.

There are a number of ways to fix this, like removing the line altogether or adding `eol=lf` after it.  I guess which one is preferable depends on why the line was added in the first place.

sle118, 2025-03-29:
I'm going to ask @philippe44 to consider the suggestion. 

In the meantime, some users have been using workarounds: 
https://github.com/sle118/squeezelite-esp32/issues/256#issuecomment-1567444671

philippe44, 2025-03-30:
I removed that

sle118, 2025-03-31:
And our build was successful!

closed 2026-10-09 by philippe44