sle118/squeezelite-esp32 #490: [Feature Request] Ability to make use of built in microphones for Home Assistant Voice

Meaparvitas111, 2025-04-19:
Hi, I hope this was not requested before.

I think it would be a big enhancement if the built in microphones could be used for Voice Assistant in Home Assistant als with the Voice Preview Edition or an ESP32-S3 Box.

The Audio Kit Board can already be used for TTS messages or sound output via squeezelite but I would be very nice if it could also be used to recognize and transfer voice commands to Home Assistant.

Perhaps it is not very diffcult to add this feature to the code but it would offer a lot of additional use cases.

Thanks in advance and kind regards.

Keep up the good work!

Meaparvitas

DavidSpivey, 2025-07-05:
+1 for this feature. I realize this would involve a non-trivial amount of additional coding, but these A1S devices seem to have the capability, if it could be programmed. It would be a huge deal for those who are trying to keep their voice data local.

trackhacs, 2025-09-20:
+2

rginon314, 2026-01-17:
+3