sle118/squeezelite-esp32 #492: AirPlay stops after 10 minutes

larryqiann, 2025-05-15:
To help us resolve your issue as quickly as possible, please follow these guidelines when submitting an issue. Providing all the necessary information will save both your time and ours.

### Describe the bug
A clear and concise description of what the bug is.
AirPlay receiver stops playing audio after ten minutes or so
### Preliminary Information

1. **Firmware Version**: Specify the version of the firmware you are using.
I2S-4MFlash-1681-16bit
2. **Plugin Version**: Mention the version of the plugin installed on your LMS (Logitech Media Server).
N/A
### Hardware Details

Please describe your hardware setup:

- **ESP32 Module**: For example, ESP32 WROVER, ESP32-S3, etc.
- **Board Type**: If applicable, e.g., ESP32 audio kit, etc.
- **DAC Chip**: Specify the DAC chip you are using.
- **Additional Hardware**: Include details about any other hardware like rotary controls, buttons, screens (SPI, I2C), Ethernet, IO expansion, etc.

ESP32-A1s (AC101 version)
### NVS Settings

Follow these steps to share your NVS settings:

1. Open the web UI of your device.
2. Click on the "Credit" tab.
3. Enable the "Show NVS Editor" checkbox. This allows you to view or change the NVS configuration even when not in recovery mode.
4. Navigate to the "NVS Editor" tab.
5. Scroll to the bottom and click "Download Config".
6. Share the downloaded content here.

<details>
<pre><code>
  {
  "a2dp_ctmt": "1000",
  "a2dp_ctrld": "500",
  "a2dp_dev_name": "Squeezelite",
  "a2dp_sink_name": "SMSL BT4.2",
  "a2dp_spin": "0000",
  "actrls_config": "",
  "airplay_name": "ESP32-AirPlay-1fe1c0",
  "airplay_port": "5000",
  "ap_channel": "1",
  "ap_ip_address": "192.168.4.1",
  "ap_ip_gateway": "192.168.4.1",
  "ap_ip_netmask": "255.255.255.0",
  "ap_pwd": "squeezelite",
  "ap_ssid": "squeezelite-1fe1c0",
  "autoexec": "1",
  "autoexec1": "squeezelite -o I2S -b 500:2000 -d all=info -C 30 -W -n squeezelite-1fe1c0",
  "bat_config": "",
  "board_model": "ESP32-A1s",
  "bt_name": "ESP32-BT-1fe1c0",
  "bt_sink_pin": "1234",
  "bt_sink_volume": "127",
  "bypass_wm": "0",
  "cspot_config": "{\"deviceName\":\"squeezelite-1fe1c0\",\"bitrate\":160,\"zeroConf\":1,\"volume\":0}",
  "dac_config": "model=AC101,bck=27,ws=26,do=25,di=35,sda=33,scl=32",
  "dac_controlset": "",
  "dhcp_tmout": "8",
  "display_config": "",
  "enable_airplay": "1",
  "enable_bt_sink": "1",
  "enable_cspot": "1",
  "equalizer": "",
  "eth_config": "",
  "ethtmout": "8",
  "gpio_exp_config": "",
  "host_name": "squeezelite-1fe1c0",
[... 92 lines cut ...]
[00:16:59.596] rtp_request_resend:858 [0x3fad2620]: SENDTO failed (Not enough space)
[00:16:59.661] rtp_request_resend:858 [0x3fad2620]: SENDTO failed (Not enough space)
[00:16:59.736] rtp_request_resend:858 [0x3fad2620]: SENDTO failed (Not enough space)
</code></pre>
</details>

### Issue Description

1. **Observed Behavior**: Describe what you think is wrong.

The AirPlay audio stream breaks up sometimes. When this happens, the other functionality i.e. Spotify Connect also breaks. The "not enough space" error continues to show
2. **Expected Behavior**: Describe what you expect should happen.

The AirPlay audio stream should self recover once the network is stable, and the device should not slow down or crash when the AirPlay network encounters errors.
3. **Steps to Reproduce**: Provide a step-by-step guide on how to replicate the issue.

Flash onto blank device, play audio from AirPlay on iPhone or Mac, wait ten or so minutes. 

The device is placed next to the router, so the signal should be excellent. The device often becomes unstable once AirPlay encounters the SENDTO failed errors. I have tried to check whether the problem mentioned previously in #446 is relevant; the pins are at 3.3V as expected whether or not the serial monitor is connected.

Wondering what might be the cause of this issue. Perhaps the RAM is being used by the generated errors building up?

I understand that the device is not able to buffer significant amounts of audio, but it seems like more than a networking failure; receiving too little data appears to overwhelm the device. Would using a device with a larger PSRAM help?

Thanks!

mistikdonbass, 2025-05-16:
I have a similar problem, appeared with the latest update LMS 9.0.2.0 in Home assistant. Regardless of radio or local music playing, playback is interrupted for a few seconds and so on regularly. There is nothing wrong in the HA add-on logs, but in the esp32 logs there is 'timeout'.

[image]

philippe44, 2026-10-09:
Telnet and other issues have been fixed recently, included many crashes that would cause audio to stop

closed 2026-10-09 by philippe44