sle118/squeezelite-esp32 #505: Unable to build & flash squeezelite on esp32-s3

pkuttiyam1, 2025-09-20:
### Describe the bug
I want to be able to build and run Spotify Connect on my esp32-s3 (ESP32-S3 Wroom Devkit-C1)
### Preliminary Information

1. **Firmware Version**: Specify the version of the firmware you are using.
2. **Plugin Version**: Mention the version of the plugin installed on your LMS (Logitech Media Server).

### Hardware Details

- **ESP32 Module**: ESP32-S3 Wroom Devkit-C1
- **Board Type**: If applicable, e.g., ESP32 audio kit, etc.
- **DAC Chip**: MAX98357

### Details
I tried two things:
1. Build via Docker environment:
idf.py set-target esp32s3
idf.py build 

sdkconfig or the idf.py menuconfig - does not show that it is getting built for esp32s3. It still shows esp32.
Is there another sdkconfig that I should be using when building for ESP32-S3 ?

2. I actually just need the ESP32-S3 loaded with squeezelite binaries. So, I tried the web installer:
https://sle118.github.io/squeezelite-esp32-installer/

I chose my device via USB, and clicked on Generic/I2S, then Clicked on Install "I2S-4MFlash-16", it returned an error saying:
Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL.

Can someone tell me how I can install these images on to my ESP32-S3 devices ?

trackhacs, 2025-09-20:
here you can flash a recompiled version for the S3 - not the latest version - but works for me 

https://wizmo2.github.io/TAudio-Case/

wizmo2, 2025-11-05:
watch out for windows line endings (search closed issues on the 'nanopb' error).  There's also some additional dependencies (https://github.com/sle118/squeezelite-esp32?tab=readme-ov-file#building-squeezeesp32)

anabolyc, 2026-02-01:
Hey everyone

I did some work on SDKCONFIG and updated to the latest master code. It seems to work now with both ESP32 and ESP32-S3, you are welcome to test it: https://sonocotta.github.io/esp32-audio-dock/

anabolyc, 2026-02-03:
> [@anabolyc](https://github.com/anabolyc), thank you for the work. For some reason the "hifi s3" version failed to download so I had to use the "hifi s3-plus", and remove a bunch of configs I didn't need from NVS. I noticed BT was configured there. Did you manage to get BT working on the S3? I disabled it because I heard it isn't working on s3, but if I can help testing LMK.
> 
> Airplay is somewhat working, but it crackles randomly. I'm not sure if it's my config or DAC or what though; new hardware.

Thanks, fixed that. Mistype on my side.

Can you point me where BT was configured? That's obviously a mistake, since S3 does not possess it :)

Airplay was working for me quite smoothly, although i din't run it for too long